Sparronator9999/YAMDCC #2: SVC ctd

porkmanager, 2024-09-02:
Service crashin on my GF66-12UG. Probably wrong config file. Seems i should to wait for the configs tutorial  
[yamdccsvc.exe.1.log.gz](https://github.com/user-attachments/files/16840898/yamdccsvc.exe.1.log.gz)

Sparronator9999, 2024-09-02:
Hi @porkmanager, can you please try the latest build when it's ready?

I think the issue was possibly caused by your system language, which I've hopefully fixed in the latest commit.

Sparronator9999, 2024-09-02:
> Seems i should to wait for the configs tutorial

Funnily enough, I don't think the problem was caused by a bad config. The service didn't launch far enough to even attempt to load a config

porkmanager, 2024-09-03:
> Hi @porkmanager, can you please try the latest build when it's ready?
> 
> I think the issue was possibly caused by your system language, which I've hopefully fixed in the latest commit.

Oh, my win lang is RU. Now its working as well with the latest build. Thank you bro! I will test it. Very uncial software, there no  alternatives for MSI bloatware on the windows os

porkmanager, 2024-09-03:
Damn, seems it is full functional on my GF66-12UG. Im so happy! I tested almost every function(didnt tested battery limit yet) and its works properly. Seems all fresh msi laptops has the same EC codes. Before your app I used RWeverything for setting fan curves lol.
Thank you for your work!

closed 2024-09-03 by Sparronator9999