Sparronator9999/YAMDCC #3: 2 issues. Self disabling and wrong crash logfile default name

porkmanager, 2024-09-03:
Not working selfdisabling feature and inaccessible default logfile name
pinned a video

[crash-2024.txt](https://github.com/user-attachments/files/16849543/crash-2024.txt)

Sparronator9999, 2024-09-03:
It looks like I can reproduce both issues on my end.

For the crash report saving issue, it looks like my sh*tty code is trying to set a default filename with colons ":" in it (which aren't allowed in a filename in Windows).

This is what happens when you don't test features, people...

I will fix both issues later.

Sparronator9999, 2024-09-04:
Hi, can you please test the latest build from GitHub Actions when it finishes?

Both issues are now fixed on my end.

porkmanager, 2024-09-04:
> Both issues are now fixed on my end.

Yep. Ive just tried latest build and it has been fixed. Thank u!

Sparronator9999, 2024-09-04:
No problem!

closed 2024-09-04 by Sparronator9999