Sparronator9999/YAMDCC #5: Service crash on startup wrong config

FeelFried, 2024-09-06:
### Prerequisites
- [ ] I am using the correct [issue template](https://github.com/Sparronator9999/MSIFanControl/issues/new/choose), i.e. I am asking the developer of MSI Fan Control a question. Bug reports and feature requests should be opened using their respective templates.
- [ ] I have read the [FAQ](https://github.com/Sparronator9999/MSIFanControl#faq). Your question might already be answered there.
- [ ] I have searched the [issue tracker](https://github.com/Sparronator9999/MSIFanControl/issues?q=label:"type:+question") for duplicate or similar questions. Your issue will be closed if it is a duplicate of another issue.
- [ ] I have not answered truthfully to the above questions.

### Question
so... Seems I broken it with wrong config and my stupidity... I loaded broken config, that crashed sevice, I tried to run it again - it crashed on startup, I tried to delete and reinstall service - it still crashes on startup ( 

Service cannot be started. System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at YAMDCC.Service.svcFanControl.ApplySettings() in D:\a\YAMDCC\YAMDCC\YAMDCC.Service\svcFanControl.cs:line 317
   at YAMDCC.Service.svcFanControl.OnStart(String[] args) in D:\a\YAMDCC\YAMDCC\YAMDCC.Service\svcFanControl.cs:line 117
   at System.ServiceProcess.ServiceBase.ServiceQueuedMainCallback(Object state)

Sparronator9999, 2024-09-06:
Hi, can you please upload the following files from `C:\ProgramData\Sparronator9999\YAMDCC`:
- `CurrentConfig.xml` - This is the config the service is trying to load
- In the `Logs` folder, `yamdccsvc.exe.log` (or one of the `.gz` files that shows the above crash. You can use 7-zip to extract them)

Sparronator9999, 2024-09-06:
I have assigned the Bug tag as this appears to be caused by a config loader bug.

Sparronator9999, 2024-09-06:
>     * [x]  I have not answered truthfully to the above questions.

Please read the template again more carefully.

FeelFried, 2024-09-06:
full folder 
[Logs.zip](https://github.com/user-attachments/files/16913550/Logs.zip)

Sparronator9999, 2024-09-06:
Looks like the problem is being caused by the service trying to select a performance mode that doesn't exist (a `ModeSel` value of 7 was in the XML under the `PerfModeConf` tag, when there's only 4 performance modes to choose from).

Did you edit the XML directly, or was this done through the configuration utility (YAMDCC.exe)?

Either way, the crash fix would be adding a simple bounds check when loading the config.

FeelFried, 2024-09-06:
I was editing it directly. 
btw... not related to crash but... do you know why all performance mods except silent limits CPU PL1 to 45 Watts? )

Sparronator9999, 2024-09-06:
I have updated YAMDCC to check for out-of-bounds conditions for a couple of config settings.

> btw... not related to crash but... do you know why all performance mods except silent limits CPU PL1 to 45 Watts? )

I'm not sure, but I think it's a limit imposed by the laptop BIOS/firmware to keep the laptop within thermal/power limits. My laptop already runs at 80+ degrees when at 45W.

Another thing to note is that this limit drops to 25W a little bit after you start using the dGPU at the same time as the CPU. Again, I think this is to keep things within the thermal capacity of the laptop.

FeelFried, 2024-09-06:
"I think this is to keep things within the thermal capacity of the laptop." I thought same but.... If I choose silent mode it uses PL1 that I specify in bios (tested with 80W works fine) but also it sets fan speed to 60% max and not reacting to sliders in YAMDCC ( except turbo cooling mod which still can be activated )

Sparronator9999, 2024-09-06:
> turbo cooling mod

Is this the Full Blast (Cooler Booster in MSI Center) setting?

As for 80W CPU limit, I think that would absolutely melt my CPU, especially at a 60% limit. (my laptop only has one fan)

Anyways, if you haven't already, can you test the latest commit to check that the service no longer crashes when attempting to set an out-of-bounds `ModeSel` setting?

Sparronator9999, 2024-09-06:
> If I choose silent mode it uses PL1 that I specify in bios

I find it quite fascinating that the Silent mode allows a *higher* CPU power draw than Balanced or High Performance settings, despite limiting the fan speed to 60%. You haven't installed any modded BIOS/EC firmware on your laptop, have you? I know of at least one that raises the CPU power limit on a specific laptop model, but haven't tested it due to my laptop not being supported.

FeelFried, 2024-09-06:
> Use the same config you uploaded to me
it just crashes on startup with this 
[yamdccsvc.exe.1.log.gz](https://github.com/user-attachments/files/16913676/yamdccsvc.exe.1.log.gz)

> You haven't installed any modded BIOS/EC firmware
Nope default latest BIOS from MSI, but laptop is MSI Katana 15 B13VGK  with i713600H and 4070, with Full Blast on it's able to maintain 80 degreese under AIDA stress test with PL1 = 80W, I just wanned to set it a bit lower ( to 80% fan speed )

FeelFried, 2024-09-07:
also... balance mode in MSI center limits PL1 to 60 Watts, but YAMDCC  with balance mode limits it to 45

FeelFried, 2024-09-07:
oh... I just realized.... seems MSI mixed up High performance and Silent mode power limits... that's why silent mode uses BIOS power limits and High performance mod lock PL1 to 45W...

Sparronator9999, 2024-09-07:
> it just crashes on startup with this [yamdccsvc.exe.1.log.gz](https://github.com/user-attachments/files/16913676/yamdccsvc.exe.1.log.gz)

Hmm, that log looks fine. Try running the service again, then upload the latest log (`yamdccsvc.exe.log`) *without* restarting the service.

> I had to install Intel DDT xD

What's Intel DDT? I use ThrottleStop to set my CPU power limits (and apply a modest undervolt)

Sparronator9999, 2024-09-07:
The logs ending with .gz are archived logs (there should be up to 9 of them), with higher numbered ones being older. Upon exit of the service (either graceful or after crashing), there should be one `yamdccsvc.log`, and up to 9 `yamdccsvc.X.log.gz` files, where X is a number from 1 to 9.

FeelFried, 2024-09-07:
> Try running the service again

srry mate, I haven't replaced the service exe with the new one last time, now with new exe from new build it runned fine  and cleared config on load.

> What's Intel DDT

since 11th gen with latest bios updates Intel requiers to install Dynamic Tuning Technology driver otherwise MSI performance plans from EC wouldn't work properly 

> I use ThrottleStop to set my CPU power limits

yeap I was trying it as well, but without DDT driver installed MSI performance mode overrides any power limits that you set in BIOS or in ThrottleStop, but now after I installed it everything is fine.

Sparronator9999, 2024-09-07:
> srry mate, I haven't replaced the service exe with the new one last time

That's fine. Good to see it works fine now!

> since 11th gen with latest bios updates Intel requiers to install Dynamic Tuning Technology driver otherwise MSI performance plans from EC wouldn't work properly

Hmm, my laptop doesn't appear to have/need a DTT driver. Unless it's included in the OEM GPU drivers, but I ain't installing those (they're both incredibly out of date).

Anyways, my laptop already roasts itself at 45W continuous, so I don't think I need to worry about increasing the power limit.

> without DTT driver installed MSI performance mode overrides any power limits that you set in BIOS or in ThrottleStop

In my experience, this doesn't happen if I *decrease* the power limit below 45W, but it can if I *increase* the power limit above 45W, or the dGPU becomes active (when it gets capped at 25W).

Anyways, I'm closing this issue now because the original bug reported is solved now.

closed 2024-09-07 by Sparronator9999