Sparronator9999/YAMDCC #13: Additional powersaving

porkmanager, 2024-09-07:
Spawned in idea how to improve @max battery@ perf. mode. If it possible would be good to add changing display refresh rate to 60hz when enabling max battery mode

Sparronator9999, 2024-09-07:
Maybe. My laptop doesn't have a high refresh rate display, though, so I have no way to test if a potential solution would work.

If I do add this feature, it probably won't be until after the 1.0 release.

I will add refresh rate switching to the planned feature list.

Sparronator9999, 2024-09-07:
Actually, on second thought, does Windows provide a power plan option for refresh rate?

Sparronator9999, 2024-09-07:
Okay, I did some research and it looks like changing the display refresh rate shouldn't be too hard.

However, I also don't think this should be in the base YAMDCC.

BUT, before you complain, I am thinking of adding a "plugin" system to YAMDCC, where knowledgeable people can program their own features into YAMDCC. I might add a refresh rate changer to one of these plugins.

This will still be after v1.0, though. So you might be waiting a while, especially since I'm working on another big project (which isn't public yet, by the way).

porkmanager, 2024-09-08:
> Actually, on second thought, does Windows provide a power plan option for refresh rate?

yea. windows power plans has the setting about intel graphics device state, but it does nothing
[image]

> This will still be after v1.0, though. So you might be waiting a while, especially since I'm working on another big project (which isn't public yet, by the way).

Thats sounds great!

Sparronator9999, 2025-01-01:
Hi, I have created a new project to auto-change your refresh rate when you switch to battery power:
[RefreshRateTuner](https://github.com/Sparronator9999/RefreshRateTuner).

I know this isn't quite what you want, but YAMDCC plugins/extensions are probably quite a while off.

porkmanager, 2025-01-02:
> Hi, I have created a new project to auto-change your refresh rate when you switch to battery power: [RefreshRateTuner](https://github.com/Sparronator9999/RefreshRateTuner).
> 
> I know this isn't quite what you want, but YAMDCC plugins/extensions are probably quite a while off.

Nice one! Thanks mate! I will try it.

For now im using simple combo with 2 programs for changing refrashrate automatically: [power event provider](https://github.com/hirschmann/powereventprovider) and an old cmd utility Qres. I added to windows task scheduler trigger for power event to set refreshrate 60hz