Sparronator9999/YAMDCC #6: ec monitoring

porkmanager, 2024-09-07:
[image]

Latest build for now. Doesent work ec monitoring

Sparronator9999, 2024-09-07:
Damn it, it's happening to you as well.

I think it has something to do with the fact that the service is *trying* to crash, but isn't for some reason.

Sparronator9999, 2024-09-07:
What's causing it is the latest update to add keyboard backlight support.

I'm assuming you haven't added the new `KeyLightConf` XML tag to your config yet (required for keyboard backlight to function), otherwise stuff breaks currently.

porkmanager, 2024-09-07:
> If you go to the Extras tab, you'll notice a new greyed out option for adjusting the keyboard backlight (at least I hope it's greyed-out. If not, I swear...)

Yes it is greyed

> I'm assuming you haven't added the new `KeyLightConf` XML tag to your config yet (required for keyboard backlight to function), otherwise stuff breaks currently.

yep i didnt add to the config that value. And with your config MSI-GF63-Thin-11SC.xml ec still not working and kb baklit option also greyed out

porkmanager, 2024-09-07:
> And with your config MSI-GF63-Thin-11SC.xml ec monitoring still not working and kb baklit option also greyed out

Oh, I have restarted the service and everything is working properly with your config. Thx bro for this great update!

Sparronator9999, 2024-09-07:
Oh... I was just about to ask you to try the latest build that attempts to log more exceptions (which I think is what was happening before you updated).

I should probably fix the config system yet again so that this doesn't happen again...

closed 2024-09-07 by Sparronator9999