Sparronator9999/YAMDCC #7: Saved config view

porkmanager, 2024-09-07:
Here is how your config looks in editor:
[image]

And here is how looks like saved config from yamdcc:
[image]

In web browser window saved config looks properly, thats happenin just in text editor
[image]

Not a big issue

Sparronator9999, 2024-09-07:
About that... it's kind of intentional when saving to the service's working config (`CurrentConfig.xml`), to save configs that look like that.

As for configs saved by the user to an XML wherever they choose (e.g. Desktop), you're right, they might want it to be "beautified" XML for easier editing (especially since you can't edit everything in a config from the configuration app).

I will look into this.

Sparronator9999, 2024-09-07:
The latest commit has a fix for this issue.

closed 2024-09-07 by Sparronator9999