Sparronator9999/YAMDCC #8: Here we go again ;) Doesent working stop and uninstall feature

porkmanager, 2024-09-07:
https://github.com/user-attachments/assets/e531487d-b3da-4143-8228-6bf5a5afdcb0

Sparronator9999, 2024-09-07:
Deja Vu
I've Just Been In This Place Before

In all seriousness, are you sure it's not that particular application?

I checked in Task Manager and Windows' built-in Services applet, and the service seems to get stopped correctly.

Sparronator9999, 2024-09-07:
Actually, what's the name of the program so I can test it?

porkmanager, 2024-09-07:
> Actually, what's the name of the program so I can test it?

https://github.com/winsiderss/systeminformer

porkmanager, 2024-09-07:
> Task Manager and Windows' built-in Services applet

It is also shows that is yamdcc service is runin

Sparronator9999, 2024-09-07:
Weird, I just tested with System Informer, and it actually notified me about two services being deleted (`WinRing0_1_2_0` (the internal name of WinRing0 that is used by YAMDCC), and `yamdccsvc.exe`).

porkmanager, 2024-09-07:
Oh sorry, thats issue on my end. I starting yamdcc from desktop context menu and seems it has not enough privileges for uninstall service. Just started yamdcc directly  from the app's folder ind its working properly

Sparronator9999, 2024-09-07:
I have added an Administrator check in the latest version. Something I never thought I'd need to do, since YAMDCC is configured to always run as Administrator (via an application manifest).

Sparronator9999, 2024-09-07:
Please try opening YAMDCC again via whatever means that was breaking service uninstall permissions (on the latest version, of course).

porkmanager, 2024-09-07:
https://github.com/user-attachments/assets/1131bd8d-dcb1-44ad-9b07-cd4359bb5a31

Latest build also the same. Starting it with context menu cant delete service but deleting temporary files and settings also can stop service

Sparronator9999, 2024-09-07:
I have made a couple of commits:
- Show an error dialog if the service failed to uninstall
- Attempt to stop the service first before uninstalling it (on my system, it doesn't matter whether I stop the service before uninstalling it or not, the service will uninstall and presumably be stopped first)

porkmanager, 2024-09-07:
> Can you test these changes?

Sure I will

porkmanager, 2024-09-07:
> I have made a couple of commits

Yea, now uninstall feature working properly from any type of launch program

Sparronator9999, 2024-09-07:
Well.

I guess I'm leaving the "stop before uninstall" functionality in, then.

That's that solved. Even if I don't like the solution.

closed 2024-09-07 by Sparronator9999