Sparronator9999/YAMDCC #9: Need to restart service at first run

porkmanager, 2024-09-07:
Noticed that while testing app
First run of the yamdcc dont working properly, need to restart the service for normal run

https://github.com/user-attachments/assets/e5e86530-703c-4c45-ae9c-832979ca29ed

Sparronator9999, 2024-09-07:
Fixed in latest commit.

porkmanager, 2024-09-07:
This one still in with the latest build for now. Ill show you how to reproduce it

Sparronator9999, 2024-09-07:
Are you sure you're using the latest build?

I tested all the options (saying Yes and No at both "service not installed" and "service not running" dialogs), and it worked for me.

porkmanager, 2024-09-07:
https://github.com/user-attachments/assets/a2bd9bda-00b3-4d04-829a-84cc837e15e8

> Are you sure you're using the latest build?

yep, ive downloaded the latest one

Sparronator9999, 2024-09-07:
Okay, I just did some testing, and the issue isn't caused by my latest changes (I think).

If you delete `CurrentConfig.xml` from the program data directory (`C:\ProgramData\Sparronator9999\YAMDCC` and re-run the config app, the same thing happens.

I have made another issue for this (#12), and will now close this one (as the bug in this issue is fixed).

closed 2024-09-07 by Sparronator9999