Sparronator9999/YAMDCC #15: Add support for 10th gen models and older

tangelogh, 2024-09-09:
### Prerequisites
- [x ] I am using the correct [issue template](https://github.com/Sparronator9999/MSIFanControl/issues/new/choose), i.e. I am requesting a feature be added to YAMDCC. Bug reports and questions should be opened using their respective templates.
- [ x] I have searched the [issue tracker](https://github.com/Sparronator9999/MSIFanControl/issues?q=label:"type:+enhancement") for duplicate or similar feature requests. Your issue will be closed if it is a duplicate of another issue.
- [ ] I have not answered truthfully to the above questions.

### Feature request
Good morning Sparronator9999, seems like you have done the heavy lifting on the EC side.
The 10th gen and older models that used Dragon Center have a different EC memory map. Would you be open to support those older models? I could help you with that, my only issue is on how to detect the correct mapping to use, but I get its easy to get the CPU model and infer it that way, worse case, having a setting to select the laptop gen would also work.

What do you think?

Sparronator9999, 2024-09-09:
TL;DR: Soon™.

I am already planning on doing this after config generation support is added (i.e. the ability to read and save your laptop's default settings to its own config).

Thanks to the open-source community, 10th-gen and older Intel support shouldn't be too hard, as the EC registers are already documented in other projects. All I need to do is copy them into a new config when config generation support is added.

PS:
I believe you have to *replace* the space between the check box brackets with an "x" for Github's check boxes to work properly.

Sparronator9999, 2024-09-09:
> Good morning Sparronator9999

Meanwhile in Australia...

closed 2024-09-09 by Sparronator9999