Sparronator9999/YAMDCC #17: Disabling perf. mode after sleep

porkmanager, 2024-09-26:
When chosen "max.battery" perf mode 10watt pl disables after windows sleep then awaking. 
On the fresh booted system its working properly and limits cpu power to 10w but after sleep it does not limit cpu power

Sparronator9999, 2024-09-27:
Hi, I can't seem to re-create this issue on my system.

Please do the following:

- Install [NoteBook FanControl](https://github.com/UraniumDonut/nbfc-revive/releases)
- Re-apply your settings, reboot your computer, do whatever you need to, to get 10W power limit working again.
- From an administrator Command Prompt, run `ec-probe read 210`, before putting computer to sleep and after, and post the output.

(side note: I plan on including the EC register read/dump functionality in the CLI when it finally releases)

porkmanager, 2024-09-27:
before sleep
[image]

After sleep
[image]

the registers are same before-after.
If you havent that issue probably that is on my side, something of the tweaked bios settings, latter i will try to reset bios settings and retry to repeat it. Before I reseted EC and it doesent help me with this issue

Sparronator9999, 2024-09-27:
Yeah, let's go with that.

You can uninstall NBFC now if you want, or keep it and play around with ec-probe :)

closed 2024-09-27 by Sparronator9999

porkmanager, 2024-10-06:
I did some research. What Ive understood: power limit keep holding normal after sleep in s0 sleep mode but I prefer to use classic s3 standby mode, after awaking with s3 power limit discards and you cant set it in yamdcc again untill reboot the system. Can you confirm it on your machine?
Everything else in yamdcc is works properly and does not resets after s3 sleep state

Sparronator9999, 2024-10-07:
I'm not sure why the power limit is the only thing that doesn't re-apply after sleep. Maybe it saves between sleep and awake, but not reboot?

I can't reproduce this on my machine, so I think it might be a model-specific issue.

Regardless, I plan on adding a "verify" feature to the YAMDCC service to make sure settings were correctly applied to the EC, but not for a few weeks.

porkmanager, 2024-10-11:
> Maybe it saves between sleep and awake, but not reboot?

Probably. Ive reseted the EC and reseted the bios to defaults the result is the same. Power limit for 10w is able till next sllep(s3), after that its available only after system reboot. Even recently I have updated my system to 11 (iot) enterprise ltsc 2024 and the behavior the same.

> since I find disabling the CPU turbo with ThrottleStop

May I ask u why do you use ThrottleStop for turboboost disabling?

Sparronator9999, 2024-10-11:
> May I ask u why do you use ThrottleStop for turboboost disabling?

Because it reduces my system's power usage during light tasks, meaning less heat (reducing the time my laptop's fan needs to turn on) and longer battery life (not that I use my laptop off the charger much).

For example, watching a 1080p YouTube video:

- CPU default settings (4.6 GHz): ~6-7 W
- CPU turbo disabled (2.3 GHz): ~3.5-4 W

I re-enable turbo when i need it, e.g. playing heavier games.

Testing done on an Intel Core i7-11800H.

<details>
<summary>Notes on testing (click to expand)</summary>

I left all settings on default, except for checking or clearing the "Disable Turbo" check box. All tests conducted using fresh ThrottleStop config after a laptop sleep cycle (to clear previous ThrottleStop settings).
</details>

Sparronator9999, 2024-10-11:
Oh, and on ThrottleStop, you can use it to enforce the 10W power limit (or any other arbitrary power limit up to your laptop's thermal capacity). The page you're looking for is accessed using the `TPL` button. Search the web for a tutorial.

porkmanager, 2024-10-14:
> I left all settings on default, except for checking or clearing the "Disable Turbo" check box. All tests conducted using fresh ThrottleStop config after a laptop sleep cycle (to clear previous ThrottleStop settings).

I though there are some another reasons for using ThrottleStop. As for me I prefer just to have 2 windows power plans, one for heavy tasks with enabled turboboost and fixed maxed freq. another one for some idle tasks(internet browsing, watching some videos) with disabled turboboost and freq. from 5 to 100%. 

> Oh, and on ThrottleStop, you can use it to enforce the 10W power limit (or any other arbitrary power limit up to your laptop's thermal capacity). The page you're looking for is accessed using the `TPL` button. Search the web for a tutorial.

Yea, I played with it many times when I searched best power limits for my cpu, then I just entered it in the bios and have no reasons for tweaking it more.