Sparronator9999/YAMDCC #19: Latest one build CTD

porkmanager, 2024-09-29:
Crashing cuz cant find CurrentConfig.xml
[crash-2024-09-29T13-47-24.txt](https://github.com/user-attachments/files/17179068/crash-2024-09-29T13-47-24.txt)

Sparronator9999, 2024-10-06:
Hi, I can't reproduce this bug on my end, but I have a possible fix, which will be uploaded shortly after this message is written.

Briefly, I changed the way that the service checks for `CurrentConfig.xml`'s existence, hopefully fixing the bug.

porkmanager, 2024-10-06:
Hey. I tried the latest build and previous bug still in on my side. Additional ive received a bug in the windows eventlog with this new one build instead a crashdump window in previous build

https://github.com/user-attachments/assets/65fa2b97-edb5-406b-bfc0-fb1cfb1a6f2c

porkmanager, 2024-10-06:
Last build which launches/runs without any troubles on my pc is https://github.com/Sparronator9999/YAMDCC/actions/runs/11087389635
all subsequent ctd with CurrentConfig.xml arguments (only with clean instal)

Sparronator9999, 2024-10-06:
Can you upload the files in `C:\ProgramData\Sparronator9999\YAMDCC`?

You may delete or move `LastConfig` (located in the above folder) before uploading to avoid leaking your PC's username if you want.

Sparronator9999, 2024-10-06:
Okay, this is kind of embarrassing.

Turns out I removed (or otherwise forgot to implement) the check in the config app for `CurrentConfig.xml`'s existence.

A new build will be published shortly.

porkmanager, 2024-10-07:
[image]
The bug still in with the latest build. ;p
[yamdccsvc.exe.log](https://github.com/user-attachments/files/17274285/yamdccsvc.exe.log)
Win. event log:
[yamdcc.zip](https://github.com/user-attachments/files/17274300/yamdcc.zip)

Sparronator9999, 2024-10-07:
Come on man, at least the config app isn't crashing anymore.

Anyways, I fixed the service crash in the latest commit.

porkmanager, 2024-10-07:
Great! Now its fixed.

Sparronator9999, 2024-10-07:
Nice.

closed 2024-10-07 by Sparronator9999