Sparronator9999/YAMDCC #20: Service failed to install, try running the yamdcc configurator again.

CConstantinee8, 2024-10-06:
Hi, when i run YAMDCC.exe i get error in title, i did check services and i only got winring0x64 but no yamdcc service, i used YAMDCC-main-a9ebc7a-Release, i tried windows installer troubleshooting but everything seems to be ok. do i need to do something before trying to run the exe? also in case it matters im using win 10 ltsc. thanks

Sparronator9999, 2024-10-06:
Hi, please provide the YAMDCC service log, and crash report if present.

They are located in `C:\ProgramData\Sparronator9999\YAMDCC`.

Sparronator9999, 2024-10-06:
> also in case it matters im using win 10 ltsc.

LTSC shouldn't matter, it's what I'm using as well.

CConstantinee8, 2024-10-06:
folder is empty

Sparronator9999, 2024-10-06:
Please go to the Event Viewer (search for it in the Start menu), then go to `Windows Logs` > `Application` (in the sidebar), then look for a group of three Error logs and copy and paste the text from each of them into a file.

You may want to remove your computer's username from the places it shows up in the event log. They are shown in the example output as a series of five asterisks (*****).

<details>
<summary>Example Event Log output (click to expand)</summary>

From "YAMDCC Service":
```
Service cannot be started. System.NullReferenceException: Object reference not set to an instance of an object.
   at YAMDCC.Service.svcFanControl.ApplyCurve() in C:\Users\*****\source\repos\YAMDCC\YAMDCC.Service\svcFanControl.cs:line 260
   at YAMDCC.Service.svcFanControl.OnStart(String[] args) in C:\Users\*****\source\repos\YAMDCC\YAMDCC.Service\svcFanControl.cs:line 119
   at System.ServiceProcess.ServiceBase.ServiceQueuedMainCallback(Object state)
```

From ".NET Runtime":
```
Application: yamdccsvc.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.NullReferenceException
   at YAMDCC.Service.svcFanControl.ApplyCurve()
   at YAMDCC.Service.svcFanControl.OnStart(System.String[])
   at System.ServiceProcess.ServiceBase.ServiceQueuedMainCallback(System.Object)
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.ServiceProcess.ServiceBase.Run(System.ServiceProcess.ServiceBase[])
   at YAMDCC.Service.Program.Main()
```

From "Application Error":
```
Faulting application name: yamdccsvc.exe, version: 0.0.0.0, time stamp: 0xb72a8b4f
Faulting module name: KERNELBASE.dll, version: 10.0.19041.4648, time stamp: 0x5a4af933
Exception code: 0xe0434352
Fault offset: 0x000000000003b699
Faulting process ID: 0x4c18
Faulting application start time: 0x01dade488017a267
Faulting application path: C:\Users\*****\source\repos\YAMDCC\YAMDCC.GUI\bin\Debug\net48\yamdccsvc.exe
Faulting module path: C:\Windows\System32\KERNELBASE.dll
Report ID: dd4e64af-e39b-423a-95f4-cb9d0cf3a28e
Faulting package full name: 
Faulting package-relative application ID: 
```
</details>

CConstantinee8, 2024-10-07:
no logs in event viewer, i thought i might have disabled logs when disabling telemetry in win, but i did try to install acrobat reader after and that one did show up in event viewer (failed install because no internet). i also tried installing in safe mod but same result, only logs i got are from revo installer.

Name    Operation    Old Name    
C:\Windows\Prefetch\DLLHOST.EXE-25B82684.pf    Overwritten        
C:\Windows\Prefetch\INSTALLUTIL.EXE-7A9F4A33.pf    Overwritten        
C:\Windows\Prefetch\CONHOST.EXE-E6AFC9F5.pf    Overwritten        
C:\Windows\Prefetch\YAMDCC.EXE-F82949FB.pf    Overwritten       

Key    Operation    Value    Data    Type    Old Data    
HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Services\bam\State\UserSettings\S-1-5-21-1518805375-4121542670-3783713298-1001\    Set    \Device\HarddiskVolume5\Windows\System32\dllhost.exe    3291cfa5918db10000000000002000    REG_BINARY    b043875a5818db10000000000002000    
HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Services\bam\State\UserSettings\S-1-5-21-1518805375-4121542670-3783713298-1001\    Set    \Device\HarddiskVolume6\Programy\Revo Uninstaller Pro\RevoAppBar.exe    a18e4085918db10000000000002000    REG_BINARY        
HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Services\bam\State\UserSettings\S-1-5-21-1518805375-4121542670-3783713298-1001\    Set    SequenceNumber    35    REG_DWORD    33    
HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Services\bam\State\UserSettings\S-1-5-21-1518805375-4121542670-3783713298-1001\    Set    \Device\HarddiskVolume5\Users\Co\Desktop\YAMDCC-main-a9ebc7a-Release\YAMDCC.exe    79a08cc5918db10000000000002000    REG_BINARY    8c43a595918db10000000000002000    
HKEY_USERS\S-1-5-21-1518805375-4121542670-3783713298-1001\SOFTWARE\Microsoft\Windows\CurrentVersion\Internet Settings\ZoneMap\    Set    ProxyBypass    1    REG_DWORD    1    
HKEY_USERS\S-1-5-21-1518805375-4121542670-3783713298-1001\SOFTWARE\Microsoft\Windows\CurrentVersion\Internet Settings\ZoneMap\    Set    IntranetName    1    REG_DWORD    1    
HKEY_USERS\S-1-5-21-1518805375-4121542670-3783713298-1001\SOFTWARE\Microsoft\Windows\CurrentVersion\Internet Settings\ZoneMap\    Set    UNCAsIntranet    1    REG_DWORD    1    
HKEY_USERS\S-1-5-21-1518805375-4121542670-3783713298-1001\SOFTWARE\Microsoft\Windows\CurrentVersion\Internet Settings\ZoneMap\    Set    AutoDetect    0    REG_DWORD    0

Sparronator9999, 2024-10-07:
Well, that didn't really help me much.

However, I think your bug is related to #19, which I (hopefully) fixed in the latest commit. Try that build, then report back.

CConstantinee8, 2024-10-07:
it works thank you

closed 2024-10-07 by Sparronator9999