Sparronator9999/YAMDCC #21: YACTD (Yet Another Crash To Desktop) :D

porkmanager, 2024-10-07:
[image]
What I done: Just started yamdcc, choose the profile, enabled ECmonitoring and forgot to close the window after applying what I wanted. It was run in background on my desktop for a few minutes, then I found it with this error.

[crash-2024-10-07T20-15-31.txt](https://github.com/user-attachments/files/17282960/crash-2024-10-07T20-15-31.txt)
[yamdccsvc.exe.log](https://github.com/user-attachments/files/17282961/yamdccsvc.exe.log)

Sparronator9999, 2024-10-08:
Congratulations, you've managed to crash the IPC component of YAMDCC.

Basically, the YAMDCC service stopped responding to the config app after a while, but the config app continued to pile on requests to the service (EC monitoring), until the queue reached max capacity and crashed.

I will look into this more later.

Oh, and nice issue title :)

Sparronator9999, 2024-10-08:
Hi, can you please update to the latest commit, then try to reproduce the above error?

I didn't fix the bug, but the output from the latest commit should help me figure out the problem (keyword: **should*)

porkmanager, 2024-10-08:
Hey, sure I will try the last one. Btw I tried to reproduce this bug on the yesterday's version again and i cant repeat it one more time

Sparronator9999, 2024-10-08:
I'm still not exactly sure what happened either. I think the service froze or crashed, and the config app isn't set up to handle service crashes after launch (yet). This is upcoming, but I've been too busy with both work and uni to do anything more than fixing my stupid mistakes/other smaller bugs.

Sparronator9999, 2024-10-08:
Okay, I've managed to reproduce the bug on my end: the service seems to stop taking requests from the config app after some time, then the config app crashes after the "queue" of service requests builds up too much (I didn't let it get to that stage).

Now I have to figure out why the service freezes after a long EC monitoring session.

Sparronator9999, 2024-10-08:
Yep, it's happening for me as well.

Closing and re-opening the config app temporarily fixes this issue for me.

Sparronator9999, 2024-10-08:
Worst of all, my crash logging "fixes" don't seem to be doing anything.

Sparronator9999, 2024-10-09:
Hi, I am going to add this issue to the list of known issues on the Issues Megathread for now.

Just... don't leave the EC monitor running for a long time for now?

I will fix this eventually, but probably not for a few weeks.

Sparronator9999, 2024-12-30:
> I will fix this eventually, but probably not for a few weeks.

Wow, that last comment aged like milk.

Anyways, I *think* I know why EC monitoring bugs out after a while.

Basically my theory is the IPC library starts to lag behind by one (or potentially more) command for some reason (which I need to figure out why), particularly after a lot of commands have been sent back and forth.

You can test this by enabling EC monitoring, wait for it to bug out, disable EC monitoring, then try enabling/disabling full blast. You'll notice the fan(s) turning on when full blast is "disabled" (in the config editor), or turning off/slowing down when full blast is "enabled".

Note that you may have to restart the YAMDCC config editor to fix the full blast setting (by checking and unchecking the full blast check box).

Sparronator9999, 2024-12-30:
Hi, could you please test the latest build of YAMDCC (available from GitHub Actions) and report your findings?

Sparronator9999, 2024-12-31:
Closing this issue since the bug seems to be fixed on my end.

Please re-open if the EC monitor is still broken on your end.

closed 2024-12-31 by Sparronator9999

porkmanager, 2025-01-02:
> Please re-open if the EC monitor is still broken on your end.

Seems i cant reproduce that issue anymore. Tested latest build for now

Sparronator9999, 2025-01-03:
> upd.
> was leaved running app with enabled EC monitoring for ca 10 minutues

Please update to the latest build from GitHub actions, and re-test.

I was discussing a similar issue with @tedomi2705, and I determined that the EC status timeout might be too short for some laptops.

Sparronator9999, 2025-01-03:
By the way, does your laptop have two fans?

Currently, all YAMDCC configs only report fan RPMs for one fan (including on two-fan laptops), but I can change this if I know which EC register the second fan RPM is stored in.

porkmanager, 2025-01-04:
> Please update to the latest build from GitHub actions, and re-test.

Okay I'll test that build.

> By the way, does your laptop have two fans?

mine gf66 12ug has 2 fans and both fan curves works properly for me

> Currently, all YAMDCC configs only report fan RPMs for one fan (including on two-fan laptops), but I can change this if I know which EC register the second fan RPM is stored in.

Seems yeah. EC monitoring shows only one fan speed now. iiric in older builds it was fine and i saw different  values per fan

https://github.com/user-attachments/assets/ccd3f56c-3b6f-415d-9fcf-4402dc8528ca

porkmanager, 2025-01-04:
> but I can change this if I know which EC register the second fan RPM is stored in.

Could this short test with the table be helpful for?
https://github.com/user-attachments/assets/17386708-9c16-45c9-ae97-74f918170532

Sparronator9999, 2025-01-04:
> Seems yeah. EC monitoring shows only one fan speed now

This has actually always been the case, since my laptop only has one fan (and only one fan RPM to report), so I used the same RPM register for both "virtual" fans.

> Could this short test with the table be helpful for?

If you post the output of `ec-probe monitor` while playing around with the fan speeds (change them one at a time), I should be able to tell you which one it is (and fix the config for you).

To start you off: the RPM register that YAMDCC reports (for >= 10th-gen Intel) is 0xC8, and the other one will probably be two registers away from this one (i.e. 0xC6 or 0xCA) and report a very similar value if both fans are set to the same speed.

porkmanager, 2025-01-04:
> Please update to the latest build from GitHub actions, and re-test.

I think you solved that issue. I leaved yamdcc runing in background with enabled ECmonitor for an hour and have had no issues.

> But if you post the output of `ec-probe monitor` while playing around with the fan speeds (change them one at a time), I should be able to tell you which one it is (and fix the config for you).

[image]

Sparronator9999, 2025-01-04:
> So this one is good for me. Now i see 2nd fan rpm correct.
> I set `<ReadReg>202</ReadReg>` and its works

Good job.

I will update the generic configs for >= 10th-gen laptops.

porkmanager, 2025-01-05:
> EDIT: which fan did you update (CPU Fan or GPU Fan)?

GPU fan

Sparronator9999, 2025-01-05:
> GPU fan

Cool, I guessed correctly then.

Since my last comment, I have updated all generic configs to have single- and dual-fan versions (the only difference between them is the fan RPM reporting).

kcos0320, 2025-02-04:
I keep getting this error every time I try to launch this program.

[image]

I have changed nothing to use this application.

Sparronator9999, 2025-02-04:
> I keep getting this error every time I try to launch this program.

Hi, could you please create a new issue, providing the YAMDCC service log (located at `C:\ProgramData\Sparronator9999\YAMDCC\Logs\yamdccsvc.log`)?