Sparronator9999/YAMDCC #22: Full blast hokey ?

grimy400, 2024-10-17:
### Prerequisites
- [ ] I am using the correct [issue template](https://github.com/Sparronator9999/MSIFanControl/issues/new/choose), i.e. I am requesting a feature be added to YAMDCC. Bug reports and questions should be opened using their respective templates.
- [ ] I have searched the [issue tracker](https://github.com/Sparronator9999/MSIFanControl/issues?q=label:"type:+enhancement") for duplicate or similar feature requests. Your issue will be closed if it is a duplicate of another issue.
- [ ] I have not answered truthfully to the above questions.

### Feature request
Is there a way to add a hotkey to enable and disable full blast would be very useful as I use my laptop closed with a monitor.

Sparronator9999, 2024-10-17:
Woah, someone using the issue templates? Impossible!

In all seriousness though, this should be possible.

I will add it to the upcoming feature list.

grimy400, 2024-10-17:
haha ! Sounds good thanks for the quick response this program actually saved my msi laptop lol I wasnt able to use cooler booster at all anymore

Sparronator9999, 2024-10-17:
Would you mind sharing your MSI laptop model so I can add it to the list of community-tested laptops? (e.g. my laptop is an MSI GF63 Thin 11SC)

grimy400, 2024-10-17:
Yes no problem ! I have the msi ge66 12ugs

Sparronator9999, 2025-02-23:
> is this going to also support changing the power profiles?

You can set the performance mode to be applied per-profile as of v1.0.2. I will see if I can add a way to change the default performance mode (applied when not overridden by a fan profile).

> on most msi laptops where the f7 key changes it with a small popup message at the bottom showing the profile(when using msi center that is)

On my laptop, the Fn+F7 key combo just opened MSI Center if I remember correctly (haven't had it installed since at least last year). I want to add support for this special key to Hotkey Handler, but I'm not sure if it works reliably on all MSI laptops (in fact, it's currently been removed from the current (pre-commit) source code until I find a way to reliably detect this key press).

You can test if it works properly on your laptop with the latest Hotkey Handler (you might need to compile it or scroll back a fair bit in GitHub Actions), it should open the Config Editor by default.

Sparronator9999, 2025-02-24:
Okay, time for a long one. Read if you dare. Skip to bottom for a keyboard test program to (hopefully) get the shortcut that the MSI Center key is pressing.

> I was just asking if it will be possible to use a keybind like this to switch between them quickly like how MSI Center does it.

I think I already mentioned above (or rather, said I'd think about it), but I *do* plan to allow switching these performance modes via hotkey combination (although I'm not sure if I will be able to get the MSI Center key working reliably enough to implement your described behaviour).

> Seems to not work unfortunately

I just committed an update which removed the MSI Center key function since it wasn't working for anyone but me, and I replaced (or rather, will replace) the global keyboard hook (see above documentation links) with calls to [`RegisterHotKey`](https://learn.microsoft.com/en-us/windows/win32/api/winuser/nf-winuser-registerhotkey) and [`UnregisterHotKey`](https://learn.microsoft.com/en-us/windows/win32/api/winuser/nf-winuser-unregisterhotkey).

> However I do know that using that function key actually triggers one of the invisible F13 to F24 [...] keys to switch the profiles, which is what MSI Center binds it to

Why, I do know about those keys (last I heard of them, the new "copilot" key on some of Microsoft's new laptops presses some convoluted shortcut that ends in F23).

However, when I use a small keyboard test app I wrote to test the MSI Center key (not currently publicly available), I get a virtual key code of 255, scan code of 10, and the "extended key" flag set after I press it at least twice, otherwise nothing happens (see documentation for [`LowLevelKeyboardProc`](https://learn.microsoft.com/en-us/windows/win32/winmsg/lowlevelkeyboardproc) and the struct that gets passed to it, [`KBDLLHOOKSTRUCT`](https://learn.microsoft.com/en-us/windows/win32/api/winuser/ns-winuser-kbdllhookstruct), for more info on these parameters). The webcam key is similar, but produces a scan code of 110. However this key already works as expected, so no work needed there :)

Interestingly enough, testing the touchpad disable button (F3 on my keyboard) produces the following shortcut: `Ctrl`+`Win`+`F24`, and same with F2 (which maps to `Win`+`P`, or the "Project" popup). None of the other keys exhibit this behaviour.

Anyways, I have attached the keyboard tester mentioned above [here](https://github.com/user-attachments/files/18949531/KeyboardTest.zip). If you don't mind, could you please test the MSI center key (and others, if you want) and report back on their behaviour? You can leave MSI Center running if you want, it might produce more interesting output.

porkmanager, 2025-02-25:
> I have attached the keyboard tester mentioned above [here](https://github.com/user-attachments/files/18949531/KeyboardTest.zip). If you don't mind, could you please test the MSI center key (and others, if you want) and report back on their behaviour?

Mine are the same as [ios7jbpro](https://github.com/ios7jbpro) has shown before.
I remember I was able to remap those combinations with the https://github.com/randyrants/sharpkeys utility when I needed to remap some keybindings. It shows different values. It probably would be useful in your keybinding workaround.
For example this code sharpkeys parse when I push fn+f7(msi key):

[image]

That key can be remapped with sharpkey and it wroks.

Sparronator9999, 2025-02-25:
Here's what my MSI Center key (same Fn+F7 combo) maps to:

[image]

It seems like it uses a different key for every laptop...

@ios7jbpro, could you post what key bind appears when you use SharpKeys? Or is it the same as mine or @porkmanager's?

Sparronator9999, 2025-03-11:
@grimy400, I have just released the [first v1.1 beta](https://github.com/Sparronator9999/YAMDCC/releases/tag/v1.1.0-beta.1), which contains the new Hotkey Handler. Can you please test it and report back?

grimy400, 2025-03-12:
> [@grimy400](https://github.com/grimy400), I have just released the [first v1.1 beta](https://github.com/Sparronator9999/YAMDCC/releases/tag/v1.1.0-beta.1), which contains the new Hotkey Handler. Can you please test it and report back?

Thank you so much !! I just tested it and It is working pretty well on my end 👍 I have noticed that you need to have the handler minimized for the hotkeys to work and also when I click apply  twice it gives me an error and stop working until I remove the hotkey and add it again.

Sparronator9999, 2025-03-12:
>  I have noticed that you need to have the handler minimized for the hotkeys to work

This is intended behaviour. If I didn't disable the hotkeys when the config window is shown, you'd be potentially triggering the hotkeys while trying to rebind them :)

> when I click apply twice it gives me an error and stop working until I remove the hotkey and add it again

This, on the other hand, is very much *not* intended behaviour. I will fix this Later™.

grimy400, 2025-03-12:
> This is intended behaviour. If I didn't disable the hotkeys when the config window is shown, you'd be potentially triggering the hotkeys while trying to rebind them :)
>
>
> This, on the other hand, is [...]

Make sense just thought I would mention it I will keep testing the beta and report back if something else comes up ! But I am thrilled I can finally use fullblast on my external keyboard :)

porkmanager, 2025-03-12:
2 times pressed "apply" button causes ctd

https://github.com/user-attachments/assets/00ef4984-8325-476e-aca6-96df88d34e31

Подробная информация об использовании оперативной 
(JIT) отладки вместо данного диалогового 
окна содержится в конце этого сообщения.

************** Текст исключения **************
System.InvalidOperationException: Коллекция была изменена; невозможно выполнить операцию перечисления.
   в System.ThrowHelper.ThrowInvalidOperationException(ExceptionResource resource)
   в System.Collections.Generic.Dictionary`2.KeyCollection.Enumerator.MoveNext()
   в YAMDCC.HotkeyHandler.MainForm.UnregisterHotkeys() в D:\a\YAMDCC\YAMDCC\YAMDCC.HotkeyHandler\MainForm.cs:строка 562
   в YAMDCC.HotkeyHandler.MainForm.RegisterHotkeys() в D:\a\YAMDCC\YAMDCC\YAMDCC.HotkeyHandler\MainForm.cs:строка 548
   в System.Windows.Forms.Control.OnClick(EventArgs e)
   в System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   в System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   в System.Windows.Forms.Control.WndProc(Message& m)
   в System.Windows.Forms.ButtonBase.WndProc(Message& m)
   в System.Windows.Forms.Button.WndProc(Message& m)
   в System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Загруженные сборки **************
mscorlib
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
HotkeyHandler
    Версия сборки: 1.1.0.0
    Версия Win32: 1.1.0.0
    CodeBase: file:///C:/Program%20Files/YAMDCC/HotkeyHandler.exe
----------------------------------------
System.Windows.Forms
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9220.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
YAMDCC.IPC
    Версия сборки: 2.2.0.0
    Версия Win32: 2.2.0.0
    CodeBase: file:///C:/Program%20Files/YAMDCC/YAMDCC.IPC.DLL
----------------------------------------
YAMDCC.Common
    Версия сборки: 1.1.0.0
    Версия Win32: 1.1.0.0
    CodeBase: file:///C:/Program%20Files/YAMDCC/YAMDCC.Common.DLL
----------------------------------------
System.Configuration
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9220.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Версия сборки: 4.0.0.0
[... 45 lines cut ...]
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Windows.Forms.resources
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_ru_b77a5c561934e089/System.Windows.Forms.resources.dll
----------------------------------------

************** Оперативная отладка (JIT) **************
Для подключения оперативной (JIT) отладки файл .config данного
приложения или компьютера (machine.config) должен иметь
значение jitDebugging, установленное в секции system.windows.forms.
Приложение также должно быть скомпилировано с включенной
отладкой.

Например:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

При включенной отладке JIT любое необрабатываемое исключение
пересылается отладчику JIT, зарегистрированному на данном компьютере,
вместо того чтобы обрабатываться данным диалоговым окном.

Sparronator9999, 2025-03-12:
@porkmanager @grimy400 I have fixed the double-click Apply to crash bug (along with a related bug) in e84a861 (not released as a beta yet).

I want to make some other changes before releasing Beta 2. You can test the latest changes by using GitHub Actions.

grimy400, 2025-03-12:
> [@porkmanager](https://github.com/porkmanager) [@grimy400](https://github.com/grimy400) I have fixed the double-click Apply to crash bug (along with a related bug) in [...]

Tested it with the github Actions and its working perfectly now.

Sparronator9999, 2025-03-14:
@porkmanager @grimy400 I have released the fixes from e84a861 in v1.1 Beta 2.