Sparronator9999/YAMDCC #25: Crush on apply fan settings

injitools, 2024-12-04:
### Prerequisites
- [x ] I am using the correct [issue template](https://github.com/Sparronator9999/MSIFanControl/issues/new/choose), i.e. I am reporting a bug or crash in YAMDCC. Feature requests and questions should be opened using their respective templates.
- [x ] I have searched the [issue tracker](https://github.com/Sparronator9999/MSIFanControl/issues?q=label:"type:+bug") for duplicate or similar bug reports. Your issue will be closed if it is a duplicate of another issue.
- [ ] I have not answered truthfully to the above questions.

### Describe the bug
When i apply settings for cpu fan i get bug window with this

(called from Application.ThreadException)
System.ArgumentOutOfRangeException: Значение '-128' недопустимо для 'Value'. 'Value' должно лежать в диапазоне от 'Minimum' до 'Maximum'.
Имя параметра: Value
   в System.Windows.Forms.Control.MarshaledInvoke(Control caller, Delegate method, Object[] args, Boolean synchronous)
   в System.Windows.Forms.Control.Invoke(Delegate method, Object[] args)
   в YAMDCC.GUI.MainWindow.IPC_MessageReceived(Object sender, PipeMessageEventArgs`2 e)
   в YAMDCC.IPC.NamedPipeConnection`2.ReadPipe()
   в YAMDCC.IPC.Threading.Worker.DoWorkImpl(Object oAction)

### To Reproduce
Steps to reproduce the behavior:
1. install YAMDCC [this build](https://github.com/Sparronator9999/YAMDCC/actions/runs/12147119661) to MSI Vector 17 HX A14VHG
2. Open program
3. Load config MSI-10th-gen-or-newer.xml
4. Click apply
5. See error

### Expected behavior
No errors hapens

### Additional context
Add any other context (including screenshots) relating to the problem here.

Sparronator9999, 2024-12-04:
Hmm, looks like the keyboard backlight feature isn't working on your laptop and is crashing YAMDCC.

I will investigate this and report back soon.

Sparronator9999, 2024-12-04:
Actually, can you please download and install [NoteBook FanControl](https://github.com/UraniumDonut/nbfc-revive), then in an Adminstrator command prompt:

`ec-probe dump`

then copy and paste the output here?

injitools, 2024-12-04:
> Actually, can you please download and install [NoteBook FanControl](https://github.com/UraniumDonut/nbfc-revive), then in an Adminstrator command prompt:
> 
> `ec-probe dump`
> 
> then copy and paste the output here?

[image]

Sparronator9999, 2024-12-04:
Ah, I think I know what's going on now.

Try [disabling VBS (Core virtualization)](https://www.thewindowsclub.com/disable-vbs-windows), then run `ec-probe dump` again.

injitools, 2024-12-04:
> Ah, I think I know what's going on now.
> 
> Try [disabling VBS (Core isolaton)](https://www.thewindowsclub.com/disable-vbs-windows), then run `ec-probe dump` again.

after disable virtualization

```
   | 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
---|------------------------------------------------
00 | 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00
10 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20 | 00 00 00 00 00 00 00 00 0A 05 00 00 00 00 19 59
30 | 03 01 00 0D 01 00 50 81 2B 16 28 3C 37 02 C0 00
40 | 2B 16 3B 00 1F 15 00 00 5A 0C 9A 3D C7 0B C0 44
50 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
60 | 00 00 00 00 00 00 00 00 35 00 32 39 40 47 4E 55
70 | 64 0F 05 0F 12 19 1E 28 64 00 03 03 03 03 03 03
80 | 00 00 32 39 40 47 4E 55 64 05 05 0F 12 19 1E 28
90 | 64 5F 03 03 03 03 03 03 02 00 00 02 00 00 00 00
A0 | 31 37 53 31 49 4D 53 32 2E 31 30 37 30 32 30 34
B0 | 32 30 32 34 31 36 3A 30 32 3A 32 31 00 00 00 08
C0 | 00 00 06 31 00 00 00 00 02 15 01 6B 00 00 00 00
D0 | 00 00 C4 00 4D 00 04 BC 00 00 00 00 00 04 00 00
E0 | E2 00 00 1F 15 00 00 40 00 00 00 00 00 D1 00 00
F0 | 00 00 70 00 2C 7F 05 00 44 3A 00 00 00 00 00 00
```

Sparronator9999, 2024-12-04:
Looking at your EC dump, it looks like your laptop doesn't support keyboard backlight adjustment with YAMDCC.

Disabling VBS will allow YAMDCC to set your fan curves now (if you didn't, you would've found that your fan settings won't be applied).

I made a few changes to YAMDCC while you were sending `ec-probe` dumps, and it should no longer crash due to bad keyboard backlight values (an error will instead be displayed).

I also uploaded a modified version of `MSI-10th-gen-or-newer.xml`, `MSI-10th-gen-or-newer-nokeylight.xml` which should work fine for your laptop (except for the keyboard backlight settings, of course).

TL;DR: keyboard backlight adjustment won't work with your laptop (using YAMDCC). VBS would've prevented changing fan settings if left on. There's a new config made for laptops like yours with full RGB keyboards, but YAMDCC won't crash anymore if you use `MSI-10th-gen-or-newer.xml`.

injitools, 2024-12-04:
> Disabling VBS will allow YAMDCC to set your fan curves now (if you didn't, you would've found that your fan settings won't be applied).
> 

settings for cpu and gpu fans working with vbs. i sucessfuly adjust settings with this error - just clossing crash window

Sparronator9999, 2024-12-04:
> settings for cpu and gpu fans working with vbs

Are you sure that's with VBS *enabled*? From your `ec-probe` output before (the first one), changing the fan settings should've not worked, unless VBS only prevents *reading* the EC, but not *writes* to it (which seems to defeat the purpose of VBS).

injitools, 2024-12-04:
> > settings for cpu and gpu fans working with vbs
> 
> Are you sure that's with VBS _enabled_? From your `ec-probe` output before (the first one), changing the fan settings should've not worked, unless VBS only prevents _reading_ the EC, but not _writes_ to it (which seems to defeat the purpose of VBS).

i disable vbs only after send you zero dump. 
i make new try with enabled vbs and then post results

injitools, 2024-12-04:
> > settings for cpu and gpu fans working with vbs
> 
> Are you sure that's with VBS _enabled_? From your `ec-probe` output before (the first one), changing the fan settings should've not worked, unless VBS only prevents _reading_ the EC, but not _writes_ to it (which seems to defeat the purpose of VBS).

try with this enabled components 
[image]

all works, YAMDCC and WSL2. 
`ec-probe dump` also works.
core isolation disabled now and before. i disable it after installing windows.
so i don't know why `ec-probe dump` first time make zero dump, and make right dump after disabling Virtual Machine Platform.

but hey, now all working without errors, thank you =)

closed 2024-12-04 by injitools

Sparronator9999, 2024-12-05:
Now that I think of it, `ec-probe dump` occasionally gives me a zero dump as well.

> but hey, now all working without errors, thank you =)

You're welcome!