Sparronator9999/YAMDCC #26: bit weird UI on high resolution

porkmanager, 2024-12-10:
[image]
Noticed that on external display. 4k res. screen scaling 300%. 
On the native FHD display everything working properly without any ui distortion

Sparronator9999, 2024-12-11:
WinForms is known to have weird scaling issues at high DPI.

I only test YAMDCC at 100% (96 DPI) and 125% (120 DPI) scaling, since 125% is the default setting of my display, and I take screenshots at 100% so they don't appear too big.

Sparronator9999, 2024-12-11:
Unfortunately I don't have a 4K display to test on, so this issue is likely going to be very hard to work around, short of moving to a different UI library (i.e. WPF)

porkmanager, 2024-12-11:
Alright I understood that. Isnt a big issue just aesthetic. Didnt you thought about another gui library? wfp is more flexible and modern then winforms.
edit.
Already found an answer :)

> I have little to no experience with other UI kits (e.g. WPF).

Sparronator9999, 2024-12-11:
Thank my high school programming class for only teaching WinForms (to be fair, it is easy to learn...)

I might try to fix the scaling issues, but maybe it's time to try out WPF again...

closed 2024-12-11 by Sparronator9999