Sparronator9999/YAMDCC #33: Switch 2 fan profiles

porkmanager, 2025-01-05:
Would be useful to have ability to switch both fans profiles(cpufan+gpufan) at one time.
For example i have 2 profiles "gaming" and "idle". For complete enabling one of those fan profiles i should set at first CPU fan profile then switch to the GPU fan profile and set it to needed profile.

https://github.com/user-attachments/assets/2cc731c6-ee03-419c-b6c0-31ac1fd455ac

Sparronator9999, 2025-01-05:
I will begin working on it.

Sparronator9999, 2025-01-06:
Implemented in b2da722.

closed 2025-01-06 by Sparronator9999

porkmanager, 2025-01-06:
Thx 4 your work bro! You are the best!

Sparronator9999, 2025-01-06:
No problem!

I will release another beta soon if you have no more bugs/feature requests for me?

porkmanager, 2025-01-06:
> bugs

Catched one bug with this new feature.
While "Switch all fans" option enabled i get CTD on creation or deletion profiles.

System.IndexOutOfRangeException: Индекс находился вне границ массива.
   в YAMDCC.ConfigEditor.MainWindow.cboProfSel_IndexChanged(Object sender, EventArgs e)
   в System.Windows.Forms.ComboBox.OnSelectedIndexChanged(EventArgs e)
   в System.Windows.Forms.ComboBox.set_SelectedIndex(Int32 value)
   в YAMDCC.ConfigEditor.MainWindow.AddFanProfile()
   в System.Windows.Forms.Control.OnClick(EventArgs e)
   в System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   в System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   в System.Windows.Forms.Control.WndProc(Message& m)
   в System.Windows.Forms.ButtonBase.WndProc(Message& m)
   в System.Windows.Forms.Button.WndProc(Message& m)
   в System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Sparronator9999, 2025-01-06:
Huh, that's weird.

When I tested this before, it worked fine. Back to the IDE I go...

Sparronator9999, 2025-01-06:
> When I tested this before, it worked fine

I might have forgotten to enable the "switch all fan profiles" check box during testing...

Anyways, the crash should now be fixed (along with another bug I caught during *actual* testing).

Can you test the latest build from GitHub Actions?

porkmanager, 2025-01-06:
> Can you test the latest build from GitHub Actions?

Yep. Cam confirm. Now it fixed