Sparronator9999/YAMDCC #35: Unavailable to edit single configs

porkmanager, 2025-01-07:
Before latest update i was able to edit single configs in the config editor but now it isnt avalible, the curves are grayed out like on "Default" config.

[image]

The config
[IDLE.zip](https://github.com/user-attachments/files/18336401/IDLE.zip)

Sparronator9999, 2025-01-07:
Hi, that latest update changed how YAMDCC checks for the "default" fan profile. Originally I just looked for it by name, but now I just use the first profile and assume it's the default for the computer (and make it non-editable.

To fix:
1. Make a copy of the "Idle" config (by clicking the "+" button or `Options` > `Add profile`). You should now be able to edit the copy.
2. Run the EC-to-config feature. This will get your default fan curve and copy it to the first fan profile, **regardless of its name** ("Default" or otherwise). This means the "Idle" profile we just made a copy of will get overwritten!
3. Rename the "Copy of Idle" config to "Idle" and fix the description (it would've had `(Copy of Idle)` added to the start)

...or I can just fix my code. Yep, that seems like the right thing to do.

Fix incoming.

Sparronator9999, 2025-01-07:
Okay, the default config detection behaviour should be fixed now.

Can you please test the EC-to-config changes and make sure it doesn't crash/overwrite your custom config?

porkmanager, 2025-01-07:
Hey

Yep. [Latest build](https://github.com/Sparronator9999/YAMDCC/actions/runs/12658990232) works properly, ability to edit single profiles has been restored. Thank you!

Sparronator9999, 2025-01-07:
No problem!

closed 2025-01-07 by Sparronator9999