Sparronator9999/YAMDCC #37: Question about config PerfModeConf

porkmanager, 2025-01-08:
Hey!
Is it possible to make fan profiles with inclusion PerfModeConf?
For example i have multiply fan configs for different scenarios and performance mode is one for all of them.
I would to have a silent fan profile and when it enabled performance mode switching to "max. battery"(<ModeSel>0</ModeSel>).

It would be useful to make some profiles for battery using and it can prevent overheating. for example I enabled silent fan profile(fans are stopped) and frogot it, then launch some heavy task like a game i will get overheating. But if i could get that config with the max battery performance mode it will limit cpu tdp to 10w and overheating will be literally impossible.

tedomi2705, 2025-01-08:
On MSI Modern 15, there's a register at 0xf4 that can be 0x1d for silent fan mode (RegConf)

Sparronator9999, 2025-01-08:
Hi @tedomi2705, I've made a new issue for your comment. I will continue discussion over there.

porkmanager, 2025-01-09:
> If I'm reading correctly, you want to be able to switch fan profiles automatically when you switch performance modes?

No. I mean that fan profile can contain `<ModeSel>0</ModeSel>`. And  when i switching to my "Silent" fan profile then Power profile sets to "Maximum battery life". Can that be don via config?

Sparronator9999, 2025-01-09:
> No. I mean that fan profile can contain `<ModeSel>0</ModeSel>`.

Oh, I see. That makes more sense, since I can just have a toggle to make the performance mode per-config or per-fan-profile.

This will still be a v1.1 feature.

porkmanager, 2025-01-09:
btw short bug report. im testing new build with an updater. after uninstalling the task "YAMDCC auto-update task" still in windows scheduler.

Sparronator9999, 2025-01-09:
> after uninstalling yamdcc, the task "YAMDCC auto-update task" still in windows scheduler.

Ah, I knew I forgot something.

Fixed in latest commit.

Sparronator9999, 2025-02-15:
@porkmanager I have implemented your feature request in the latest commit. Would you like to test it for me?

closed 2025-02-15 by Sparronator9999

porkmanager, 2025-02-15:
> I have implemented your feature request in the latest commit. Would you like to test it for me?

Woah! TYVM mate! I'd be happy to test it!

commit 0d48cc6c (2025-02-15): Add per-profile performance mode toggle

- Remove support for inequal number of fan profiles per fan config

Closes #37