Sparronator9999/YAMDCC #38: Add RegConf for MSI Modern 15 for silent fan mode

Sparronator9999, 2025-01-08:
On MSI Modern 15, there's a register at 0xf4 that can be 0x1d for silent fan mode (RegConf)

_Originally posted by @tedomi2705 in https://github.com/Sparronator9999/YAMDCC/issues/37#issuecomment-2578277141_

Sparronator9999, 2025-01-08:
I can create a new RegConf for you, but I would like to get the next beta out first.

You'll have to wait for v1.1 if you want this to be applied as part of a PerfModeConf.

Sparronator9999, 2025-01-11:
@tedomi2705, I just added a new setting for the fan mode (with Auto, Silent, Basic and Advanced (default) options). Can you please test it and report back?

tedomi2705, 2025-01-12:
Seem everything is working fine
This model doesn't have basic fan mode

And I found out that the silent fan mode in my model isnt actually limit fan speed but it limit cpu tdp to 15w

closed 2025-01-12 by Sparronator9999