Sparronator9999/YAMDCC #39: unexpected behavior. ctd

porkmanager, 2025-01-14:
https://github.com/user-attachments/assets/b9fa3c82-d922-41a0-843b-0d35de544cb0

System.NullReferenceException: Ссылка на объект не указывает на экземпляр объекта.
   в YAMDCC.ConfigEditor.MainWindow.tsiAdvanced_Click(Object sender, EventArgs e)
   в System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
   в System.Windows.Forms.ToolStripMenuItem.OnClick(EventArgs e)
   в System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
   в System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
   в System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
   в System.Windows.Forms.ToolStripDropDown.OnMouseUp(MouseEventArgs mea)
   в System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   в System.Windows.Forms.Control.WndProc(Message& m)
   в System.Windows.Forms.ToolStrip.WndProc(Message& m)
   в System.Windows.Forms.ToolStripDropDown.WndProc(Message& m)
   в System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

Sparronator9999, 2025-01-14:
Fix incoming.

Sparronator9999, 2025-01-14:
You wouldn't believe it, but the check I put in place to prevent this crash from happening in the first place didn't work.

Fixed in commit 1052cf3.

closed 2025-01-14 by Sparronator9999