Sparronator9999/YAMDCC #40: missing tooltips

porkmanager, 2025-01-14:
marked the places without tooltips. everything else seems has balloon tooltips
[image]
[image]
[image]
[image]

Sparronator9999, 2025-01-14:
Yep, I'm aware I haven't added tooltips to everything yet.

Thanks for pointing out which ones need to be done, though.

Sparronator9999, 2025-01-14:
The fan mode should have a tooltip already.

If you're talking about the labels to the left of the actual checkboxes/dropdowns/slider, none of them have tooltips (I only added them to the actual checkbox/dropdown/slider for each option)

Sparronator9999, 2025-01-14:
The remaining tooltips have been added in the latest commit.

closed 2025-01-14 by Sparronator9999