Sparronator9999/YAMDCC #44: updater not working

porkmanager, 2025-01-17:
I tired to test an updater. Downloaded a version below(beta8 non hotfix) and it cant update yamdcc properly. Update process error and updater.exe just hangin.

https://github.com/user-attachments/assets/a4e18620-91f4-4dc6-a0db-eb33ae0301df

yamdcc logfile: [yamdccsvc.exe.zip](https://github.com/user-attachments/files/18451350/yamdccsvc.exe.zip)

.net error log:
`Подробная информация об использовании оперативной 
(JIT) отладки вместо данного диалогового 
окна содержится в конце этого сообщения.

************** Текст исключения **************
System.IO.DirectoryNotFoundException: Не удалось найти часть пути "C:\Program".
   в YAMDCC.Common.Dialogs.ProgressDialog.Worker_RunWorkerCompleted(Object sender, RunWorkerCompletedEventArgs e)
   в System.ComponentModel.BackgroundWorker.OnRunWorkerCompleted(RunWorkerCompletedEventArgs e)

************** Загруженные сборки **************
mscorlib
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
Updater
    Версия сборки: 1.0.0.0
    Версия Win32: 1.0.0.0
    CodeBase: file:///C:/Program%20Files/YAMDCC/Old/Updater.exe
----------------------------------------
System.Windows.Forms
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9220.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
YAMDCC.Common
    Версия сборки: 1.0.0.0
    Версия Win32: 1.0.0.0
    CodeBase: file:///C:/Program%20Files/YAMDCC/Old/YAMDCC.Common.DLL
----------------------------------------
System.Configuration
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9220.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.ServiceProcess
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.ServiceProcess/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.ServiceProcess.dll
----------------------------------------
mscorlib.resources
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_ru_b77a5c561934e089/mscorlib.resources.dll
----------------------------------------
System.Windows.Forms.resources
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_ru_b77a5c561934e089/System.Windows.Forms.resources.dll
----------------------------------------

************** Оперативная отладка (JIT) **************
Для подключения оперативной (JIT) отладки файл .config данного
приложения или компьютера (machine.config) должен иметь
значение jitDebugging, установленное в секции system.windows.forms.
Приложение также должно быть скомпилировано с включенной
отладкой.

Например:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

При включенной отладке JIT любое необрабатываемое исключение
пересылается отладчику JIT, зарегистрированному на данном компьютере,
вместо того чтобы обрабатываться данным диалоговым окном.

`

porkmanager, 2025-01-17:
Havent seen that is ver 8beta has bugged updater.

Sparronator9999, 2025-01-17:
That crash should not be happening.

Also, that crash dialog you see is the .NET default crash dialog, not the "YAMDCC has crashed!" one you're probably used to seeing by now.

If you click the left button (should say "Details" or something similar), you'll get a stack trace similar to YAMDCC's crash report that I can use to figure out where the bug is. Please copy and paste it into a comment (or screenshot/record it, I don't really care...)

> Havent seen that is ver 8beta has bugged updater.

More like every version since the updater was introduced is buggy in some way...

porkmanager, 2025-01-17:
> you'll get a stack trace similar to YAMDCC's crash report that I can use to figure out where the bug is. Please copy and paste it into a comment (or screenshot/record it, I don't really care...)

I sent it before I edited my post xD.

Here is it:

`Подробная информация об использовании оперативной 
(JIT) отладки вместо данного диалогового 
окна содержится в конце этого сообщения.

************** Текст исключения **************
System.IO.DirectoryNotFoundException: Не удалось найти часть пути "C:\Program".
   в YAMDCC.Common.Dialogs.ProgressDialog.Worker_RunWorkerCompleted(Object sender, RunWorkerCompletedEventArgs e)
   в System.ComponentModel.BackgroundWorker.OnRunWorkerCompleted(RunWorkerCompletedEventArgs e)

************** Загруженные сборки **************
mscorlib
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
Updater
    Версия сборки: 1.0.0.0
    Версия Win32: 1.0.0.0
    CodeBase: file:///C:/Program%20Files/YAMDCC/Old/Updater.exe
----------------------------------------
System.Windows.Forms
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9181.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9220.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
YAMDCC.Common
    Версия сборки: 1.0.0.0
    Версия Win32: 1.0.0.0
    CodeBase: file:///C:/Program%20Files/YAMDCC/Old/YAMDCC.Common.DLL
----------------------------------------
System.Configuration
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9220.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.ServiceProcess
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.ServiceProcess/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.ServiceProcess.dll
----------------------------------------
mscorlib.resources
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_ru_b77a5c561934e089/mscorlib.resources.dll
----------------------------------------
System.Windows.Forms.resources
    Версия сборки: 4.0.0.0
    Версия Win32: 4.8.9032.0 built by: NET481REL1
    CodeBase: file:///C:/windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_ru_b77a5c561934e089/System.Windows.Forms.resources.dll
----------------------------------------

************** Оперативная отладка (JIT) **************
Для подключения оперативной (JIT) отладки файл .config данного
приложения или компьютера (machine.config) должен иметь
значение jitDebugging, установленное в секции system.windows.forms.
Приложение также должно быть скомпилировано с включенной
отладкой.

Например:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

При включенной отладке JIT любое необрабатываемое исключение
пересылается отладчику JIT, зарегистрированному на данном компьютере,
вместо того чтобы обрабатываться данным диалоговым окном.

`

Sparronator9999, 2025-01-17:
It looks like the updater can't handle spaces in the path names for some reason.

I'm more surprised it didn't crash because you installed it to Program Files (which usually needs UAC permissions to modify anything inside).

Sparronator9999, 2025-01-17:
Finally remembered I can write "Fixes <issue>" in the commit message to close an issue...

closed 2025-01-17 by Sparronator9999

Sparronator9999, 2025-01-17:
Oh by the way, you can test the newer updater by checking for updates from the "Check for updates" button when running `Updater.exe` directly, then clicking the "force-reinstall" link in the "up-to-date" or "dev version detected" message.

Thanks for your dedication to bug hunting in this project.

porkmanager, 2025-01-17:
> I'm more surprised it didn't crash because you installed it to Program Files (which usually needs UAC permissions to modify anything inside).

I do not using uac and i have only one administrator account in windows

commit 5e45ff61 (2025-01-17): Fix updater handling of file paths with spaces

Fixes #44 (hopefully)