Sparronator9999/YAMDCC #46: updater still buggy xD

porkmanager, 2025-01-17:
I tried to do force update in the latest dev build. Got ctd

Version: 1.0.0-beta.8-hotfix.1
Revision (git): 7455b41a2171fb469bd43f8c2be890b09d2ba956
Crash report:
System.UnauthorizedAccessException: Отказано в доступе по пути "ConfigEditor.exe".
   в System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   в System.IO.FileInfo.Delete()
   в YAMDCC.Updater.Program.<>c__DisplayClass1_0.<InstallUpdate>b__0()
   в System.Threading.Tasks.Task`1.InnerInvoke()
   в System.Threading.Tasks.Task.Execute()
--- Конец трассировка стека из предыдущего расположения, где возникло исключение ---
   в System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   в System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   в YAMDCC.Common.Dialogs.ProgressDialog`1.<OnLoad>d__9.MoveNext()
--- Конец трассировка стека из предыдущего расположения, где возникло исключение ---
   в System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()

Sparronator9999, 2025-01-17:
Now THAT'S the crash I was half expecting.

Did this happen during the install phase, or while downloading/extracting the update?

If it happened during the install phase (where we should be running with UAC permissions), then there's a problem.

porkmanager, 2025-01-17:
> Did this happen during the install phase, or while downloading/extracting the update?

Seems that was happen during installing phase. Let me check it again. I'll record short video

porkmanager, 2025-01-17:
yep. while installin
https://github.com/user-attachments/assets/dc7f573f-93db-4a8f-8409-6d9863846092

Sparronator9999, 2025-01-17:
You left the config editor open, and so the updater can't delete the old version.

Time to implement process killing as part of the update process...

porkmanager, 2025-01-17:
Yea. Its working without running config editor. I thought updater is able to stop it, cuz it is possible to update it right in the config editor xD

Sparronator9999, 2025-01-17:
> I thought updater is able to stop it, cuz it is possible to update it right in the config editor xD

Um... forgor

commit 14966ecd (2025-01-17): Kill all running instances of YAMDCC before installing an update

Fixes #46
closed 2025-01-17 by Sparronator9999