Sparronator9999/YAMDCC #47: one more updater issue

porkmanager, 2025-01-17:
Updating process deletes current Configs folder. Every custom user config wipes permanently.

Sparronator9999, 2025-01-17:
It should be moved to the `Old` folder, which doesn't get deleted after an update.

I should probably not move the Configs folder during an update, but then any configs that exist in the updated YAMDCC will overwrite the existing (potentially modified) ones.

porkmanager, 2025-01-17:
> Any existing configs (as well as the previous YAMDCC version) should be moved to the `Old` folder, which doesn't get deleted after an update.

Have not noticed that folder. There are spawned too much folders in the new yamdcc version xD

Sparronator9999, 2025-01-17:
> There are spawned too much folders in the new yamdcc version xD

Blame the Task Scheduler NuGet package I use - it contains translations for a few languages, which is where all the extra folders come from.

If I were to set up a translation system (where contributors could translate the program to their native language), this would happen anyway, so I just left it.

Sparronator9999, 2025-01-17:
> Have not noticed that folder

The only ones you need to worry about are:

- **Configs**: where bundled configs are stored
- **Old**: the previous version of YAMDCC after an update
- **Update**: a temporary folder that holds the extracted YAMDCC update before it is installed (this gets deleted automatically on successful update)

The rest contain translations for the Task Scheduler plugin the updater uses to schedule automatic update checks.

Closing since existing configs don't appear to be getting deleted (just moved to `Old` if present in the YAMDCC install directory)

closed 2025-01-17 by Sparronator9999