Sparronator9999/YAMDCC #53: YAMDCC Service Crashed when Starting

kcos0320, 2025-02-05:
Hi,

Whenever I try to launch YamDCC, I get

[image]

 the following error message.

[yamdccsvc.exe.1.log](https://github.com/user-attachments/files/18679845/yamdccsvc.exe.1.log)

Sparronator9999, 2025-02-05:
That's weird, the service doesn't appear to actually be crashing, but starting, then stopping shortly after starting.

Can you please do the following:

1. Open Event Viewer `eventvwr.msc`, then go to `Windows Logs` > `Application` and wait for it to load.
2. Find an Error event log that looks similar to the following:

```
Application: yamdccsvc.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.FormatException
   at System.DateTime.ParseExact(System.String, System.String, System.IFormatProvider)
   at YAMDCC.Service.FanControlService.OnStart(System.String[])
   at System.ServiceProcess.ServiceBase.ServiceQueuedMainCallback(System.Object)
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.ServiceProcess.ServiceBase.Run(System.ServiceProcess.ServiceBase[])
   at YAMDCC.Service.Program.Main()
```

(the exception info will be different, this is just one I had while developing the EC firmware version feature)

3. Post the Event log here. If you couldn't find one, let me know.

You might find it easier to try attempting to start the YAMDCC service and look for any new event logs created during service startup.

kcos0320, 2025-02-06:
I ran the program, then immediately went to event viewer, but nothing ever showed up under application

However, I did see this Under: WindowsLog/System

Log Name:      System
Source:        Service Control Manager
Date:          2/6/2025 2:39:29 PM
Event ID:      7000
Task Category: None
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      MSI
Description:
The YAMDCC Service service failed to start due to the following error: 
The system cannot find the file specified.
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Service Control Manager" Guid="{555908d1-a6d7-4695-8e1e-26931d2012f4}" EventSourceName="Service Control Manager" />
    <EventID Qualifiers="49152">7000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>0</Task>
    <Opcode>0</Opcode>
    <Keywords>0x8080000000000000</Keywords>
    <TimeCreated SystemTime="2025-02-06T19:39:29.5510586Z" />
    <EventRecordID>16754</EventRecordID>
    <Correlation />
    <Execution ProcessID="1544" ThreadID="6000" />
    <Channel>System</Channel>
    <Computer>MSI</Computer>
    <Security />
  </System>
  <EventData>
    <Data Name="param1">YAMDCC Service</Data>
    <Data Name="param2">%%2</Data>
    <Binary>790061006D006400630063007300760063000000</Binary>
  </EventData>
</Event>

kcos0320, 2025-02-06:
Log Name:      System
Source:        Service Control Manager
Date:          2/6/2025 2:39:29 PM
Event ID:      7000
Task Category: None
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      MSI
Description:
The YAMDCC Service service failed to start due to the following error: 
The system cannot find the file specified.
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Service Control Manager" Guid="{555908d1-a6d7-4695-8e1e-26931d2012f4}" EventSourceName="Service Control Manager" />
    <EventID Qualifiers="49152">7000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>0</Task>
    <Opcode>0</Opcode>
    <Keywords>0x8080000000000000</Keywords>
    <TimeCreated SystemTime="2025-02-06T19:39:29.5510586Z" />
    <EventRecordID>16754</EventRecordID>
    <Correlation />
    <Execution ProcessID="1544" ThreadID="6000" />
    <Channel>System</Channel>
    <Computer>MSI</Computer>
    <Security />
  </System>
  <EventData>
    <Data Name="param1">YAMDCC Service</Data>
    <Data Name="param2">%%2</Data>
    <Binary>790061006D006400630063007300760063000000</Binary>
  </EventData>
</Event>

Sparronator9999, 2025-02-07:
> The YAMDCC Service service failed to start due to the following error:
> The system cannot find the file specified.

That should not be happening. Unfortunately Windows is being cryptic with its errors as always, and not telling you *what* file it can't find.

Did you try updating to the latest version of YAMDCC? The version you are/were running is quite out of date (five more betas, an RC, and a full release of YAMDCC have been made since v1.0 Beta 4).

marcyiu, 2025-02-10:
Hi! I am also experiencing the same issue as kcos0320. 
The error pops up when trying to start the YANDCC service when launching the Config Editor. 

<details>
<summary>Error from Event Viewer</summary>

---

**The YAMDCC Service service failed to start due to the following error: 
The system cannot find the file specified.** 

Log Name: System
Source: Service Control Manager
Event ID: 7000
Level: Error
User: N/A
OpCode: Info
Logged: 10/02/2025 12:44:14
Task Category: None
Keywords: Classic
Computer: DESKTOP-P9D3GLK

---
</details>

I initially experienced this error trying to launch an earlier beta release, but the error still persists after updating to the latest release (v1.0.0 Release). 
I am using an MSI Vector 16 HX A14V. 

> You might need to update your config, or start again from one provided with YAMDCC 

I'm sorry if this was meant to be straight forward, but could you tell me how could I do it without being able to launch the Config Editor? 
Also, is there any way to do a clean install of the YAMDCC service?

Thanks in advance!

Sparronator9999, 2025-02-10:
> I am also experiencing the same issue as kcos0320

Hmm, that's very weird. I almost feel like Windows is failing to find the service for some reason?

> > You might need to update your config, or start again from one provided with YAMDCC
> 
> I'm sorry if this was meant to be straight forward, but could you tell me how could I do it without being able to launch the Config Editor?

If you go to `C:\ProgramData\Sparronator9999\YAMDCC`, you will see two files (`CurrentConfig.xml` and `GlobalConfig.xml`) and a folder (`Logs`).

You can manually edit `CurrentConfig.xml` in a text editor (or copy a new one there), then restart the YAMDCC service (or run the Config Editor and click `Apply`).

> is there any way to do a clean install of the YAMDCC service?

You can try the following to clean-uninstall YAMDCC:

1. Open an Administrator Command Prompt and `cd` to the following folder: `C:\Windows\Microsoft.NET\Framework64\v4.0.30319`.
2. Locate your copy of `yamdccsvc.exe`, hold `Shift`, and right-click it, then select `Copy as path`.
3. Run the following command in the Command Prompt you opened earlier: `installutil /u "C:\path\to\yamdccsvc"`, replacing the path with the one you copied.
4. Verify that `YAMDCC Service` no longer appears in `services.msc`.
5. Move or delete `C:\ProgramData\Sparronator9999\YAMDCC` (this is where the active config and logs are stored).

You should now be able to start over from scratch.

If the `installutil` command failed, you can also try `sc delete yamdccsvc` in an admin Command Prompt (I think that was the command). This will force-delete the YAMDCC service.

marcyiu, 2025-02-11:
I no longer encounter the error after a clean install. Thank you very much!

Sparronator9999, 2025-02-11:
> I no longer encounter the error after a clean install. Thank you very much!

No problem!

I will update the "crash on start" error to also state that the YAMDCC service could have been moved from its original location at install-time (which also causes the "crash" message).

closed 2025-02-11 by Sparronator9999