Sparronator9999/YAMDCC #54: EC monitoring issue

porkmanager, 2025-02-08:
[image]

[yamdccsvc.exe.log](https://github.com/user-attachments/files/18722239/yamdccsvc.exe.log)

Hey. I thought this issue was solved along ago but suddenly i enabled ec monitoring and minimized configurator window for a while. 
Log is short cuz only error loggin was enabled.

Sparronator9999, 2025-02-08:
> I thought this issue was solved along ago

So did I.

This "operation completed successfully" error happens because reading from and writing to the EC occasionally takes much longer than expected (from <1 ms to 10 ms, or apparently even longer if you're getting those errors).

Ideally this wouldn't happen, but the OS (and other applications, if running) access the EC as well for their own purposes. For example, I think battery charge state is stored in EC (I think YAMDCC occasionally messes with the OS's EC access, which can cause it to straight up think the battery is "missing" even when clearly running on battery power...)

porkmanager, 2025-02-08:
> I don't really want to set it much longer, since if it's set high enough, the YAMDCC service can take so long to start that the SCM (Service Control Manager, the program that manages services, whether they're running, etc.) will kill the service with an error when trying to start it.

Ok. I cant even reproduce it one more time during last hour.

> Oh, and how long were you running the EC monitor for? The log shows the error messages occurring ~2-5 seconds apart?

It ran  just few minutes.

closed 2025-02-09 by porkmanager

Sparronator9999, 2025-02-09:
> Ok. I cant even reproduce it one more time during last hour.

Yeah... that's what makes this issue so hard to track down.

I might add an option to configure EC read/write timeout for advanced users.