Sparronator9999/YAMDCC #55: Missing .exe?

DoomCon, 2025-02-09:
### Prerequisites
- [ X ] I am using the correct [issue template](https://github.com/Sparronator9999/YAMDCC/issues/new/choose), i.e. I am asking the developer of YAMDCC a question. Bug reports and feature requests should be opened using their respective templates.
- [ X ] I have read the [FAQ](https://github.com/Sparronator9999/MSIFanControl#faq). Your question might already be answered there.
- [ X ] I have searched the [issue tracker](https://github.com/Sparronator9999/YAMDCC/issues?q=label:"type:+question") for duplicate or similar questions. Your issue will be closed if it is a duplicate of another issue.
- [ ] I have not answered truthfully to the above questions.

### Question
So I'm likely missing something, but I believe the YAMDCC-v1.0.0-Release.zip does not include a YAMDCC.exe file. When I try to open yamdccsvc.exe it tells me to use that first file instead, but I cannot find it in the folder. Is there some precompiling, additional service or another step I'm missing? I couldn't find any installation instructions on the github, but should I compile the program instead? Additionally, I assume this is related but the Updater.exe gives a crash report, I've added the report in the attachments. I admit I'm not used to github, so if there's a wiki/tutorial I'd need to watch please give me an idea on where to search.

[YAMDCC crash report.txt](https://github.com/user-attachments/files/18724118/YAMDCC.crash.report.txt)

DoomCon, 2025-02-09:
Allright, I seem to have found the issue, I hadn't run ConfigEditor.exe yet. So, my issue is solved. 
However if my ignorant self might propose something, I would add a small 'Install' section and/or a brief mention of the fact you have to run this exe first, as its not called installer and thus a bit unintuitive to find.

closed 2025-02-09 by DoomCon

Sparronator9999, 2025-02-10:
> However if my ignorant self might propose something, I would add a small 'Install' section and/or a brief mention of the fact you have to run this exe first, as its not called installer and thus a bit unintuitive to find.

First, in case you didn't find them, there are install instructions on [the Wiki](https://github.com/Sparronator9999/YAMDCC/wiki/Installation)

I was actually originally going to package YAMDCC into an installer (I still have an old InnoSetup script from before the project was called YAMDCC somewhere), but ended up having Config Editor do the service management as a "temporary" solution during development.

I do agree that the config editor probably shouldn't manage the YAMDCC service installation. I might move it to another app when I have time.