Sparronator9999/YAMDCC #61: The YAMDCC service fail to start after update and reset bios

VisionCrizzal, 2025-02-14:
So, after i'm update the bios (and reset bios because i'm fck up something on bios),YAMDCC is showing this:

The YAMDCC service failed to start.
This could be due to a service crash on startup, or the system couldn't
find the YAMDCC service (due to being moved or deleted since it was
installed).
See the FAQ for more information:
https://github.com/Sparronator9999/YAMDCC/wiki/FAQ

This is my log:

[14/02/2025 22:48:15.875] [Info]   OS version: Microsoft Windows NT 10.0.19045.0
[14/02/2025 22:48:15.875] [Info]   Service version: 1.0.1-release+bd52c9924982e537c36cf60d8fc4b421d77f883b
[14/02/2025 22:48:15.902] [Debug]  Log level is set to debug mode.
[14/02/2025 22:48:15.910] [Info]   Starting YAMDCC service...
[14/02/2025 22:48:15.910] [Info]   Reloading service config...
[14/02/2025 22:48:16.009] [Info]   Config loaded successfully!
[14/02/2025 22:48:16.009] [Info]   Loading WinRing0 driver for EC access...
[14/02/2025 22:48:16.010] [Info]   WinRing0 driver loaded successfully!
[14/02/2025 22:48:16.023] [Debug]  EC firmware version: 158PIMS1.114

VisionCrizzal, 2025-02-15:
.NET Runtime:

`Log Name:      Application
Source:        .NET Runtime
Date:          15/02/2025 11:09:40 am
Event ID:      1026
Task Category: None
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      WIN-556M0H3R3GC
Description:
Application: yamdccsvc.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.FormatException
   at YAMDCC.Service.FanControlService.OnStart(System.String[])
   at System.ServiceProcess.ServiceBase.ServiceQueuedMainCallback(System.Object)
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.ServiceProcess.ServiceBase.Run(System.ServiceProcess.ServiceBase[])
   at YAMDCC.Service.Program.Main()

Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name=".NET Runtime" />
    <EventID Qualifiers="0">1026</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>0</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2025-02-15T04:09:40.4141905Z" />
    <EventRecordID>4275</EventRecordID>
    <Correlation />
    <Execution ProcessID="0" ThreadID="0" />
    <Channel>Application</Channel>
    <Computer>WIN-556M0H3R3GC</Computer>
    <Security />
  </System>
  <EventData>
    <Data>Application: yamdccsvc.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.FormatException
   at YAMDCC.Service.FanControlService.OnStart(System.String[])
   at System.ServiceProcess.ServiceBase.ServiceQueuedMainCallback(System.Object)
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.ServiceProcess.ServiceBase.Run(System.ServiceProcess.ServiceBase[])
   at YAMDCC.Service.Program.Main()

</Data>
  </EventData>
</Event>
`
Application Error:

`Log Name:      Application
Source:        Application Error
Date:          15/02/2025 11:09:40 am
Event ID:      1000
Task Category: (100)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      WIN-556M0H3R3GC
Description:
Faulting application name: yamdccsvc.exe, version: 1.0.1.0, time stamp: 0xaa692ef2
Faulting module name: KERNELBASE.dll, version: 10.0.19041.5440, time stamp: 0x4b82124c
Exception code: 0xe0434352
Fault offset: 0x000000000003b699
Faulting process id: 0xe40
Faulting application start time: 0x01db7f5f6f384830
Faulting application path: C:\Users\Administrator\Downloads\YAMDCC-v1.0.1\yamdccsvc.exe
Faulting module path: C:\Windows\System32\KERNELBASE.dll
Report Id: 5e6f3f67-1b25-440e-bfd7-f46fb3305d28
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>100</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2025-02-15T04:09:40.5147122Z" />
    <EventRecordID>4276</EventRecordID>
    <Correlation />
    <Execution ProcessID="0" ThreadID="0" />
    <Channel>Application</Channel>
    <Computer>WIN-556M0H3R3GC</Computer>
    <Security />
  </System>
  <EventData>
    <Data>yamdccsvc.exe</Data>
    <Data>1.0.1.0</Data>
    <Data>aa692ef2</Data>
    <Data>KERNELBASE.dll</Data>
    <Data>10.0.19041.5440</Data>
    <Data>4b82124c</Data>
    <Data>e0434352</Data>
    <Data>000000000003b699</Data>
    <Data>e40</Data>
    <Data>01db7f5f6f384830</Data>
    <Data>C:\Users\Administrator\Downloads\YAMDCC-v1.0.1\yamdccsvc.exe</Data>
    <Data>C:\Windows\System32\KERNELBASE.dll</Data>
    <Data>5e6f3f67-1b25-440e-bfd7-f46fb3305d28</Data>
    <Data>
    </Data>
    <Data>
    </Data>
  </EventData>
</Event>`

Sparronator9999, 2025-02-15:
Yep, it's as I suspected. The EC firmware date is failing to be parsed for some reason.

If you wouldn't mind, could you run `ec-inspect --dump` in an Administrator Command Prompt and post the output?

VisionCrizzal, 2025-02-16:
I'm install ghost spectre win 10,it's have build in Administrator account.
I'm run ec-inspect --dump and it's show this: 
`PS C:\Users\Administrator\Downloads\YAMDCC-v1.0.1-Release> .\ec-inspect --dump
ERROR: the YAMDCC service is not running.
Please run the YAMDCC config editor to start the service.
`

Sparronator9999, 2025-02-16:
uh i may have forgotten about ec-inspect's dependency on the YAMDCC service...

You'll need to install [NoteBook FanControl](https://github.com/UraniumDonut/nbfc-revive) and use its tool, `ec-probe`, to dump the EC.

I might update `ec-inspect` in the future so that it doesn't depend on the YAMDCC service running, but for now this will work.

VisionCrizzal, 2025-02-16:
`
   | 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
---|------------------------------------------------
00 | 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00
10 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20 | 00 00 00 00 00 00 00 00 0A 05 00 00 00 00 0B 1A
30 | 03 09 00 0D 00 00 50 81 D2 11 88 2C C8 01 E0 00
40 | 00 00 5E 00 51 11 00 00 2D 10 91 31 D5 0B 00 00
50 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
60 | 00 00 00 00 00 00 00 00 28 00 37 41 46 4B 50 55
70 | 64 26 26 2B 30 36 3C 46 55 64 08 03 03 03 03 03
80 | 00 00 37 41 46 4B 50 55 69 00 00 2B 30 36 3C 46
90 | 55 69 08 03 03 03 03 03 06 00 00 06 00 00 00 00
A0 | 31 35 38 50 49 4D 53 31 2E 31 31 34 31 32 32 35
B0 | 32 30 32 34 20 39 3A 33 37 3A 33 36 00 00 00 08
C0 | 00 00 06 22 00 00 00 0F 00 D8 00 00 00 00 00 00
D0 | 00 00 C0 83 0D 00 05 80 00 05 00 00 00 00 00 00
E0 | E2 00 00 51 11 00 00 00 01 00 00 00 00 83 00 00
F0 | 40 00 C0 00 26 F0 28 90 22 F0 00 00 00 00 00 00
`

Sparronator9999, 2025-02-16:
I figured out the problem.

The date from your EC decodes to `12252024 9:37:36`, or December 25th 2024 at 9:37:36 AM.

I think the parser I'm using expects a leading zero before the 9 in the time (i.e. `09:37:36`), causing a crash when it gets a whitespace character instead.

This will be fixed Soon.

Sparronator9999, 2025-02-16:
Hi, I have fixed the date parsing error.

Can you please test the latest build from [GitHub Actions](https://github.com/Sparronator9999/YAMDCC/actions)? You should be able to see your EC firmware date on the `Info` tab of the config editor after clicking `Get model and manufacturer`.

closed 2025-02-16 by Sparronator9999

VisionCrizzal, 2025-02-16:
I'll test it soon

VisionCrizzal, 2025-02-16:
After i clicked `Get model and manufacturer` , it's show nothing in EC firmware date on info tab.
beside that, everything work perfectly.

Sparronator9999, 2025-02-16:
That should not happen.

Can you please post your service log? (found at `C:\ProgramData\Sparronator9999\YAMDCC\Logs`)?

VisionCrizzal, 2025-02-16:
[yamdccsvc.exe.log](https://github.com/user-attachments/files/18813812/yamdccsvc.exe.log)

Sparronator9999, 2025-02-16:
I figured out the problem.

I decided to only get the EC firmware version on service boot with whatever config it loaded at the time, but you appeared to have an invalid (or maybe outdated?) config loaded, so the YAMDCC service couldn't get your EC firmware date/version.

Try restarting the YAMDCC service, then try clicking `Get model and manufacturer` again.

I will fix this as well in a commit.

VisionCrizzal, 2025-02-16:
It's show!!!

[image]

After restart the YAMDCC service and reloaded config

Sparronator9999, 2025-02-16:
Looks like it works fine now.

I have (hopefully) fixed the EC firmware version not working if a config with EC firmware support wasn't loaded on service start.

A new release will be published shortly including these changes.

commit ecb66d5a (2025-02-16): Fix date parsing error for EC firmwares released before 10 AM

Fixes #61