Sparronator9999/YAMDCC #62: YAMDCC Updater Crashes on Installing Update

marcyiu, 2025-02-17:
### Prerequisites
- [x] I am using the correct [issue template](https://github.com/Sparronator9999/YAMDCC/issues/new/choose), i.e. I am reporting a bug or crash in YAMDCC. Feature requests and questions should be opened using their respective templates.
- [x] I have searched the [issue tracker](https://github.com/Sparronator9999/YAMDCC/issues?q=label:"type:+bug") for duplicate or similar bug reports. Your issue will be closed if it is a duplicate of another issue.
- [ ] I have not answered truthfully to the above questions.

### Describe the bug
YAMDCC Updater Crashes on Installing Update

### To Reproduce
Steps to reproduce the behavior:
1. Click on the update button on the 'Update available!' window
2. Wait for downloading to complete
3. 'Installing YAMDCC update...' pops up briefly
4. Crash occurs

<details>
<summary>Crash Report</summary>

```
Version: 1.0.1
Revision (git): bd52c9924982e537c36cf60d8fc4b421d77f883b
Crash report:
System.UnauthorizedAccessException: Access to the path 'WinRing0x64.sys' is denied.
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.FileInfo.Delete()
   at YAMDCC.Updater.Program.<>c__DisplayClass1_0.<InstallUpdate>b__0()
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at YAMDCC.Common.Dialogs.ProgressDialog`1.<OnLoad>d__12.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
```
</details>

### Expected behavior
Update to succeed

### Additional context
Hi, I'm the person who did the manual clean install of 1.0.0 (in https://github.com/Sparronator9999/YAMDCC/issues/53#issuecomment-2649010764)

I experienced this updater crash trying to update from 1.0.0 to 1.0.1, but thought it might be due to me messing up to my manual clean install, so I turned to doing a manual update. However, it crashed again this time. 

Screen recording: 

https://github.com/user-attachments/assets/506246b4-dc1a-4770-8d36-e9d412a6fa74

Sparronator9999, 2025-02-17:
Hmm, that's weird.

I just tried "updating" my YAMDCC and nothing out of the ordinary happened (I did find an unrelated bug though. That will be fixed soon).

Anyways, back to the reported bug. It looks like the WinRing0 driver is still being used by the system somewhere. Try uninstalling the YAMDCC service then reboot your computer (or setting its startup mode to `Manual` or `Disabled` in `Services` should be enough). Then try updating again.

marcyiu, 2025-02-18:
I set the 'startup mode' of the YAMDCC Service to `Disabled` and restarted my laptop

Both the 'ConfigEditor' and 'Updater' .exe files disappeared after I restarted. Is that supposed to happen?

I applied the update from the `\Update` folder manually, and now the Updater works when I do a forced reinstall. 

Thanks!

Sparronator9999, 2025-02-18:
> Both the 'ConfigEditor' and 'Updater' .exe files disappeared after I restarted. Is that supposed to happen?

That probably happened because the update failed halfway through.

> I applied the update from the \Update folder manually, and now the Updater works when I do a forced reinstall.

Weird. I might need to improve my driver uninstall code to check to make sure the driver is actually no longer running.

...wait.

Did you have any other programs running (in particular, hardware monitoring) while you were trying to update YAMDCC?

marcyiu, 2025-02-18:
I don't think I have any hardware monitoring programmes running (not that I know of)

I have only regular programs (browser, Teams, etc.) open

Sparronator9999, 2025-02-18:
That's very weird.

The only reason I asked is because the YAMDCC service doesn't uninstall WinRing0 if it detects another program might be using it, which would cause the `UnauthorizedAccessException` you see while installing the update. The reboot with YAMDCC service disabled ensures that nothing could possibly be using the `WinRing0x64.sys` that comes with YAMDCC (allowing it to be overwritten in an update).

Sparronator9999, 2025-02-18:
@marcyiu, can you please try the following to help me figure out the problem:

1. Make sure the YAMDCC service is running.
2. Open `regedit` and navigate to the following registry key: `HKLM\SYSTEM\CurrentControlSet\Services\WinRing0_1_2_0`.
3. Stop the YAMDCC service, then try to open the above registry key again. You should get an "not found" error. If not, WinRing0 isn't being uninstalled for some reason (which prevents it from being overwritten, leading to the original bug)

marcyiu, 2025-02-23:
Hi! Sorry for not replying the past few days, was quite busy. 

I've tried your steps, but it the `WinRing0_1_2_0` is still present after manually stopping the YAMDCC Service in Services. 
The registry key is still present in the Registry Editor after refresh. 

I'm still on 1.0.2 since I'm unable to update to 1.0.3
Here's the crash log: 
```
Version: 1.0.2
Revision (git): 046ba5110f323bcca661cec4e9f6070d8ccddaf7
Crash report:
System.UnauthorizedAccessException: Access to the path 'WinRing0x64.sys' is denied.
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.FileInfo.Delete()
   at YAMDCC.Updater.Program.<>c__DisplayClass1_0.<InstallUpdate>b__0()
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at YAMDCC.Common.Dialogs.ProgressDialog`1.<OnLoad>d__12.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
```

The same behavior of the registry key is exhibited on 1.0.3
Here's the crash report when I attempted a forced reinstall
```
Version: 1.0.3
Revision (git): 1a032a841dd8467692edebc6a7b28a4b6bf98c41
Crash report:
System.UnauthorizedAccessException: Access to the path 'WinRing0x64.sys' is denied.
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.FileInfo.Delete()
   at YAMDCC.Updater.Program.<>c__DisplayClass1_0.<InstallUpdate>b__0()
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at YAMDCC.Common.Dialogs.ProgressDialog`1.<OnLoad>d__15.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
```

Sparronator9999, 2025-02-23:
> I've tried your steps, but it the `WinRing0_1_2_0` is still present after manually stopping the YAMDCC Service in Services.

Have you tried rebooting after manually disabling the YAMDCC Service? Setting its startup type to Manual should be enough.

Is the `WinRing0_1_2_0` registry key still present after rebooting the computer? If it is, and you're sure nothing else is running that uses WinRing0, you can try force-deleting it with `sc delete WinRing0_1_2_0`. Reboot and check for the registry key's existence.

If after all that the registry key's still there, the `ImagePath` should hint you to whether YAMDCC installed it or not (and maybe even what, if any, other program is using it). If it was YAMDCC, it should point to somewhere in YAMDCC's install directory, e.g. `C:\Program Files\YAMDCC\WinRing0x64.sys`, otherwise it will point to either System32's `drivers` directory, or if you're lucky it will give away another program that uses WinRing0.

Either way, I'll have to figure out a way to work around this issue...

Sparronator9999, 2025-05-04:
Closing this issue since the Updater (in the latest betas) no longer does the actual installation (it's handled by [Inno Setup](https://jrsoftware.org/isinfo.php) instead).

closed 2025-05-04 by Sparronator9999