Sparronator9999/YAMDCC #64: YAMDCC update crash

VisionCrizzal, 2025-02-21:
Hey,it's me again.
Yes,i were read about the issuse when update the app but i'm put the error log here for....... just in case you need.

[image]

`Version: 1.0.2
Revision (git): 046ba5110f323bcca661cec4e9f6070d8ccddaf7
Crash report:
System.IO.DirectoryNotFoundException: Could not find a part of the path 'C:\Windows\system32\Configs'.
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.FileSystemEnumerableIterator`1.CommonInit()
   at System.IO.FileSystemEnumerableIterator`1..ctor(String path, String originalUserPath, String searchPattern, SearchOption searchOption, SearchResultHandler`1 resultHandler, Boolean checkHost)
   at System.IO.DirectoryInfo.InternalGetFiles(String searchPattern, SearchOption searchOption)
   at YAMDCC.Updater.UpdateForm.<btnUpdate_Click>d__11.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
`

Sparronator9999, 2025-02-21:
I fixed this already in v1.0.3. Earlier versions will have this issue, since YAMDCC uses the included Updater up until the new version of YAMDCC is successfully installed.

> how can i paste the log correctly to look like a "code"?

Using three backtick symbols (these things: \`) around the content you want to look like a "code".

> i'm update successful when follow your guide

Is this related to [#62](https://github.com/Sparronator9999/YAMDCC/issues/62)? That issue was caused by a file still being in use (for some reason, I haven't figured it out yet) during an update.

porkmanager, 2025-02-21:
I had the same issue today while updating from 1.0.2 to 1.0.3. The issue was in the autoupdater task. When I opened the config editor and ran the update through it, the update completed successfully.

Sparronator9999, 2025-02-21:
> I had the same issue today while updating from 1.0.2 to 1.0.3

Are you talking about this issue, or [#62](https://github.com/Sparronator9999/YAMDCC/issues/62)?

> The issue was in the autoupdater task. When I opened the config editor and ran the update through it, the update completed successfully.

That's weird, both auto-update task and update via config editor (or running with `--autoupdate` and `--checkupdate` respectivey) do the same thing, except one shows a "YAMDCC is up to date" message box if up to date.

Sparronator9999, 2025-02-21:
I just had a closer look at the crash log, and noticed that the updater is trying to access SYSTEM32 FOR SOME REASON???

I am astounded. A fix will be released shortly.

Sparronator9999, 2025-02-21:
Okay, NOW the updater crashes should be fixed.

Hopefully Windows doesn't find some weird way to screw with the relative file paths again.

commit 51bc21e5 (2025-02-21): (hopefully) Fix updater crash if Configs folder doesn't exist in old version once and for all

This also fixes potential issues with the Updater trying to access System32 folders (due to the way that Windows resolves relative paths)

Fixes #64
closed 2025-02-21 by Sparronator9999