Sparronator9999/YAMDCC #65: Best to warn EC risks

ViamoIam, 2025-03-07:
### Prerequisites
- [ ] I am reporting a bug or feature request in YAMDCC's Readme.
- [ ] I have searched the for duplicate issues
- [ ] I have answered truthfully to the above questions. 

### Describe the bug
A EC risk warning may good as well <actually not joking>

### To Reproduce
Steps to reproduce the behavior:
1. Go down the list
2. Click on load a config
3. Scroll down to config that has same name, but different product code.
4. Find your system shutting down after a minute after start

### Expected behavior
I hope the warnings can be improved a bit so they inform better, and prevent people from making a mistake  

### Additional context
(I hope I or others could contribute here as this sort of thing is a waste of devs time, and users if running into issues that could be avoided or resolved quickly)

(I'll try to look at later again within 48 hrs) Here is a 1st draft. Maybe good enough?)

**Warning the Embedded Controller (EC)  manages the keyboard, charging, power, cooling, and ?temp reporting iirc? While sometimes it can be recovered by reseting the EC or reflashing <add links> Your system may be no longer function or even be permanently damaged.

I have used other EC projects like msi-ec, mcontrolcenter, and others. 

I suspect my hardware may be having a separate hardware board issue preventing communication from Ec to OS. EC reset  were needed more often lately with ryzenadj, uxtu and ghelper. This happens to be first time a EC reset or reflashing the EC doesn't resolve all EC issues. Msi center and ryzenadj were not showing normal values and functioning normally.

ViamoIam, 2025-03-07:
I was suggested to share. But i still doubt the usefulness. Maybe a warning to not forget the product code. Never load the wrong config.

Sparronator9999, 2025-03-08:
I mean, there is *a* warning in the README, but I agree it could be a bit better.

As for loading the "wrong" config, this ***shouldn't*** matter too much unless you load a config for a laptop with an Intel 9th-gen CPU or earlier on a laptop with an Intel 10th-gen CPU or later, or vice-versa.

Assuming the loaded config and laptop's generations match, the only things wrong will be the default fan profiles, and potentially some extra features (in particular the keyboard backlight) not working or giving an error.

Maybe I should add a warning on first startup of YAMDCC about the potential risks of modifying the EC registers (and update the README warning). I will get to work on this Soon™.

ViamoIam, 2025-03-26:
Perhaps a warning to not use the wrong config would be appropriate. This is actually an Alpha 15 b5eek AMD system. Unfortunately a later model is actually an Intel system. You mention the difference between the 9th and earlier and 10th and later, but there is probably a difference in the AMD systems. AMD Smartshift and Intel/Nvidia Dynamic shared power could be different for example.

Sorry, I haven't been able to do something constructive for this 'issue'. Access to resources/time turned out to me more limited. The system will always restart after about a minute. I can't seem to fix system normally and then test to replicate the issue.

There could have been a hardware issue, though I believe my error was using the MSI Alpha 17 C7VF (Intel system) config on my MSI Alpha b5eek (AMD Zen3) system. (Yes this was a dumb tired move) The MSI Bravo 15 B7ED (AMD Zen4) config was tried to see if it would help, but no luck. It isn't obviously the loading of the config but commiting a change to the EC. My terminology will be wrong here as I know little other then iirc the embedded controller is its own system on chip, so I think of the firmware as its own system with flash storage, power cycles, etc. 

(Standard MSI embedded controller firmware trouble shooting was applied. The EC was power cycled (EC reset). EC firmware was flashed from windows. System still restarted after a minute or so. Then BIOS CMOS was reset, then BIOS could be 'updated'. Bios update seemed to rewrite the EC after the restart, though this was done blind. The screen ferrite is blown from a unrelated issue and is waiting on repair. I did confirmed the EC being written through the expected behaviour described in bios update documentation, because normally the system does not have the screen on during this procedure.  Basically the power button flashed orange to white after the system restarted as normal while EC was written. )

ViamoIam, 2025-03-26:
Here is the commit link that added support for this system to the MSI-EC project. I don't think there is a hardware issue in the EC itself, but I do know that they have released new EC firmware versions even in 2023, about 2 years after release.

MSI Alpha B5eek support added to msi-ec https://github.com/BeardOverflow/msi-ec/commit/7bdec2ff7d0cb571d65a099f3e10aa488888f570

The MSI Alpha 15 spec/support page. I believe I was still running E158LAMS.106 instead of E158LAMS.10F
https://ca.msi.com/Laptop/Alpha-15-B5EX/support?sub_product=Alpha-15-B5EEK

I found you can download prior bios and firmware versions from MSI support page by altering url to the correct release. Here i have changed the filename to the release that was installed on my system iirc. 
https://download.msi.com/bos_exe/nb/E158LAMS.106.zip