Sparronator9999/YAMDCC #68: CMD when? XD

porkmanager, 2025-03-18:
As we can see you are done with the hotkeys handler. Does that means that soon we can wait command line utility for yamdcc? 
CMD is most awaited feature of yamdcc for now in particular for me.
Thank you for your awesome app!

Sparronator9999, 2025-03-18:
> Thank you for your awesome app!

Why, you're very welcome!

> CMD is most awaited feature of yamdcc for now in particular for me.

The CLI is next, assuming nothing else goes wrong with the rest of YAMDCC.

> As we can see you are done with the hotkeys handler

I mean, it's still in beta, but no one appears to be finding any issues with it, so might be time for a full release?

porkmanager, 2025-08-14:
Hey! Im testing cmd now. Could i find a changelog of implemented features? Some of those functions works well some does not works.  All "Program information" and "basic commands" works fine. -chargelim also works properly. 
But for exmpl  -keylight isnt, i see success in console but kb light does nothing.

[image]

Cant understood or it does not finished yet but fan control features does not works for me. command like: yamdcc.exe -F 1 -P 3 does nothing.

Sparronator9999, 2025-08-14:
All features should be implemented, so any that aren't working should be reported. I will look into the `-keylight` command.

Also, is it just me, or is your CMD window slightly transparent?

> Cant understood or it does not finished yet but fan control features does not works for me. command like: yamdcc.exe -F 1 -P 3 does nothing.

Please post command line output (or does it not print anything?)

porkmanager, 2025-08-15:
> Also, is it just me, or is your CMD window slightly transparent?

Yep, it is semi transparent. Default in the windows 11.

> You need to add the `--apply` (or `-A`) command to apply the fan settings.

Ah sure. I was sleepy to read the cmd output well XD
But anyway it has no affect on configs with -apply or -A arguments. Console output is: Connecting to YAMDCC service...
Connected successfully! but changes nothing, like with -keylight command.
`C:\Program Files\YAMDCC>yamdcc.exe -I
                                     _                                  _   _
                                    | |                                | | |_|
  _    _   ______   _________   ____| |  ______   ______       ______  | |  _
 | |  | | |  __  | |  _   _  | |  __  | |  ____| |  ____|     |  ____| | | | |
 | |  | | | |  | | | | | | | | | |  | | | |      | |          | |      | | | |
 | |__| | | |_/  | | | | | | | | |__| | | |____  | |____      | |____  | | | |
 |____  | |___/|_| |_| |_| |_| |______| |______| |______|     |______| |_| |_|
  ____| |
 |______|  v1.2.0-beta.2

Copyright c Sparronator9999 (and Contributors) 2025
This program is Free Software; run with the "-L" argument to print full license info.

~~~~~ Config info ~~~~~
Author: porkmanager
Laptop manufacturer: Micro-Star International Co., Ltd.
Laptop model: Katana GF66 12UG
EC firmware version: 1583EMS1.110
EC firmware date: 23.04.2024 9:40:13

~~~~~ 0: CPU Fan ~~~~~
Fan speed range: 0-100%
Current fan profile: Default
Available fan profiles:
  0: Default
  1: TEST

Current fan profile settings:
  Speed (%):  38  43  48  54  60  70  85
    Up (°C):   L  55  62  69  76  82  87
  Down (°C):  47  58  65  74  79  85   H

~~~~~ 1: GPU Fan ~~~~~
Fan speed range: 0-100%
Current fan profile: Default
Available fan profiles:
  0: Default
  1: TEST

Current fan profile settings:
  Speed (%):   0  43  48  54  60  70  85
    Up (°C):   L  55  61  67  72  77  82
  Down (°C):  47  58  64  69  74  79   H

~~~~~ Extras ~~~~~
Charge threshold: 60%
Win/Fn key swap: disabled

Default performance mode: Balanced
Available performance modes:
  0: Maximum battery life
  1: Silent
  2: Balanced
  3: High performance

C:\Program Files\YAMDCC>`

Then i trying to switch fan 0(cpu) to 1(TEST) profile: yamdcc.exe -F 0 -P 1 -A
It prints: success, but profile isnt switches, the same as -keylight command. Prints OK, but nothing happens.

<img width="1103" height="639" alt="Image" src="https://github.com/user-attachments/assets/6368577a-b08c-482d-b0ea-0ffc18493fe9" />

Also tested -perfmode argument, works properly, it sets global performance mode.

porkmanager, 2025-08-15:
Short showcase
https://github.com/user-attachments/assets/45d5c207-252e-4bed-b6d1-db2df367dd2f

Sparronator9999, 2025-08-15:
okay so uh I *might* have forgotten to implement fan profile switching

To *edit* a fan profile, you'd need the `-speed/-S` switch, which combine with the `-fan/-F` and `-profile/-P` switches to edit a single fan profile temperature and/or speed. But obviously you just want to change the fan profile...