Sparronator9999/YAMDCC #69: Service failed to install

tomsLt0, 2025-03-18:
### Prerequisites
- [yes ] I am using the correct [issue template](https://github.com/Sparronator9999/YAMDCC/issues/new/choose), i.e. I am reporting a bug or crash in YAMDCC. Feature requests and questions should be opened using their respective templates.
- [yes ] I have searched the [issue tracker](https://github.com/Sparronator9999/YAMDCC/issues?q=label:"type:+bug") for duplicate or similar bug reports. Your issue will be closed if it is a duplicate of another issue.
- [ ] I have not answered truthfully to the above questions.

### Describe the bug
When installing the service the service fails to install. 

[image]

### To Reproduce
Steps to reproduce the behavior:
1.Open ConfigEditor.exe as admin
2. press yes to installing the service
3. See error

### Expected behavior
for the app to successfully install and run

### Additional context

[image]
yesterday i installed fresh windows 11 and debloated it with talon

Sparronator9999, 2025-03-18:
> [yes ] I am using the correct [issue template](https://github.com/Sparronator9999/YAMDCC/issues/new/choose), i.e. I am reporting a bug or crash in YAMDCC. Feature requests and questions should be opened using their respective templates.
[yes ] I have searched the [issue tracker](https://github.com/Sparronator9999/YAMDCC/issues?q=label:%22type:+bug%22) for duplicate or similar bug reports. Your issue will be closed if it is a duplicate of another issue.

You can use an X (replacing the space) to show actual check boxes (so it would look like [X]). But that's just a nitpick.

Now, you wouldn't happen to be running Windows Defender? As of a few days ago, I've heard reports of that antivirus (and probably others now) reporting WinRing0 (the driver we use for EC access) as malware.

> debloated it with talon

I'm not familiar with that piece of software (personally I run Windows 10 LTSC on my own computers), so not sure what I can do if it ends up being a problem with that.

tomsLt0, 2025-03-18:
Tried without windows defender still the same thing, btw I noticed the installutil.exe logs are being deleted could you explain why that was done, I think having them might have made it easier to resolve this, but anyways i installed the service manually with cmd "sc create yamdccsvc binPath= "C:\Users\Tomas\Documents\YAMDCC-v1.0.3-Release\yamdccsvc.exe" start= auto" the service is running but now i got a new problem, idk if its because of the way i installed the service but the option to get fan profile from EC is grayed out 

[image]

Sparronator9999, 2025-03-18:
> I noticed the installutil.exe logs are being deleted could you explain why that was done

Not sure why I did it in the first place, probably to keep the install directory clean.

The latest commit (at time of writing) removes this functionality. Perhaps you can post the InstallUtil logs now?

tomsLt0, 2025-03-18:
You'll need to give me the compiled version. Hahaha didnt see that u might need to add a pop up or something, and i think my manual service installation didnt work.

[image]

Sparronator9999, 2025-03-18:
> You'll need to give me the compiled version

You can download the latest builds from [GitHub Actions](https://github.com/Sparronator9999/YAMDCC/actions). 

> i think my manual service installation didnt work.

I don't see any reason for `sc` to not install the YAMDCC service properly. If anything, it'll be missing the service description (and name?) from `Services.msc`.

Sparronator9999, 2025-03-20:
>  all i had to do was press that unblock button 😭

i just love [Mark Of The Web](https://en.wikipedia.org/wiki/Mark_of_the_Web) (/s). Funnily enough you're the first one to report an issue with it.

I'm assuming you ended up finding a working config for your laptop? Could you share your laptop model (and the config you used) so I can add it to the supported laptop's list?

jjrobertlol, 2025-03-20:
Hi, I'm very unfamiliar with github and whatnot, but I had the same issue as toms at the beginning, and just to note, in his description of "sc create yamdccsvc binPath= "C:\Users\Tomas\Documents\YAMDCC-v1.0.3-Release\yamdccsvc.exe" start= auto", there needs to be no space after the equals sign at the end, so that it should be "sc create yamdccsvc binPath= "C:\Users\Tomas\Documents\YAMDCC-v1.0.3-Release\yamdccsvc.exe" start=auto" instead. I am however now not sure how to proceed, my laptop is an msi delta 15, and I'm not sure which generic profile is close to that. I tried all of the ones listed, though it seems that the 9th gen or older ones (both of them), don't do anything when loaded into the program. Do you think you could help me out too? I used the ec-config file, but I don't know what it output, or how to give it to you.

Sparronator9999, 2025-03-20:
Hi @jjrobertlol, I searched for your laptop, and it appears to have an RGB keyboard, which aren't supported by YAMDCC (as evidenced by the `GetKeyLightBright` command error). I would recommend one of the generic configs with a "-nokeylight" suffix at the end of the config name.

As for Intel 9th-gen or earlier vs Intel 10th-gen or newer, the main difference is the extra features (charge limit, performance mode, Win/Fn swap, and keyboard backlight brightness on monochrome keyboards), which are set by different EC registers between two "generations" of ECs. The first generation was used in laptops with a 9th-gen Intel processor (as well as all laptops before then), and the second generation was used in all laptops following that (so 10th-gen and later). I don't know what the cutoff is for AMD laptops with "first-gen" ECs, so I used Intel.

Hopefully that clears the config choices up a bit.

Now, as for the YAMDCC service not installing, try the latest build from [GitHub Actions](https://github.com/Sparronator9999/YAMDCC/actions), which changes the service installation to keep the `InstallUtil` logs. Can you please upload the `InstallUtil.InstallLog` and `yamdccsvc.InstallLog` files (located in the same directory as YAMDCC)?

**NOTE: you may want to remove your computer's username from several places in those logs if you're installing the YAMDCC service from your user folder.**

@tomsLt0 can you try installing the service from YAMDCC and post your logs as well?

jjrobertlol, 2025-03-20:
I have the files, but how do I upload them? This is their contents:
Running a transacted installation.

Beginning the Install phase of the installation.
See the contents of the log file for the C:\Users\anony\Downloads\YAMDCC-main-1eeba79-Release\yamdccsvc.exe assembly's progress.
The file is located at C:\Users\anony\Downloads\YAMDCC-main-1eeba79-Release\yamdccsvc.InstallLog.

The Install phase completed successfully, and the Commit phase is beginning.
See the contents of the log file for the C:\Users\anony\Downloads\YAMDCC-main-1eeba79-Release\yamdccsvc.exe assembly's progress.
The file is located at C:\Users\anony\Downloads\YAMDCC-main-1eeba79-Release\yamdccsvc.InstallLog.

The Commit phase completed successfully.

The transacted install has completed.

and

Installing assembly 'C:\Users\anony\Downloads\YAMDCC-main-1eeba79-Release\yamdccsvc.exe'.
Affected parameters are:
   logtoconsole = 
   assemblypath = C:\Users\anony\Downloads\YAMDCC-main-1eeba79-Release\yamdccsvc.exe
   logfile = C:\Users\anony\Downloads\YAMDCC-main-1eeba79-Release\yamdccsvc.InstallLog
Installing service yamdccsvc...
Service yamdccsvc has been successfully installed.
Creating EventLog source yamdccsvc in log Application...
Committing assembly 'C:\Users\anony\Downloads\YAMDCC-main-1eeba79-Release\yamdccsvc.exe'.
Affected parameters are:
   logtoconsole = 
   assemblypath = C:\Users\anony\Downloads\YAMDCC-main-1eeba79-Release\yamdccsvc.exe
   logfile = C:\Users\anony\Downloads\YAMDCC-main-1eeba79-Release\yamdccsvc.InstallLog

I managed to get the full blast thing to work, but I can't adjust any of the other options, like the fan curve.

[image]

This was with the single-fan model without keylight, the same happens with the dual-fan model without keylight

nm17, 2025-03-20:
I think I found a definitive fix for the "serivce failed to install" issue. Make sure YAMDCC is located in the Program Files folder. It works fine from there. 

P.S. @Sparronator9999 you should probably make a YAMDCC package for Chocolatey, it would save a lot of time installing.

nm17, 2025-03-20:
> I tried all of the ones listed, though it seems that the 9th gen or older ones (both of them), don't do anything when loaded into the program.

Same here

Sparronator9999, 2025-03-20:
> Make sure YAMDCC is located in the Program Files folder

I've been installing the YAMDCC service from my user folder for the past two years and had no issues.

> > I tried all of the ones listed, though it seems that the 9th gen or older ones (both of them), don't do anything when loaded into the program.
> 
> Same here

What laptop model do you have? I'm going to take a guess and say that MSI switched to "second-gen" ECs around 2019-2020 or so (when the 10th-gen Intel CPUs released). I need more users with AMD-based MSI laptops to report which config works for them (10th-gen or newer, or 9th-gen or earlier) along with at least their laptop model (for the supported laptops list, and so I can look up which CPU model was used).

> you should probably make a YAMDCC package for Chocolatey, it would save a lot of time installing.

I have an old InnoSetup script somewhere that I used to make an MSI Fan Control (old name for YAMDCC) installer back in like 2023, that I could repurpose into a YAMDCC installer.

The reason why YAMDCC's Config Editor is used to manage the YAMDCC service is because it saves me time during development by not having to constantly use `net`, `installutil`, or `services.msc` to manage the service manually, which eventually made it into the final releases...

VisionCrizzal, 2025-03-21:
Hey, i just heard about you need more about AMD based Msi laptop report about config.
Currently i'm using MSI bravo 15 B7ED(not the B7EDP),with the spec:
AMD Ryzen 5 7535HS
Gpu AMD Radeon Rx 6550m
I'm confirm that the script 10th-gen or newer laptop work for me,and i'm use that script and follow the instructions to create my own laptop config.
(I'm use full AMD combo because in the future,when i switch to linux,i dont have to configure the gpu driver like NVIDIA

Sparronator9999, 2025-03-21:
> Hey, i just heard about you need more about AMD based Msi laptop report about config.
> Currently i'm using MSI bravo 15 B7ED(not the B7EDP),with the spec:
> AMD Ryzen 5 7535HS
> Gpu AMD Radeon Rx 6550m [...]

> (I'm use full AMD combo because in the future,when i switch to linux,i dont have to configure the gpu driver like NVIDIA 

Haha, fair. I've tried out Linux Mint on my GF63 Thin 11SC (i7-11800H + NVIDIA GeForce 1650 Max-Q), and still haven't figured out how to get the "Optimus" mode to work properly (i.e. the GPU stays "awake" even when it's not being used sometimes).

Thanks for that. I don't need *exact* CPU specs, just whether it's an Intel or AMD, and the CPU generation/year of laptop release.

Sparronator9999, 2025-03-24:
> Win Defender Off Win11 Core i5-11400h
> 
> Programm not install (from configeditor.exe)
> 
> updater.exe:
> 
> ```
> System.Net.Http.HttpRequestException: Произошла ошибка при отправке запроса. ---> System.Net.WebException: Базовое соединение закрыто: Не удалось установить доверительные отношения для защищенного канала SSL/TLS. ---> System.Security.Authentication.AuthenticationException: Удаленный сертификат недействителен согласно результатам проверки подлинности.
>    в System.Net.TlsStream.EndWrite(IAsyncResult asyncResult)
>    в System.Net.ConnectStream.WriteHeadersCallback(IAsyncResult ar)
>    --- Конец трассировки внутреннего стека исключений ---
>    в System.Net.HttpWebRequest.EndGetResponse(IAsyncResult asyncResult)
>    в System.Net.Http.HttpClientHandler.GetResponseCallback(IAsyncResult ar)
>    --- Конец трассировки внутреннего стека исключений ---
>    в System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
>    в System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
>    в YAMDCC.Updater.Updater.<GetLatestReleaseAsync>d__4.MoveNext()
> --- Конец трассировка стека из предыдущего расположения, где возникло исключение ---
>    в System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
>    в System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
>    в YAMDCC.Updater.UpdateForm.<CheckUpdate>d__17.MoveNext()
> ```

Hmm, looks like a possible certificate/authentication issue? Maybe?? Doesn't help that I have to rely on Google Translate...

This is a bit out of my league. The only thing I can suggest is try again after some time.

Oh, and this is a separate issue, unrelated to the YAMDCC Service installation error originally reported. Open another issue next time.

nm17, 2025-03-25:
> What laptop model do you have?

MSI GF75 Thin 9SCXR

Miskler, 2025-03-25:
> Hmm, looks like a possible certificate/authentication issue? Maybe?? Doesn't help that I have to rely on Google Translate...
>
> This is a bit out of my league. The only thing I can suggest is try [...]
I get exactly the installation error that is described in this thread, the error updater.exe I put it in bulk :)

MSI GF63 11UC. I tried to install it on a completely clean system.

Sparronator9999, 2025-04-29:
Hi, closing this issue since the @tomsLt0 fixed their issue on their own.

@nm17, I have no experience with Chocolatey. A "proper" installer (using [Inno Setup](https://jrsoftware.org/isinfo.php)) will be added Soon™ (in the full v1.1 release, but probably in betas before that to allow testing).

@Miskler, my best guess (assuming you still have your updater issue) is a problem with the SSL certificates on your computer? That's still a pretty wild guess on my end, and the Russian exception text doesn't make it easier for me (I haven't figured out how to get YAMDCC to reliably print exception text in English so that I don't have to rely on Google Translate lol).