Sparronator9999/YAMDCC #72: Gui is not active

anthony-is-dev, 2025-04-04:
### Prerequisites
- [X] I am using the correct [issue template](

[yamdccsvc.exe.log](https://github.com/user-attachments/files/19599720/yamdccsvc.exe.log)

), i.e. I am asking the developer of YAMDCC a question. Bug reports and feature requests should be opened using their respective templates.
- [X] I have read the [FAQ](https://github.com/Sparronator9999/MSIFanControl#faq). Your question might already be answered there.
- [X] I have searched the [issue tracker](https://github.com/Sparronator9999/YAMDCC/issues?q=label:"type:+question") for duplicate or similar questions. Your issue will be closed if it is a duplicate of another issue.
- [ ] I have not answered truthfully to the above questions.

### Question
For some reason, I can't edit the cooler settings through the program, only through xml. What is the reason for this?

Sparronator9999, 2025-04-04:
Did you create a new profile (using the `+` button next to the profile name)?

The default fan profile is locked in the Config Editor, since it's intended to hold your laptop's default fan profile (which can be fetched using `Options` > `Get default fan profile from EC...`). That being said, as you discovered, you can edit it manually using a text editor.

closed 2025-04-04 by Sparronator9999

anthony-is-dev, 2025-04-04:
Thank you, everything works!