Sparronator9999/YAMDCC #74: Switch between integrated graphics and discrete graphics

f0re1gnKey, 2025-04-06:
Will there be a function to switch between integrated graphics and discrete graphics in the future?

Sparronator9999, 2025-04-06:
That depends on if someone has a laptop that supports GPU switching from MSI Center/Feature Manager, and can implement it.

My laptop doesn't appear to support this feature, so I can't implement it for you (sorry).

I can leave this issue open with a "help wanted" tag, though.

f0re1gnKey, 2025-04-06:
> 这取决于某人是否有一台支持从 MSI Center/Feature Manager 切换 GPU 的笔记本电脑，并且可以实现它。
> 
> 我的笔记本电脑似乎不支持此功能，所以我无法为您实现它（抱歉）。
> 
> 不过，我可以用“需要帮助”的标签来保留这个问题。

I have a notebook like this, how can I help you?

f0re1gnKey, 2025-04-06:
> That depends on if someone has a laptop that supports GPU switching from MSI Center/Feature Manager, and can implement it.
> 
> My laptop doesn't appear to support this feature, so I can't implement it for you (sorry).
> 
> I can leave this issue open with a "help wanted" tag, though.

I have a notebook like this, how can I help you?

Sparronator9999, 2025-04-06:
> I have a notebook like this, how can I help you?

First off, does your laptop support changing GPU using MSI Center (or Feature Manager)? I know this seems like a stupid question but better to check lol

For context: Feature Manager is basically MSI Center but without the bloat/tracking. It's an official alternative provided for MSI* laptops sold in a particular country's market that have very strict privacy laws (I can't remember which one off the top of my head). Maybe @porkmanager can tell you more.

f0re1gnKey, 2025-04-06:
> > I have a notebook like this, how can I help you?
> 
> First off, does your laptop support changing GPU using MSI Center (or Feature Manager)? I know this seems like a stupid question but better to check lol
> 
> For context: Feature Manager is basically MSI Center but without the bloat/tracking. It's an official alternative provided for MSI* laptops sold in a particular country's market that have very strict privacy laws (I can't remember which one off the top of my head). Maybe [@porkmanager](https://github.com/porkmanager) can tell you more (EDIT: see [this discussion](https://github.com/Sparronator9999/YAMDCC/discussions/57))

Yes, I confirmed that my laptop supports MSI Center/Future Manager for switching graphics cards. I used to use MSI Center, but later I found it too bloated, so I switched to Future Manager.

Sparronator9999, 2025-04-06:
> Yes, I confirmed that my laptop supports MSI Center/Future Manager for switching graphics cards

Good.

I wrote [this small program]](https://github.com/user-attachments/files/19619784/MsiGpuSwitcher.zip) for someone else during my previous research into GPU switching, can you test it to see if it works, please?

[MsiGpuSwitcher.zip](https://github.com/user-attachments/files/19619784/MsiGpuSwitcher.zip)

f0re1gnKey, 2025-04-06:
> > Yes, I confirmed that my laptop supports MSI Center/Future Manager for switching graphics cards
> 
> Good.
> 
> I wrote [this small program](https://github.com/user-attachments/files/19619784/MsiGpuSwitcher.zip) for someone else during my previous research into GPU switching, can you test it to see if it works, please?
> 
> EDIT: typo

[image]

Sparronator9999, 2025-04-06:
> [image]

Close the message box, right-click on the background of the program, then click `Dump MSI DC UEFI variable`, and post the output.

f0re1gnKey, 2025-04-06:
> > [image]
> 
> Close the message box, right-click on the background of the program, then click `Dump MSI DC UEFI variable`, and post the output.

[image]

Sparronator9999, 2025-04-06:
> [image]
>
> I am currently using discrete graphics

Hmm, interesting that this matches my MsiDcVarData output. Maybe this has nothing to do with UEFI variable modification at all, or I'm missing something...

Give me a bit to try something else.

Sparronator9999, 2025-04-06:
Okay, I think I remember what else Feature Manager was doing behind the scenes (I did a little decompiling to try to figure out the GPU switching).

Can you please post the output of `ec-inspect dump`? (Run this command from within YAMDCC's install directory)

f0re1gnKey, 2025-04-06:
> Okay, I think I remember what else Feature Manager was doing behind the scenes (I did a little decompiling to try to figure out the GPU switching).
> 
> Can you please post the output of `ec-inspect dump`? (Run this command from within YAMDCC's install directory)
> 
> EDIT: make sure you run the Command Prompt as administrator.

sure

[image]

Sparronator9999, 2025-04-06:
Okay, I'm definitely missing something.

Feature Manager writes to both `0xD1` and `0xBE`, but both values in your EC dump appear to be zeroed out.

Can you try switching to a different GPU mode (i.e. integrated only, or hybrid mode), and try again? I will continue to dissect Feature Manager's inner workings in the meantime.

f0re1gnKey, 2025-04-06:
> Okay, I'm definitely missing something.
>
> Feature Manager writes to both `0xD1` and `0xBE`, but both values in your EC dump appear to be zeroed out.
>
> Can you try switching to a different GPU mode [...]

Yes，i switch from discrete to mshybird

[image]

Sparronator9999, 2025-04-06:
Does running `Get-WmiObject -Class MSI_AP` give a valid result? (I ***think*** this is the method that ends up getting called by Feature Manager to query for something. Note that you might need MSI Center (not sure about Feature Manager) already installed for this to work...)

f0re1gnKey, 2025-04-06:
> Does running `Get-WmiObject -Class MSI_AP` in a PowerShell window give a valid result? (I _**think**_ this is the method that ends up getting called by Feature Manager to query for something. Note that you might need MSI Center (not sure about Feature Manager) already installed for this to work...)
> 
> EDIT: if that didn't work, I probably either gave you a wrong command, or the WMI class I'm looking for doesn't exist on your computer...

This class cannot be found when only future manager is installed, but I noticed that there is an MSIAPService process running. I don't want to install msi center again because it is too difficult to clean up lol

Sparronator9999, 2025-04-06:
> This class cannot be found when only future manager is installed, but I noticed that there is an MSIAPService process running

Try `Get-WmiObject -Namespace "root\WMi" -Query "SELECT * FROM MSI_System"`, and see if that returns any classes?

> I don't want to install msi center again because it is too difficult to clean up lol

Fair, lol. Don't worry about re-installing MSI Center.

f0re1gnKey, 2025-04-06:
> Try `Get-WmiObject -Namespace "root\WMi" -Query "SELECT * FROM MSI_System"`, and see if that returns any classes?
>
>
> Fair, lol. Don't worry about re-installing MSI Center.

I reinstalled the msi center, and still got an error when executing these two commands in PowerShell

+ Get-WmiObject -Namespace "root\WMi" -Query "SELECT * FROM MSI_System"
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (:) [Get-WmiObject]，ManagementException
    + FullyQualifiedErrorId : GetWMIManagementException,Microsoft.PowerShell.Commands.GetWmiObjectCommand

Sparronator9999, 2025-04-06:
> still got an error when executing these two commands in PowerShell

Did you run the commands as Administrator? probably should've mentioned that before lol. If it still doesn't work, then I don't currently know how to change the GPU from within Windows.

You should still be able to change the GPU mode from your advanced BIOS settings (`Right Ctrl` + `Right Shift` + `Left Alt` + `F2`). For my laptop (GF63 Thin 11SC), you go to `Advanced` > `System Agent (SA) Configuration` > `Graphics Configuration` > `Primary Display`, then pick `HG` for hybrid graphics, `Intel IGFX` for integrated graphics only, or `PCH PCI` for discrete graphics only.

f0re1gnKey, 2025-04-06:
> > still got an error when executing these two commands in PowerShell
> 
> Did you run the commands as Administrator? probably should've mentioned that before lol. If it still doesn't work, then I don't currently know how to change the GPU from within Windows.
> 
> You should still be able to change the GPU mode from your advanced BIOS settings (`Right Ctrl` + `Right Shift` + `Left Alt` + `F2` after booting into the BIOS using the `Delete` key). For my laptop (GF63 Thin 11SC), you go to `Advanced` > `System Agent (SA) Configuration` > `Graphics Configuration` > `Primary Display`, then pick `HG` for hybrid graphics, `Intel IGFX` for integrated graphics only, or `PCH PCI` for discrete graphics only.

I am sure it is running under an administrator account, because I only use the administrator account. My laptop is Alpha-17-C7VG, which uses an AMD CPU. Currently, there is no option to switch the graphics card in the BIOS, so it must be switched under Windows.

Sparronator9999, 2025-04-06:
> Currently, there is no option to switch the graphics card in the BIOS

Did you try the advanced settings key combo? When in the BIOS, press `Right Ctrl`, `Right Shift`, `Left Alt`, and `F2` (F2 must be the last key pressed) to reveal a *lot* of hidden options. Be careful, you can make your laptop unbootable if you change the wrong setting (ask me how I know). AFAICT, this feature is present on all MSI laptop BIOSes. Then follow the rest of my instructions above (your GPU setting might be in a different location).

f0re1gnKey, 2025-04-06:
> Did you try the advanced settings key combo? When in the BIOS, press `Right Ctrl`, `Right Shift`, `Left Alt`, and `F2` (F2 must be the last key pressed) to reveal a _lot_ of hidden options. Be [...]

I know how to access the advanced settings, and in fact I have already set various advanced settings such as overclocking, undervoltage, memory timing, etc. There is a similar option in AMD CBS to switch graphics cards, but it doesn't work after I set it, so I don't know where to go. (Complaint: AMD's bios contains a lot of bugs)

Sparronator9999, 2025-04-06:
Oh, is your laptop an AMD system? Mine runs on Intel, so yeah, the settings would be different between CPU vendors.

> (Complaint: AMD's bios contains a lot of bugs)

I've heard the AMD GPU drivers contain a lot of bugs as well (ask the Dolphin and Xenia developers, plus probably others).

Peter42R, 2025-05-01:
@Sparronator9999 if you need something from someone with an intel cpu (i9) i can run some scripts or get some data dumps for you also. However i tried running the commands you mentioned in an elevated powershell prompt and it gave error as below:
`PS C:\Windows\system32> Get-WmiObject -Class MSI_AP
Get-WmiObject : Invalid class "MSI_AP"
At line:1 char:1
+ Get-WmiObject -Class MSI_AP
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidType: (:) [Get-WmiObject], ManagementException
    + FullyQualifiedErrorId : GetWMIManagementException,Microsoft.PowerShell.Commands.GetWmiObjectCommand

PS C:\Windows\system32> Get-WmiObject -Namespace "root\WMi" -Query "SELECT * FROM MSI_System"
Get-WmiObject : Not supported
At line:1 char:1
+ Get-WmiObject -Namespace "root\WMi" -Query "SELECT * FROM MSI_System"
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (:) [Get-WmiObject], ManagementException
    + FullyQualifiedErrorId : GetWMIManagementException,Microsoft.PowerShell.Commands.GetWmiObjectCommand`

Peter42R, 2025-05-01:
Hi what do you mean have msi center installed and try again - you mean the powershell commands? Or to just accept i need to use MSI Center to switch the gfx? ;) I DO have msi center installed but set the msi foundation and msi center service to not startup, and instead the YAMDCC starts up at boot, then when i need to switch i have to disable YAMDCC and enable the two msi services, open msi center do the switch and restart.. a bit of a pain but luckily it is not often i need to switch.

Sparronator9999, 2025-05-01:
> Hi what do you mean have msi center installed and try again - you mean the powershell commands?

Oh sorry, yes I meant the PowerShell commands. I don't remember why I suggested them in the first place, either way.

RobaloAmigo, 2025-06-25:
It is a huge loss that this feature is not available in YAMDCC. In fact, it is the only thing I am missing. Mentioned [mini program](https://github.com/Sparronator9999/YAMDCC/issues/74#issuecomment-2781227398) to change the GPU mode unfortunately does not work with the MSI Vector A18 HX (it supposedly switches the mode then tells to restart system, but after restarting the iGPU is still active, even though I selected dGPU only mode). Hope somebody find solution on this finally.

FengXi7420, 2025-08-20:
> It is a huge loss that this feature is not available in YAMDCC. In fact, it is the only thing I am missing. Mentioned [mini [...]

MSI Vector A18 HX(MS-182L) can't disable iGPU completely

RobaloAmigo, 2025-08-20:
> > > > It is a huge loss that this feature is not available in YAMDCC. In fact, it is the only thing I am missing. Mentioned [mini program](https://github.com/Sparronator9999/YAMDCC/issues/74#issuecomment-2781227398) to change the GPU mode unfortunately does not work with the MSI Vector A18 HX (it supposedly switches the mode then tells to restart system, but after restarting the iGPU is still active, even though I selected dGPU only mode). Hope somebody find solution on this finally.
> > > 
> > > 
> > > MSI Vector A18 HX(MS-182L) can't disable iGPU completely
> > 
> > 
> > My mistake, I meant Vector 18HX (Intel based).
> 
> Still the same it you disabled it completely you will encounter bitlocker and pin problems

I have 100% of time usage of dGPU with external monitor connected and not single second usage of iGPU so I really have no idea what are you writing about, maybe some kind of other way of disabling iGPU that I am not aware of, anyway I mean MUX switch and it works completely fine on my Vector 18HX and this is what I would like to achieve with YAMDCC (same thing what MSI Center does). Nothing special I would say.

FengXi7420, 2025-08-20:
> I have 100% of time usage of dGPU with external monitor connected and not single second usage of iGPU so I really have no idea what are you writing about, maybe some kind of other way of disabling [...]

OK, go to BIOS, unlock hidden menus, PCH-IO, Graphics configuration, Disable internal graphics