Sparronator9999/YAMDCC #76: Fan speed does not decrease when temperature drop

Peter42R, 2025-04-08:
### Prerequisites

- [x] This issue is not a [bug report](https://github.com/Sparronator9999/YAMDCC/issues/new?template=bug_report.md) or [feature request](https://github.com/Sparronator9999/YAMDCC/issues/new?template=feature_request.md).
- [x] I have searched the [FAQ](https://github.com/Sparronator9999/YAMDCC#faq) for answers.
- [x] I have searched the [issue tracker](https://github.com/Sparronator9999/YAMDCC/issues?q=label:"type:+question") for duplicate or similar questions.
- [x] I have not answered the above questions truthfully.

### Question

It seems like CPU fan speed does not drop when the temperature drops. I have attached my config, i can create a new fresh one from EC if that would help. Right now my fan speed is stuck at 75% for cpu as can be seen, and when i change the 75-78 degrees segment the fan speed adjusts according to that.

Could you please assist me with what may be going wrong here (i have disabled the Windows Defender feature (Memory integrity) but not uninstalled the feature thouhg - I am not seeing any errors either, it is scaling up, just not down it seems like. It might be scaling too far up somehow.

I have attached some images (as seen the gpu looks fine, but I did not have any load on that right now either).

Best regards and hope someone can help!

[Vector-16-HX-AI-A2XWIG.txt](https://github.com/user-attachments/files/19646112/Vector-16-HX-AI-A2XWIG.txt)

[image]
[image]

Peter42R, 2025-04-08:
Hi like this?

[image]

YAMDCC EC inspector

 0x | 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
-----|----------------------------------------------
 00 | 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
 10 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
 20 | 00 00 00 00 00 00 00 00 0A 05 00 00 00 24 5B 5B  .............$[[
 30 | 03 09 00 0D 01 00 50 81 2B 16 28 3C 37 02 E0 00  ......P.+.(<7...
 40 | 00 00 5E 00 9A 14 00 00 40 13 4E 42 BE 0B 00 00  ..^.....@.NB....
 50 | 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00  ................
 60 | 00 00 00 00 24 00 30 10 2E 00 37 3E 45 4B 51 57  ....$.0...7>EKQW
 70 | 69 28 00 28 32 3C 4B 59 67 00 28 31 38 3E 44 4A  i(.(2<KYg.(18>DJ
 80 | 28 00 32 37 3C 41 46 4B 61 00 00 37 3C 46 50 5D  (.27<AFKa..7<FP]
 90 | 70 00 2B 30 35 3A 3F 44 02 00 00 02 00 00 3C 00  p.+05:?D......<.
 A0 | 31 35 4D 33 45 4D 53 31 2E 31 30 35 30 32 32 37  15M3EMS1.1050227
 B0 | 32 30 32 35 30 39 3A 33 90 3A 34 36 00 00 00 08  202509:3.:46....
 C0 | 00 00 07 24 00 00 A6 00 00 C6 00 C8 00 00 00 00  ...$............
 D0 | 00 00 C1 80 0D 00 05 D0 00 01 00 00 00 07 00 00  ................
 E0 | E2 00 00 9A 14 00 00 C1 00 00 00 00 00 C2 00 00  ................
 F0 | 00 00 70 00 26 7F 05 2A 7F 05 00 00 00 01 00 00  ..p.&..*........
C:\Program Files\YAMDCC>

Peter42R, 2025-04-08:
Now after some time the fan stopped or atleast slowed down a lot (YAMDCC is not running). And the F0 line changed a bit if that does any difference ;)

[image]

Sparronator9999, 2025-04-08:
> Hi like this?

Yep, just like that. However, if you copy and paste the command line output (like you've done), can you put it in a fenced code block next time? It looks something like this:

```cs
Console.WriteLine("Use three backticks (```) to start and end a fenced code block.");
```

Anyways, back to the EC dump you provided.

Congratulations, you're the first YAMDCC user to have a laptop store the temperature down thresholds in a different way to literally every other YAMDCC-supported laptop.

Basically, every MSI laptop except yours (and probably others released around the same time) store the temperature down threshold as an offset of the up threshold. For my laptop, this is 3°C below each up threshold by default (e.g. a 50°C up threshold would have a calculated down threshold of 47°C using the above offset).

Yours, on the other hand, stores *absolute* down threshold values, and in fact there appears to be quite a large gap between up and down threshold for your laptop. Refer to [this diagram](https://github.com/Sparronator9999/YAMDCC/blob/main/Media/MSI-GF63-Thin-11SC-EC.png) I created for an EC research doc.

Or I could be completely wrong. Either way, I will write a patch for you to test at some point.

Sparronator9999, 2025-04-08:
> Now after some time the fan stopped or atleast slowed down a lot (YAMDCC is not running). And the F0 line changed a bit if that does any difference ;)

The F0 line is not used by YAMDCC at all currently, but then again your laptop appears to operate quite differently from any other one that's used this project.

Sparronator9999, 2025-04-08:
> Would it help if I supplied a clean EC loaded config for you that I have not modified myself

Please.

Not sure if it will matter in the end, but it might help having a reference config.

Peter42R, 2025-04-08:
It is attached here as you can see the values for DownThreshold are static values. So it makes sense what you say that speed will drop when it changes with this amount i guess.

[Vector-16-HX-AI-A2XWIG-Clean.txt](https://github.com/user-attachments/files/19646917/Vector-16-HX-AI-A2XWIG-Clean.txt)

Sparronator9999, 2025-04-08:
It's funny you post your config now, I've just pushed a new commit to try to resolve this issue.

Can you please add the following line to your config:

`<OffsetDT>true</OffsetDT>`

below the `FirmDate` tag? Then try running EC-to-config again to get new down threshold values.

While you're at it, can you make sure the other features work? The keyboard backlight slider is a known issue on newer laptops with RGB keyboards, but the other options might have been broken with your new laptop as well.

Peter42R, 2025-04-08:
[Vector-16-HX-AI-A2XWIG-CleanV2.txt](https://github.com/user-attachments/files/19647379/Vector-16-HX-AI-A2XWIG-CleanV2.txt)

Indeed correct it does not allow upload of xml 😄 

I added the following as you mentioned:
```
	<FirmDate>0001-01-01T00:00:00</FirmDate>
	<OffsetDT>false</OffsetDT>
	<FanConfs>
``` 

I then loaded the config into YAMDCC and applied profiles on both cpu and gpu, i then went to options and get default fan profile from EC and it gave me the attached txt file (without OffsetDT).

Come to think of it i did not install the new version you mention before doing this? should I have done that? :D 

Also how do I test the other functions, i can change the charge limit and performance mode but where do i see the impact of this to verify that it actually works? :)

Best regards

Peter42R, 2025-04-08:
Okays so... uninstalled the service and wanted to install the new one you link to - that failed with an error when trying to install it (maybe i need to restart after uninstalling to be able to install it again?) oh well.. I then reinstalled the debug version (also without restart) that worked. 

I loaded the config and tried to get from EC and it gave me this:

[Vector-16-HX-AI-A2XWIG-CleanV3.txt](https://github.com/user-attachments/files/19647824/Vector-16-HX-AI-A2XWIG-CleanV3.txt)
So now it kept the OffsetDT which i guess is progress ;)

I will test the battery thing now and get back about that!

Best regards

Peter42R, 2025-04-08:
I can confirm the battery charge limit works as well as the fn+win switch. 

It looks like the profile is also working with decreasing fan speed as temperature drops now :D 

I have yet to test the performance modes I will get back when i have if i can figure out a way to test it, but if you say it limits fan speed I might try to start a game and get fans up and running at high speed and then set it to silent if that should cap fans to 3000 :)

Best regards

Sparronator9999, 2025-04-08:
Your new config looks much more sane than previous ones, so that's an improvement (I still think ~12°C is too much of a gap between up and down threshold).

Can you confirm (using EC monitor) that the fan speed changes according to the temperatures as displayed in YAMDCC?

Peter42R, 2025-04-09:
Hi, yes i can confirm span spins up and down according to the ec monitor values now - this is my current config and it looks to be following the numbers defined here:

[image]
[image]

Sparronator9999, 2025-04-09:
> Hi, yes i can confirm span spins up and down according to the ec monitor values now

Good, it looks like this issue can be closed now.

One last thing: can you please open a [pull request](https://github.com/Sparronator9999/YAMDCC/pulls) with your current config? This is so we can add your laptop to the supported laptop list.

Peter42R, 2025-04-09:
It looks like the performance mode changes between balanced silent ect. does not do anything at all - I have been trying to get it to react to those changing but i cannot register any changes by going from silent to high performance. No fps gain in games, no changes in fan spin that i can see.

porkmanager, 2025-04-09:
> balanced silent ect. does not do anything at all

'balanced' is the same as 'high performance'. Silent mode just limits the fan's RPM to 3.6k (only for the 'default' fan profile; it does not affect customized profiles). 'Max battery' mode limits the CPU TDP to 10W (any fan profile) and probably has some fan RPM limit on the default fan profile. So, most of those modes are just MSI nonsense and little useful for the default fan profile. MSI Center does the same things.
Only 'Max battery' has sense with yamdcc on my opinion.

Sparronator9999, 2025-04-09:
Your performance mode tests match my experiences, i.e. they don't do much apart from limiting CPU power (which can be done with ThrottleStop's Turbo Power Limit settings) and fan speed (which can be unintuitive for new users, hence why Balanced is set as the default setting).

That being said, MSI Center also auto-disables the keyboard backlight (likely through software) when set to Super Battery (YAMDCC calls this setting "Maximum battery life").

porkmanager, 2025-04-10:
> EDIT: the reason why these four options are even present is because they are present in the official MSI Center as (if I remember correctly) User Scenarios.

That is good that those modes available in yamdcc, because if someone just wants the basic functionality of MSI Center, they can simply use the default fan profile and its built-in modes. Essentially, it's like MSI Center without the bloatware. The main differences, as you mentioned, are things like the keyboard backlight turning off in maximum battery mode, and the fact that MSI Center's performance mode allows for tweaking fan curves.

porkmanager, 2025-04-10:
> However when i switch profile in YAMDCC to max battery or silent i do not see any fps drop. Maybe because i am running my custome "Casual" profile as in the images above?

Max. battery mode should limit cpu power in the any fan profile. Probably some changes in the newer laptop's EC.

https://github.com/user-attachments/assets/9723e8dc-4e86-4630-ac2e-03df49c8823b

Sparronator9999, 2025-04-29:
@Peter42R, I just realised I haven't added your laptop's config to YAMDCC. Can you provide your up-to-date config in either a pull request or as a reply to this comment (using a .txt extension is fine)?

I plan to release a beta soon, and realised I didn't have a config for the new "OffsetDT" feature...

Peter42R, 2025-04-30:
Hi @Sparronator9999 sorry for the large delay, I ran into some vacation and other stuff.. I have run the same test as you and can confirm the CPU power does drop for me also when stting maximum battery life. However the other settings (silent, balanced) does not seem to do much.

The MSI Center has the option to switch between discreet and mshybrid gfx - i am guessing that this is not something that is easily implemented right? (it also requires a reboot)

Attached is my current config - if the values makes sense or allow for the pc to run too hot I am not too sure but here it is atleast ;)

[Vector-16-HX-AI-A2XWIG.txt](https://github.com/user-attachments/files/19975372/Vector-16-HX-AI-A2XWIG.txt)

Sparronator9999, 2025-04-30:
Thanks for the config, I will add it soon.

> The MSI Center has the option to switch between discreet and mshybrid gfx - i am guessing that this is not something that is easily implemented right? (it also requires a reboot)

Yeah, I haven't figured this out yet. There is an open issue for this feature though: [#74](https://github.com/Sparronator9999/YAMDCC/issues/74)