Sparronator9999/YAMDCC #78: [Question] Charging start threshold and stop threshold

f0re1gnKey, 2025-04-17:
When setting the charging value in msi center, there will be a start threshold and a stop threshold, for example (50%-60%), but yamdcc only has a stop threshold. What is the start threshold?

Sparronator9999, 2025-04-17:
The start threshold for the charging value will always be 10% lower than the stop threshold set in YAMDCC (except for 0/disabled, which will always charge to 100%). This is hardcoded on all* MSI laptops that YAMDCC supports (*AFAIK).

I thought I explained this in the corresponding tooltip, but checking it again I didn't (for some reason).

closed 2025-04-17 by Sparronator9999

f0re1gnKey, 2025-04-17:
> The start threshold for the charging value will always be 10% lower than the stop threshold set in YAMDCC (except for 0/disabled, which will always charge to 100%). This is hardcoded on all* MSI [...]

Thank you for your answer