Sparronator9999/YAMDCC #79: Crash Due to Default Fan Setting at 110%

gemSquared, 2025-04-20:
### Prerequisites

- [x] This issue is not a [feature request](https://github.com/Sparronator9999/YAMDCC/issues/new?template=feature_request.md) or [bug report](https://github.com/Sparronator9999/YAMDCC/issues/new?template=question.md).
- [x] I have searched the [issue tracker](https://github.com/Sparronator9999/YAMDCC/issues?q=label:"type:+bug") for duplicate or similar bug reports.
- [ ] I have not answered the above questions truthfully.

### Describe the bug

GS76 Stealth 11UG

Crash when loading default profile and viewing GPU fan settings.

Most likely because some reason the laptop lets you go past 100% all the way up to 150%
```
<TempThreshold>
<UpThreshold>87</UpThreshold>
<DownThreshold>82</DownThreshold>
<FanSpeed>110</FanSpeed>
</TempThreshold>
```

### To Reproduce

1. GS76 Stealth 11UG Laptop
2. Options > Get default fan profile from EC...
3. Reboot
4. Profile is saved (I have the XML if you wish)
5. Profile is loaded to YAMDCC
6. Fan Control > Fan > Dropdown Select GPU
7. Crash while trying to display GPU

### Expected behavior

It probably should've showed the GPU fan curve.

### Additional context

Crash report:
```
Version: 1.1.0-beta.4
Revision (git): c8fdcf56b4e6b63ed020d4cba5c032b48473ece5
Crash report:
System.ArgumentOutOfRangeException: Value of '110' is not valid for 'Value'. 'Value' should be between 'Minimum' and 'Maximum'.
Parameter name: Value
   at System.Windows.Forms.TrackBar.set_Value(Int32 value)
   at YAMDCC.ConfigEditor.MainForm.ProfSelChanged(Object sender, EventArgs e)
   at System.Windows.Forms.ComboBox.OnSelectedIndexChanged(EventArgs e)
   at System.Windows.Forms.ComboBox.set_SelectedIndex(Int32 value)
   at YAMDCC.ConfigEditor.MainForm.FanSelChanged(Object sender, EventArgs e)
   at System.Windows.Forms.ComboBox.OnSelectedIndexChanged(EventArgs e)
   at System.Windows.Forms.ComboBox.WmReflectCommand(Message& m)
   at System.Windows.Forms.ComboBox.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
```

Attempts at slapping duct tape at the problem:
- Opening the XML and replacing `<FanSpeed>110</FanSpeed>` with `<FanSpeed>100</FanSpeed>`
  - Crash has been avoided.

P.S. Laptop has a 3rd smaller fan that that kicks on at 66C GPU and is either 95% or off. Would like to see what can be played with.

gemSquared, 2025-04-20:
Found the max value in the xml. Set to 150 as well as per other closed posts. Would you like my config or shall we try to figure out that 3rd fan?

Sparronator9999, 2025-04-20:
Looks like potentially a UI bug causing that crash. Investigating that now.

As for the third fan, I can provide the following information on how the MSI fan profiles work:

My MSI laptop (and likely others) only have one physical fan, but can set the fan speed according to the higher of the CPU and GPU fan profile.

For laptops with two fans and a discrete GPU (NVIDIA/AMD, not the built-in Intel iGPU for example), I *think* the CPU fan profile applies to the CPU fan, and the GPU fan profile to the GPU fan, but I'd need someone who has one of these laptops to check for me.

I have no idea how three-fan MSI laptops work, but you can provide an EC dump using `ec-inspect dump` in an Administrator Command Prompt `cd`'d into the YAMDCC install directory, and I'll see if I can find anything.

Sparronator9999, 2025-04-20:
Can you please also provide your generated config XML?

I can't reproduce your exact bug, however I am getting an `IndexOutOfRangeException` when the `MaxSpeed` values differ between fan configs (different to fan *profile* configs).

gemSquared, 2025-04-20:
[MSI-Stealth-GS76-11UG_150.txt](https://github.com/user-attachments/files/19823751/MSI-Stealth-GS76-11UG_150.txt)

Didn't let me do an xml. Put a txt instead. This one I set max fan speed to 150 for both CPU and GPU, but it was 100 fresh after the reboot. Doesn't have crashing errors anymore.

gemSquared, 2025-04-20:
```YAMDCC EC inspector

 0x | 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
-----|----------------------------------------------
 00 | 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
 10 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
 20 | 00 00 00 00 00 00 00 00 0A 05 00 00 00 04 19 4B  ...............K
 30 | 03 09 00 0D 01 00 50 81 6A 18 60 3B 71 02 E0 00  ......P.j.`;q...
 40 | 00 00 5E 00 5A 12 00 00 14 11 C2 3E 03 0C 00 00  ..^.Z......>....
 50 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
 60 | 00 00 00 00 00 00 00 00 49 00 34 3C 41 4B 52 58  ........I.4<AKRX
 70 | 64 00 00 00 00 00 4B 4B 00 50 07 03 03 03 03 03  d.....KK.P......
 80 | 3B 00 37 3D 41 47 4D 57 64 00 00 00 00 00 00 00  ;.7=AGMWd.......
 90 | 00 5F 08 03 03 03 03 05 82 0F 78 06 0F 78 3A 00  ._........x..x:.
 A0 | 31 37 4D 31 45 4D 53 31 2E 31 31 33 30 34 32 32  17M1EMS1.1130422
 B0 | 32 30 32 32 31 35 3A 31 33 3A 33 31 00 00 00 20  202215:13:31...
 C0 | 00 00 07 24 00 00 00 00 00 45 00 41 00 41 00 00  ...$.....E.A.A..
 D0 | 00 00 C0 82 8D 00 05 80 00 03 00 00 00 00 00 00  ................
 E0 | E2 00 00 5A 12 00 00 C1 10 00 00 00 00 C0 00 80  ...Z............
 F0 | 40 00 70 00 3C 44 3A 3B 41 00 00 00 00 03 02 00  @.p.<D:;A.......```

Here's the EC dump. You can see the 3rd fan RPM at `0xCD`

gemSquared, 2025-04-20:
>  I think the CPU fan profile applies to the CPU fan, and the GPU fan profile to the GPU fan, but I'd need someone who has one of these laptops to check for me

Like they correlate to each other? Mine's got a mobile RTX3070. Seems like they do for the most part. GPU Fan spools up override to at least 70% when CPU gets hot though.

Trying to find what temp it kicks on to help the CPU Fan. Current guess is 75C for my laptop, but hard to find on the dot because my CPU temps are real jumpy.

Sparronator9999, 2025-04-20:
> Didn't let me do an xml. Put a txt instead.

I've noticed that with GitHub recently. I might start putting in the issue templates and my replies to upload the config as a zip file (but txt files are OK as well). Will investigate.

As for the third fan, it doesn't appear to have its own fan profile in EC, and probably won't show its RPM in YAMDCC, but you can still see it using a tool like HWiNFO (I think they show MSI laptop fan RPMs). I plan to rework how fan RPMs are shown at some point anyways, because I made an assumption that the two fan profiles apply only to their respective fan, when that doesn't appear to be the case.

> my CPU temps are real jumpy

My laptop doesn't keep its own CPU cool enough to run an up to ~2-3 core Prime95 stress test (it immediately thermal throttles for one or two cores, then by the third or fourth core it hits the 45W turbo power limit).

With that useless fact out of the way, I've noticed my CPU temps jump all over the place when I disable turbo as well, even after cleaning out the single fan for both CPU and GPU (you'd be surprised by how much dust builds up over the 3 years I've had this laptop) and repasting the CPU and GPU (you need to anyways if you're taking apart the fan, due to a stupid design choice by MSI).

gemSquared, 2025-04-20:
I haven't touched the thermal paste. But can confirm needs frequent air-dust on the vents every month or so. And the heat made the hinge plastic brittle (so the stereotypical hinge explosion)

Anyway, you think I can help testing anything else? That config I posted not sure where the Aggressive profile came from. I just picked one of the provided ones and did the EC default reboot. You can use it if you like I don't mind 👍

Sparronator9999, 2025-04-20:
I still can't reproduce your bug with your config, however I *think* I know what's going on.

There appears to be a bug in YAMDCC causing a crash when loading a config with a lower `MaxSpeed` than the old config, caused by a fan speed value from the old config being higher than the new config's `MaxSpeed`.

I will try to fix this bug Soon (TM).

> That config I posted not sure where the Aggressive profile came from

It looks like you copied another laptop's config, specifically `MSI-Titan-GT77HX-13VH.xml`. Maybe start from one of the generic configs next time, however that laptop appears to be the same EC generation anyways.

I don't know where the Aggressive fan profile came from either, maybe the uploader came up with it.

> (so the stereotypical hinge explosion)

I wish I knew that before buying this laptop, however I haven't had a problem with that surprisingly. That being said, the corner with the single fan has cracked plastic so that the screw for that corner doesn't actually do anything, and is held on with black duct tape lol (same as the laptop colour)

gemSquared, 2025-04-20:
Oh I already "fixed" it when I uploaded hence the `-150`.

Here's what it was sorry 

[MSI-Stealth-GS76-11UG-100.txt](https://github.com/user-attachments/files/19823854/MSI-Stealth-GS76-11UG-100.txt)

Sparronator9999, 2025-04-20:
Oh, that's your problem. Some configs have a `MaxSpeed` value of 100%, and some have 150% instead. I can't remember exactly which configs, but there appears to be no check on the EC-to-config feature to make sure the `MaxSpeed` is high enough to accommodate all fan speeds (I probably assumed they'd all be within the default 100%). This causes the crash you see because (obviously) 110% > 100%, and trying to set a TrackBar (the speed slider you see in the config editor) to a value higher than its maximum value will cause the exception you see.

This should've been caught by the config validation that occurs when loading the YAMDCC config, but evidently this seems to be missing or broken for some reason. Add that to the list of stuff to fix.

gemSquared, 2025-04-20:
Ah. Okay, well otherwise tool's working great so far! Thanks much!

Sparronator9999, 2025-04-20:
No problem! I've already fixed the config validator, just need to fix EC-to-config and the inconsistent MaxSpeed crash, then I will close this issue.

Sparronator9999, 2025-04-20:
All bugs should be fixed in the latest commits now, can you please test it? You may need to use a config with a MaxSpeed of 150% to test the EC-to-config feature (all generic configs are configured this way, so use one of those when testing).

One final request: can you open a pull request with your new config when you're done testing? This allows me to add your laptop to the supported laptop list.

gemSquared, 2025-04-20:
Not seeing the commit yet, but made the pull request (sorry took a while I'm a bit rusty with GIT haha)

Also this other issue seems to have this problem https://github.com/Sparronator9999/YAMDCC/issues/77

103%

Sparronator9999, 2025-04-29:
Closing since this issue was fixed with my latest commits, plus the config added in [#80](https://github.com/Sparronator9999/YAMDCC/pull/80).

closed 2025-04-29 by Sparronator9999