Sparronator9999/YAMDCC #83: installer issues

porkmanager, 2025-05-04:
I used this version https://github.com/Sparronator9999/YAMDCC/commit/be95771354f9be7b84b9366f761f4294e439e8f6
And got few issues:
1. Installer doesnt unpacked configs folder
2. I choosed Portable mode without crating shortcuts but shortcuts was created

Could we have classic archive portable versions? I hate any kind of installers especially those installers what i cant unpack using upx tools XD This one I cant open in the total commander with pressing ctrl+pgdn combo.

Pined installer log, I used side app for monitoring installer behavior.
[yamdccInstaller.log](https://github.com/user-attachments/files/20026851/yamdccInstaller.log)

Sparronator9999, 2025-05-04:
> Installer doesnt unpacked configs folder

YAMDCC no longer comes bundled with configs, you're expected to download the one for your laptop (or a generic one) from the "Configs" folder on GitHub.

> I choosed Portable mode without crating shortcuts but shortcuts was created

The shortcuts being created in portable mode shouldn't be happening though. That will be fixed soon.

> Could we have classic archive portable versions? I hate any kind of installers especially those installers what i cant unpack using upx tools XD This one I cant open in the total commander with pressing ctrl+pgdn combo.

I only did the installer in the first place because someone requested it (I think multiple people did, in fact), and it makes for a nicer setup experience. Also, I seem to remember you used to be able to extract Inno Setup installers using 7-Zip, but newer versions seem to not work.

I can bring back the old zip files, but that could cause undefined behaviour with the YAMDCC updater, plus more work for me when creating the installer.

porkmanager, 2025-05-04:
> YAMDCC no longer comes bundled with configs, you're expected to download the one for your laptop (or a generic one) from the "Configs" folder on GitHub.

Seems i missed this innovation while read changelog.

> I only did the installer in the first place because someone requested it (I think multiple people did, in fact), and it makes for a nicer setup experience. Also, I seem to remember you used to be able to extract Inno Setup installers using 7-Zip, but newer versions seem to not work.

Yeah, i think it would be useful for casual users, as i know most of users like to install something on their OS, but it still it couldnt help with configs for those casual users. Most frequency question what i have seen on the different sources where yamdcc was discussed its: "there are no my laptops model in the supported list, so i will not use it".

> I can bring back the old zip files, but that could cause undefined behaviour with the YAMDCC updater, plus more work for me when creating the installer.

nvm, unpacking portable version with the installer is enough.

Sparronator9999, 2025-05-04:
> Yeah, i think it would be useful for casual users, as i know most of users like to install something on their OS, but it still it couldnt help with configs for those casual users. Most frequency [...]

Huh, that's quite insightful. As mentioned in the planned features, I plan to get rid of the EC register-based configs and just have a CPU and GPU fan profile (along with all the other extra options), but this would become a v2.0 (since it would break literally every config ever created for YAMDCC so far). Instead, the user would either be asked when their laptop was made, or I could try to do some automatic EC generation detection if I can find a reliable method to do so.

> nvm, unpacking portable version with the installer is enough.

I didn't get a chance to edit my reply to say this, but I think I will compromise and re-instate ZIP uploads to GitHub Actions (but not Releases), without the Updater program (since running that would run the latest installer silently and possibly mess up an existing portable installation).

The Installer is currently being improved to show the "portable mode" question before selecting the extraction location (sort of similar to Winaero Tweaker's installer), and to allow disabling installation of selected components (this seems to break the installation size requirements, however).

Sparronator9999, 2025-05-05:
Oh, I know this one.

The Config Editor is probably trying to call the Updater to disable automatic updates, but since the Updater has been removed from portable versions and made optional for the installed version, it crashes instead if it's not present. This'll be fixed soon enough.

Sparronator9999, 2025-05-05:
Fixed in commit `68a66d9`.

closed 2025-05-05 by Sparronator9999

porkmanager, 2025-05-06:
xd. it is still trying to remove updater

[image]

Sparronator9999, 2025-05-06:
Oops.

I meant to change that to silently skip removing the updater task, but that's what happens when you copy and paste the code from the two other places that call the updater (to check for updates, and to enable automatic update checking on the first run of YAMDCC).