Sparronator9999/YAMDCC #84: Hello from msi-ec

glpnk, 2025-05-05:
We have bunch of research data which might help you:

+ https://github.com/BeardOverflow/msi-ec/discussions/98
+ https://github.com/BeardOverflow/msi-ec/blob/main/docs/device_support_guide.md
+ https://github.com/BeardOverflow/msi-ec/discussions/130

List of supported devices with linked issues:
+ https://github.com/BeardOverflow/msi-ec/discussions/277

glpnk, 2025-05-05:
For majority of devices you support you can get rid of kernel driver and use only WMI, but only for Generation 2 devices, which MSI call WMI2

Sparronator9999, 2025-05-05:
Oh, this was a bit unexpected. Hello!

I actually already had a look at the MSI EC project before, that's how I implemented the Fan Mode feature. But thanks for the links, I will be looking at them again at some point.

> For majority of devices you support you can get rid of kernel driver and use only WMI, but only for Generation 2 devices, which MSI call WMI2

I have absolutely no idea how to call WMI for MSI laptops, or if the temperature thresholds are even supported via WMI. If you have more information, please reply.

glpnk, 2025-05-05:
WMI is mapped to next addresses (check pics). Map is sourced from DSDT tables of some devices. For calling WMI on MSI laptops you need to install `.mof` file, which maps ACPI methods to WMI space. Often this file is built in ACPI tables (DSDT), but MSI share it in drivers. You can find decompiled version in 98 issue.

Sadly, not all addresses are mapped on older WMI1 devices. 

| [image] | [image] |
|-|-|

Also:
+ Linux driver for `lm-sensors` which reads RPM on WMI2 devices https://github.com/torvalds/linux/blob/master/drivers/platform/x86/msi-wmi-platform.c

glpnk, 2025-05-05:
As Linux GUI we use MControlCenter, which is currently being updated https://github.com/dmitry-s93/MControlCenter

glpnk, 2025-05-05:
About battery issues you've mentioned - not sure what you trying to write, or what is really being written, because last time I experimented with Windows, I had ability to only write 4 byte blocks, which is dumb and unsafe.

Sparronator9999, 2025-05-05:
What I meant is that I think my EC access conflicts with the OS's EC access (i.e. a race condition). Basically if my service accesses the EC at the same time as the OS does, my service may end up inadvertently writing to the wrong EC register, for example. But that's just a theory. I agree that is is probably unsafe, but it's the only way I can achieve EC access vure

glpnk, 2025-05-05:
Laptops have keyboard backlight register, and devices with RGB keyboards should have disabled backlight regulator logic.

For long time I had question - why this happens. But then I discovered that few devices under same EC name might have single- and RGB colored backlight.

For mute leds - you probably can subscribe to Windows audio stack issues and write to led bits. On Linux we have problem, because audio stack often don't mark sound device as primary, so kernel don't monitor it's state.

glpnk, 2025-05-05:
Have you seen reddit post with "check backdoor" function? EC is accessible not only through EC calls, but it's mapped to system memory space, which you can read on Linux and Windows (highly unsafe, but used you know where).

glpnk, 2025-05-05:
If you've used ImHex you can try to use my patterns, which maps dump to WMI-like arrays https://github.com/glpnk/hexpats

glpnk, 2025-05-05:
Would be fun if you implement WMI events. You need to enable events with some bit and subscribe to events class

Sparronator9999, 2025-05-05:
Okay, sorry for the delay in replying. This will be a long one.

> WMI is mapped to next addresses (check pics)

Do you know which laptops use WMI1 specifically? Is it all the old pre-10th-gen Intel laptops, or even older than that?

> As Linux GUI we use MControlCenter, which is currently being updated https://github.com/dmitry-s93/MControlCenter

I have heard of this project before (it's even linked somewhere as an alternative for Linux users).

> Laptops have keyboard backlight register, and devices with RGB keyboards should have disabled backlight regulator logic.

Yes, this is how I control keyboard backlight through YAMDCC. Currently I don't detect which laptops have this functionality (well, I *sort of* do, but it could be better), but it appears that the MSB of register `0xD3` is always set for laptops that have monochrome keyboard backlights, but isn't for laptops with RGB backlights (and possibly ones without a backlight at all, if they somehow exist).

> For mute leds - you probably can subscribe to Windows audio stack issues and write to led bits. On Linux we have problem, because audio stack often don't mark sound device as primary, so kernel don't monitor it's state.

I can look into this at some point if I have time.

> Have you seen reddit post with "check backdoor" function? EC is accessible not only through EC calls, but it's mapped to system memory space, which you can read/write on Linux and Windows

I have not heard of this function, and I've decompiled Feature Manager (basically MSI Center for the Turkish, who have stricter privacy/data collection laws) to try to understand the GPU switch feature.

> (highly unsafe, but used you know where).

I don't.

(...you're not talking about YAMDCC, are you??)

> If you've used ImHex you can try to use my patterns, which maps dump to WMI-like arrays https://github.com/glpnk/hexpats

I haven't used ImHex (HxD is my current hex editor).

> Would be fun if you implement WMI events. You need to enable events with some bit and subscribe to events class

Not sure what they would be useful for (note to self: look at the WMI mappings again lol)

Sparronator9999, 2025-05-05:
Okay, I had a look at the full decompiled MOF table. Assuming I can get WMI access working, would it be OK for me to incorporate those MOF definitions into YAMDCC directly? I can give @Wer-Wolf and/or the msi-ec team a shoutout in a special thanks section of some description (README/in-app?)

Also, do those MOF definitions work on both WMI1 and WMI2 laptops? Or do I need a separate one for WMI1 and WMI2 laptops?

Sparronator9999, 2025-05-05:
I'm going to convert this issue to a discussion, since that seems like a better fit for this topic.