Sparronator9999/YAMDCC #88: Fan gets stuck at the speed it was at when sleeping/hibernating, does not follow custom fan curve

ThreeScreenStudios, 2025-05-15:
### Prerequisites

- [ ] This issue is not a [feature request](https://github.com/Sparronator9999/YAMDCC/issues/new?template=feature_request.md) or [bug report](https://github.com/Sparronator9999/YAMDCC/issues/new?template=question.md).
- [x] I have searched the [issue tracker](https://github.com/Sparronator9999/YAMDCC/issues?q=label:"type:+bug") for duplicate or similar bug reports.
- [ ] I have not answered the above questions truthfully.

### Describe the bug

After installing YAMDCC, and setting a fan curve, the service works appropriately and scales up and down the fan speed based on the custom temp/speed settings. However, if the computer goes to sleep and then hibernates, it appears that the fan speed gets stuck on whatever the speed was prior to sleep/hibernation. E.g. even if the temp is at a different threshold, the fan speed will not change.

Re-applying the curve does not fix the issue, however changing the Profile on the dropdown to Default, and then back to Custom will make the fan curves work appropriately again.

### To Reproduce

1. Set a custom fan speed curve, with Profile as Custom
2. Run a benchmark or game, the fans respond as set by the curve
3. Allow computer to fall asleep and then hibernate
4. Wake the computer
5. Re-run a benchmark or game, note that the fan speed is fixed
6. Change profile in YAMDCC UI to Default, Apply, then back to Custom, and Apply.
7. Notice the fan curve works as expected again

### Expected behaviour

The set custom fan curves should remain through sleep and hibernation.

ThreeScreenStudios, 2025-05-15:
Thanks for the quick reply! Yep - laptop is a `MSI Stealth 18 AI Studio A1VHG`.

> Are you talking about the (if I remember correctly) 3-hour timeout after entering sleep, before the computer goes into hibernation? Have you tried reproducing the bug by entering hibernation directly?

Yes I believe that's it - for some reason I can't get `Hibernate` to come up as an explicit option on the start menu, but let me see if I can invoke it somehow and will get back to you.

> Okay, that's weird. Clicking Apply will always re-write the fan profiles and settings, even if nothing was changed.

Agreed, it is very odd - I just happened to be playing around with settings to see if anything could change it and stumbled upon the profile switch working. I'll see if I can get any further details for you. If I click `Apply`, It does say that the fan curve has applied successfully, but the fan remains stuck at the speed at sleep/hibernation.

Also wanted to say great work on the app! So much better than MSI Center, and it's nice to be able to set a custom fan curve independently, as well as battery charge percentage :)

Sparronator9999, 2025-05-15:
Oh wait, I figured it out.

Try loading `MSI-Vector-16-HX-AI-A2XWIG.xml`, then re-run EC-to-config. This laptop had a problem with the down thresholds recently that caused the fan speed to get stuck on a high speed.

Alternatively, you can open your current config in an XML editor (located at `C:\ProgramData\Sparronator9999\YAMDCC\CurrentConfig.xml`) and change the `<OffsetDT>` option from `true` (the default) to `false`, then restart the YAMDCC service (or close and re-open the YAMDCC Config Editor, then click `Apply`).

Sparronator9999, 2025-05-15:
> Also wanted to say great work on the app! So much better than MSI Center, and it's nice to be able to set a custom fan curve independently, as well as battery charge percentage :)

Why, thank you very much!

Don't forget the multi-fan-profile support, which MSI Center (as of last time I used it) doesn't have (you can only set one custom fan profile, and even then I don't think it remembers it if you switch back to Auto).

ThreeScreenStudios, 2025-05-16:
> Alternatively, you can open your current config in an XML editor (located at C:\ProgramData\Sparronator9999\YAMDCC\CurrentConfig.xml) and change the <OffsetDT> option from true (the default) to false, then restart the YAMDCC service (or close and re-open the YAMDCC Config Editor, then click Apply).

Ooh ok - I just checked my CurrentConfig.xml (also attached if you're curious) and it does have <OffsetDT> true, will give this a try and let you know how it goes tomorrow! Thanks again for the very quick help.

[CurrentConfig.txt](https://github.com/user-attachments/files/20235922/CurrentConfig.txt)

ThreeScreenStudios, 2025-05-16:
Also I've worked on .NET projects most of my career (along with WPF and WinUI3 recently) so if you ever need any tasks done, would be happy to help contribute :)

Sparronator9999, 2025-05-16:
I just had a look at your config, and you definitely need the `OffsetDT` option disabled. You can tell because your default fan profiles all have values of 5°C or 8°C, which is way too cold for the fans to ever decrease in speed with `OffsetDT` enabled. With your custom fan profile, the fans wouldn't slow down unless your CPU hit a chilly 3°C, which I doubt you'd be able to achieve unless you use your laptop in a freezer lol

Sparronator9999, 2025-05-16:
> Also I've worked on .NET projects most of my career (along with WPF and WinUI3 recently) so if you ever need any tasks done, would be happy to help contribute :)

Help would be appreciated, however time for chatting is probably going to be limited due to me (probaby) living on the other side of the world from you (time zone is on my bio)

closed 2025-05-16 by ThreeScreenStudios

Sparronator9999, 2025-05-16:
Should the title bar in that screenshot you attached be black? My system is set to use a similar colour theme, and it shows white text instead...

> Haha oh wow, yeah we really are on the total opposite sides of the world - let me know how you want to connect and we can figure it out, my email is woot [at] threescreenstudios.com.

The troubles of living in a country that has like 1% of the English-speaking population, that is also on the other side of the world from the other 99%...

You can reach me through Matrix (link on my bio). If you don't know what that is, it's basically Discord but decentralised and more privacy-friendly (I think. I have somehow avoided using Discord to this point, and I don't want to start today).

That being said, I'm not really sure what there is to do at the moment. I've already started work on the CLI (for the next version), so I would not recommend starting your own version in the meantime. If you do decide to look at the source code, try not to get brain cancer from my coding "style".

I noticed that you have WPF experience, which is nice, but if the GUI applications were to be rewritten for another UI (which probably won't happen for a while unless someone else does it), I would prefer if it had Linux support (which is planned for the future), and WPF is only supported on Windows (haven't heard of WinUI).

ThreeScreenStudios, 2025-05-16:
Sounds good to me! Not trying to make more work for you or insert myself, so it's just if I can be helpful at all :) Haven't used Matrix before but I'm definitely willing to give it a go.

glpnk, 2025-05-17:
Does this app set SCM bit (`d9 bit 1`) high? Today we've found that some devices do not enable fan curve, leds, screen format detection, when it unset