Sparronator9999/YAMDCC #89: Maybe this will help?

FengXi7420, 2025-05-16:
I guess this can be helpful to your development 
https://pan.hotpe.top/f/Pv6CL/NEW_9548_ECRAM_Qevent_OemScancode_Define_20210322.xlsx

Sparronator9999, 2025-05-16:
This is probably more suited to go in a discussion (maybe I should make one).

That being said, the link you provided doesn't work for me. Try uploading to GitHub directly (or in a .zip file if GitHub doesn't allow .xlsx files for some reason).

Sparronator9999, 2025-05-16:
Where did you find this? This could come in handy for some extra YAMDCC features and automatic feature support detection in the future. Thank you for sharing, now I will probably spend time messing around with this instead of working on the CLI lol.

closed 2025-05-16 by Sparronator9999

FengXi7420, 2025-08-20:
> Where did you find this? This could come in handy for some extra YAMDCC features and automatic feature support detection in the future. Thank you for sharing, now I will probably spend time messing around with this instead of working on the CLI lol.

I have something more now : MSITOOLS

[4.0.22.zip](https://github.com/user-attachments/files/21884547/4.0.22.zip)

Sparronator9999, 2025-08-20:
OK, I should start by asking what this is? The only thing I really need help with is the GPU switch feature, which MSITOOLS (on first glance) doesn't appear to have.

FengXi7420, 2025-08-20:
> OK, I should start by asking what this is? The only thing I really need help with is the GPU switch feature, which MSITOOLS (on first glance) doesn't appear to have.

MSI internal server

FengXi7420, 2025-08-20:
> OK, I should start by asking what this is? The only thing I really need help with is the GPU switch feature, which MSITOOLS (on first glance) doesn't appear to have.

Just a small tool to set fan speed

FengXi7420, 2025-08-20:
> OK, I should start by asking what this is? The only thing I really need help with is the GPU switch feature, which MSITOOLS (on first glance) doesn't appear to have.

Any useful things in here?
Oh you'd better give me private email or discord DM maybe

Sparronator9999, 2025-08-20:
I have a [Matrix account](https://matrix.to/#/@sparronator9999:matrix.org), which I'm pretty sure has private messaging (I've never actually used it for much though lol)

Sparronator9999, 2025-08-20:
> It looks like msi internal debugger for testing MSI center pressets. Is that legal? lol

In that case, I would rather not RE these programs due to potential copyright issues.

PS: @porkmanager I accidentally edited out the quote replies on your message (I hit the wrong button by accident). These have been restored.

FengXi7420, 2025-08-21:
> In that case, I would rather not RE these programs due to potential copyright issues.
>
> PS: @porkmanager I accidentally edited out the quote replies on your message (I hit the wrong button by [...]

It's OK to RE it I also have some code photos that I have to send you by private