Sparronator9999/YAMDCC #92: MSI Vector 16 HX A14V Fan Profile + How to apply settings automatically  ?

KamykJr, 2025-05-25:
Hello. I had a problem with pull requests ;) So I am adding a base profile here.

https://github.com/KamykJr/MSI-VECTOR-16-HX-A14V/blob/main/Vector-16-HX-A14VHG%20Default%20%2B%20Silent%20by%20Adam.xml

I also have a question, how to apply settings automatically after program start ? I have added a program to autostart but any time I reboot my computer I have to reclick and apply ;)

KamykJr, 2025-05-26:
I have checked services. It it set to automatic and is running but still i have to click apply in config editor.

Sparronator9999, 2025-05-26:
Well, this is certainly weird. Can you please restart the YAMDCC service (or your computer), then check the latest service log (located at `C:\ProgramData\Sparronator9999\YAMDCC\Logs\yamdccsvc.log`) and make sure that it says `Applying YAMDCC config...` shortly after the YAMDCC service starts? There will also be a bunch of log spam if the log level is set to Debug, or potentially a bunch of EC write errors if that's failing for some reason.

KamykJr, 2025-05-26:
Before clicking aplly, after reboot: 

[26.05.2025 08:41:07.443] [Info]   OS version: Microsoft Windows NT 10.0.26100.0
[26.05.2025 08:41:07.443] [Info]   Service version: 1.1.0-release+7e7db6f21942812d8ec51d3a5e42645d152a0097
[26.05.2025 08:41:07.485] [Debug]  Log level is set to debug mode.
[26.05.2025 08:41:07.494] [Info]   Starting YAMDCC service...
[26.05.2025 08:41:07.494] [Info]   Loading WinRing0 driver for EC access...
[26.05.2025 08:41:07.540] [Info]   WinRing0 driver loaded successfully!
[26.05.2025 08:41:07.542] [Info]   Reloading service config...
[26.05.2025 08:41:07.573] [Info]   Config loaded successfully!
[26.05.2025 08:41:07.589] [Debug]  EC firmware version: 15M1IMS2.111
[26.05.2025 08:41:07.604] [Debug]  EC firmware date: 1.11.2024 17:58:45
[26.05.2025 08:41:07.604] [Info]   Starting IPC server...
[26.05.2025 08:41:07.607] [Info]   Service started successfully!
[26.05.2025 08:41:07.607] [Info]   Applying YAMDCC config...
[26.05.2025 08:41:07.607] [Info]   Writing fan profile for `CPU Fan` (1/2)...
[26.05.2025 08:41:07.609] [Debug]  Wrote to EC register 0x72 successfully.
[26.05.2025 08:41:07.609] [Debug]  Wrote to EC register 0x73 successfully.
[26.05.2025 08:41:07.609] [Debug]  Wrote to EC register 0x6A successfully.
[26.05.2025 08:41:07.609] [Debug]  Wrote to EC register 0x7A successfully.
[26.05.2025 08:41:07.609] [Debug]  Wrote to EC register 0x74 successfully.
[26.05.2025 08:41:07.609] [Debug]  Wrote to EC register 0x6B successfully.
[26.05.2025 08:41:07.613] [Debug]  Wrote to EC register 0x7B successfully.
[26.05.2025 08:41:07.613] [Debug]  Wrote to EC register 0x75 successfully.
[26.05.2025 08:41:07.613] [Debug]  Wrote to EC register 0x6C successfully.
[26.05.2025 08:41:07.613] [Debug]  Wrote to EC register 0x7C successfully.
[26.05.2025 08:41:07.613] [Debug]  Wrote to EC register 0x76 successfully.
[26.05.2025 08:41:07.613] [Debug]  Wrote to EC register 0x6D successfully.
[26.05.2025 08:41:07.617] [Debug]  Wrote to EC register 0x7D successfully.
[26.05.2025 08:41:07.617] [Debug]  Wrote to EC register 0x77 successfully.
[26.05.2025 08:41:07.617] [Debug]  Wrote to EC register 0x6E successfully.
[26.05.2025 08:41:07.617] [Debug]  Wrote to EC register 0x7E successfully.
[26.05.2025 08:41:07.617] [Debug]  Wrote to EC register 0x78 successfully.
[26.05.2025 08:41:07.620] [Debug]  Wrote to EC register 0x6F successfully.
[26.05.2025 08:41:07.620] [Debug]  Wrote to EC register 0x7F successfully.
[26.05.2025 08:41:07.620] [Info]   Writing performance mode setting...
[26.05.2025 08:41:07.620] [Debug]  Wrote to EC register 0xD2 successfully.
[26.05.2025 08:41:07.620] [Info]   Writing fan profile for `GPU Fan` (2/2)...
[26.05.2025 08:41:07.620] [Debug]  Wrote to EC register 0x8A successfully.
[26.05.2025 08:41:07.620] [Debug]  Wrote to EC register 0x8B successfully.
[26.05.2025 08:41:07.620] [Debug]  Wrote to EC register 0x82 successfully.
[26.05.2025 08:41:07.620] [Debug]  Wrote to EC register 0x92 successfully.
[26.05.2025 08:41:07.620] [Debug]  Wrote to EC register 0x8C successfully.
[26.05.2025 08:41:07.620] [Debug]  Wrote to EC register 0x83 successfully.
[26.05.2025 08:41:07.620] [Debug]  Wrote to EC register 0x93 successfully.
[26.05.2025 08:41:07.627] [Debug]  Wrote to EC register 0x8D successfully.
[26.05.2025 08:41:07.627] [Debug]  Wrote to EC register 0x84 successfully.
[26.05.2025 08:41:07.627] [Debug]  Wrote to EC register 0x94 successfully.
[26.05.2025 08:41:07.629] [Debug]  Wrote to EC register 0x8E successfully.
[26.05.2025 08:41:07.629] [Debug]  Wrote to EC register 0x85 successfully.
[26.05.2025 08:41:07.629] [Debug]  Wrote to EC register 0x95 successfully.
[26.05.2025 08:41:07.629] [Debug]  Wrote to EC register 0x8F successfully.
[26.05.2025 08:41:07.629] [Debug]  Wrote to EC register 0x86 successfully.
[26.05.2025 08:41:07.629] [Debug]  Wrote to EC register 0x96 successfully.
[26.05.2025 08:41:07.629] [Debug]  Wrote to EC register 0x90 successfully.
[26.05.2025 08:41:07.629] [Debug]  Wrote to EC register 0x87 successfully.
[26.05.2025 08:41:07.629] [Debug]  Wrote to EC register 0x97 successfully.
[26.05.2025 08:41:07.629] [Info]   Writing charge limit configuration...
[26.05.2025 08:41:07.629] [Debug]  Wrote to EC register 0xD7 successfully.
[26.05.2025 08:41:07.629] [Info]   Writing fan mode setting...
[26.05.2025 08:41:07.629] [Debug]  Wrote to EC register 0xD4 successfully.
[26.05.2025 08:41:07.629] [Info]   Writing Win/Fn key swap setting...
[26.05.2025 08:41:07.636] [Debug]  Wrote to EC register 0xE8 successfully.
[26.05.2025 08:41:08.639] [Info]   Client connected to service (ID: 1)

After I click aplly:

[26.05.2025 08:48:28.238] [Info]   Reloading service config...
[26.05.2025 08:48:28.238] [Info]   Config loaded successfully!
[26.05.2025 08:48:28.238] [Debug]  EC firmware version: 15M1IMS2.111
[... 25 lines cut ...]
[26.05.2025 08:48:28.270] [Debug]  Wrote to EC register 0x8A successfully.
[26.05.2025 08:48:28.270] [Debug]  Wrote to EC register 0x8B successfully.
[26.05.2025 08:48:28.270] [Debug]  Wrote to EC register 0x82 successfully.
[26.05.2025 08:48:28.270] [Debug]  Wrote to EC register 0x92 successfully.
[26.05.2025 08:48:28.270] [Debug]  Wrote to EC register 0x8C successfully.
[26.05.2025 08:48:28.270] [Debug]  Wrote to EC register 0x83 successfully.
[26.05.2025 08:48:28.270] [Debug]  Wrote to EC register 0x93 successfully.
[26.05.2025 08:48:28.270] [Debug]  Wrote to EC register 0x8D successfully.
[... cut ...]
[26.05.2025 08:48:28.270] [Debug]  Wrote to EC register 0x97 successfully.
[26.05.2025 08:48:28.270] [Info]   Writing charge limit configuration...
[26.05.2025 08:48:28.286] [Debug]  Wrote to EC register 0xD7 successfully.
[26.05.2025 08:48:28.286] [Info]   Writing fan mode setting...
[26.05.2025 08:48:28.286] [Debug]  Wrote to EC register 0xD4 successfully.
[26.05.2025 08:48:28.286] [Info]   Writing Win/Fn key swap setting...
[26.05.2025 08:48:28.286] [Debug]  Wrote to EC register 0xE8 successfully.

Sparronator9999, 2025-06-01:
> Maby I am doing something wrong directly in your program ?

> Maby there is some problems with a default settings ? And it writes them to the registers, not the modified one ?

This seems like another manifestation of a known issue for this project, where occasionally EC reads and writes just don't work for an unknown reason (in other words, I have no idea why this is happening, and hope that switching to WMI will fix this). The default and custom profiles will always write to the same registers (they are the same for every MSI laptop - it is only some of the extra features that are different).

bullet0006, 2025-06-06:
It is possible that the YAMDCC service started first, and then MSI Center started later, overwriting the EC writes from YAMDCC. You can try setting the YAMDCC service to delayed start and see what happens.

Sparronator9999, 2025-06-07:
> It is possible that the YAMDCC service started first, and then MSI Center started later, overwriting the EC writes from YAMDCC. You can try setting the YAMDCC service to delayed start and see what happens.

I do not recommend running YAMDCC and MSI Center at the same time. *Technically* it's possible, but since both do similar things, they will end up interfering with each other. That being said, delayed start for YAMDCC would only fix a bad fan profile on startup, and do nothing if MSI Center is re-writing the fan profile after YAMDCC when waking the laptop from sleep.

glpnk, 2025-06-07:
> bad fan profile on startup

Which value `0xd4` byte have?

MJRT, 2025-06-10:
+1, same issue

MJRT, 2025-06-10:
> +1, same issue

@KamykJr @Sparronator9999 I found that was caused by MSI center service. After close MSI center app and MSI center service, fan control can meet the expected. Do not let them start automatically.