Sparronator9999/YAMDCC #95: Hotkey Handler bug??

VisionCrizzal, 2025-06-04:
### Prerequisites

- [ ] This issue is not a [feature request](https://github.com/Sparronator9999/YAMDCC/issues/new?template=feature_request.md) or [bug report](https://github.com/Sparronator9999/YAMDCC/issues/new?template=question.md).
- [X] I have searched the [issue tracker](https://github.com/Sparronator9999/YAMDCC/issues?q=label:"type:+bug") for duplicate or similar bug reports.
- [ ] I have not answered the above questions truthfully. (really???)

### Describe the bug

After i unintalled the MSI foundation for hotkey and using the hotkey handler,when i press shift+7 for open config editor: 

[image]

It's show this:

[image]

But when i manualy open config editor,it's open nomaly.

### To Reproduce

1. Go to 'Hotkey handler'
2. Config any hotkey for "open config editor
3. Apply,minimize to tray and press the hotkey
4. See error

### Expected behaviour

Press shift+7 will show config editor

### Additional context

BTW,the toggle win/fn snap is not working and toggle switch perf mode is...working or not working???(the config editor still show the default perf mode)

Sparronator9999, 2025-06-04:
I cannot reproduce this bug on my system, however this issue can happen if the Config Editor and Hotkey Handler executables are not in the same folder.

- How did you install YAMDCC (i.e. did you install it normally, or install in "portable mode")? Did you make sure both "Config editor" and "Hotkey handler" components were enabled during a normal install?
- If you open Task Manager (with Hotkey Handler open), right-click on the open Hotkey Handler instance, then click `Open file location`, can you find both `HotkeyHandler.exe` and `ConfigEditor.exe` in the same folder? If not, then there's the problem.

Sparronator9999, 2025-06-04:
I also just realised one of my preliminary questions has a typo (it should say "feature request or question", not "feature request or bug report"). The third check box is to make sure people actually *read* the questions instead of just checking every check box they see (this is inspired by some other GitHub projects).

VisionCrizzal, 2025-06-04:
> I cannot reproduce this bug on my system, however this issue can happen if the Config Editor and Hotkey Handler executables are not in the same folder.
> 
>     * How did you install YAMDCC (i.e. did you install it normally, or install in "portable mode")? Did you make sure both "Config editor" and "Hotkey handler" components were enabled if installing normally?
> 
>     * If you open Task Manager (with Hotkey Handler open), right-click on the open Hotkey Handler instance, then click `Open file location`, can you find both `HotkeyHandler.exe` and `ConfigEditor.exe` in the same folder? If not, then there's the problem.

Eh.... I have to lunch now so im not in my lap
It 11:52am in Vietnam.

VisionCrizzal, 2025-06-04:
> I cannot reproduce this bug on my system, however this issue can happen if the Config Editor and Hotkey Handler executables are not in the same folder.
>
>     * How did you install YAMDCC (i.e. did [...]

ok i'm on my lap now.
1. i'm install YAMDCC normally,all components are enable.
2. Both `HotkeyHandler.exe` and `ConfigEditor.exe` in the same folder.

Should i'm reinstall or change to portable install ???

VisionCrizzal, 2025-06-04:
> > 1. i'm install YAMDCC normally,all components are enable.
> > 2. Both HotkeyHandler.exe and ConfigEditor.exe in the same folder.
> > 
> > Should i'm reinstall or change to portable install ???
> 
> Well, this is very weird indeed. Maybe a screenshot will clear things up (but that's not seeming likely).

[image]

VisionCrizzal, 2025-06-04:
> Well, they certainly appear to be in the same directory, which is why I just pushed a commit to try and figure out what Hotkey Handler's trying to actually open (for all I know it could be somehow trying to open a `ConfigEditor.exe` located in System32...)
> 
> Can you please test the new commit once it finishes building? You can download it from [GitHub Actions](https://github.com/Sparronator9999/YAMDCC/actions). If it still fails to find the Config Editor, it should tell you where it looked in the error output now.
> 
> Also, still a Windows 10 holdout like me, I see. Nice.

It's work, no error.
And yes,win 11 it's SUCK (specially with AMD fanboy like me,F*CK MICROSOFT!!!!!!)

VisionCrizzal, 2025-06-04:
> > Well, they certainly appear to be in the same directory, which is why I just pushed a commit to try and figure out what Hotkey Handler's trying to actually open (for all I know it could be somehow trying to open a `ConfigEditor.exe` located in System32...)
> > Can you please test the new commit once it finishes building? You can download it from [GitHub Actions](https://github.com/Sparronator9999/YAMDCC/actions). If it still fails to find the Config Editor, it should tell you where it looked in the error output now.
> > Also, still a Windows 10 holdout like me, I see. Nice.
> 
> It's work, no error. And yes,win 11 it's SUCK (specially with AMD fanboy like me,F*CK MICROSOFT!!!!!!) I think i will try reinstall the official version to see the error remain.

wait, the official version after reinstall the error doesn't exit anymore???

Sparronator9999, 2025-06-04:
> wait, the official version after reinstall the error doesn't exit anymore???

Man, I love bugs that you can't consistently reproduce. I'm sure Garmin does as well (I recently got one of their watches, and even navigating the damn UI is enough to crash and reboot the watch on a bad day).

closed 2025-06-04 by Sparronator9999

VisionCrizzal, 2025-06-04:
> If you're interested in the details: to implement the directory readout, I had to first convert the supplied path to an absolute path, instead of letting the call to open the Config Editor do it for me. Meaning that you somehow uncovered an implementation difference? Or maybe you just got lucky?? Who knows, all that I care about is it's fixed now.

eh... give me few minutes for my bad English skill to understand what that mean (or using gg translate)
> 
> > And yes,win 11 it's SUCK (specially with AMD fanboy like me,F*CK MICROSOFT!!!!!!)
> 
> Does AMD have problems on windows 11? I wouldn't know because 1. I've never run Windows 11 on bare metal (only the LTSC release in a VM briefly), and 2. I don't own any computers with an AMD CPU (or are you talking about GPUs?)

If you run windows 11 23h2,it's seem like MS is trying to nerf AMD CPU (and VGAs) but on Intel cpu it's normal(sound like Win 11 is choosing intel as favor child,BRUH).And after i heard about win 11 24h2 will increase the perform of AMD, i just feel like MS just try to force anyone to update the Win version????
For more info: https://www.youtube.com/watch?v=2KoiNtJ3lHA&t=198s
(I swear not a rickroll,for sure)
> Beware, a rant follows. Click to expand.
> 
> Honestly, if you want my opinion: the UI _looks_ nice (but Windows Vista/7 will always be my favourite UI), but there's too many small changes to where old settings are located, plus several annoying decisions including FINALLY adding the volume mixer to the regular volume UI, but putting it below an option I'm sure only about 1% of users knew existed before Windows 11 came out, and not to mention the new context menus that move half the options to its own menu, among other annoyances.

I'm just a gen z dude so i still perfer the win 10 ui (when i were a child,i'm playing web game on my dad windows XP laptop and my uncle win 7 laptop (both using celeron intel) so it mean i not like average user(at least))
> 
> I have tried Linux before (in particular, Linux Mint), but I still haven't found a ThrottleStop alternative that does everything ThrottleStop does on Windows, just on Linux instead, plus anti-cheat games don't work (which is its own issue, and therefore won't get into here).

oh, in the past i'm tried Fedora,even the gpu passthrough( to make Fortnite work. Yes, i'm a Fortnite addited) but it have many issue and i just feel it not worth it.

Sparronator9999, 2025-06-04:
> If you run windows 11 23h2,it's seem like MS is trying to nerf AMD CPU (and VGAs) but on Intel cpu it's normal(sound like Win 11 is choosing intel as favor child,BRUH).And after i heard about win 11 [...]

Hmm, interesting. It seems as though AMD's branch prediction on Zen 5 CPUs needs support on the OS side for some reason? Which was only introduced in 24H2?

I remember seeing something similar when the first Intel CPUs with "E-cores" and "P-cores" came out, where Microsoft updated the thread scheduling to make better use of these two different CPU core types, except support for that was in Windows by the 21H2 LTSC... Fuck, they are picking favourites.

> I'm just a gen z dude so i still perfer the win 10 ui (when i were a child,i'm playing web game on my dad windows XP laptop and my uncle win 7 laptop (both using celeron intel) so it mean i not like average user(at least))

Gen Z here as well. My first computer was my dad's old Pentium 4 (3 GHz with hyper-threading!) XP machine, with nothing but the Intel "GMA" (which I like to call the "graphics media decelerator") for graphics, then switched to a shitty Vista laptop with a Celeron and a *maybe* slightly better GMA "graphics card"? It was good enough to run the Aero UI though, apparently (and the bloatware game trials it came with), and actually rendered the graphics of a certain movie tie-in video game correctly compared to the Pentium machine.

> oh, in the past i'm tried Fedora,even the gpu passthrough( to make Fortnite work. Yes, i'm a Fortnite addited) but it have many issue and i just feel it not worth it.

Now that I think of it, the NVIDIA GPU drivers on Linux wouldn't let my graphics card sleep half the time, which is obviously very bad for battery life. I started trying to GPU passthrough it to a Windows VM, but stopped halfway through, and even then I'm not sure it would've worked (either some magic would be needed to get the GPU output back to the iGPU, or an external monitor connected to the zero video outputs directly connected to the dGPU on my laptop)

As for Fortnite, last time I played it was when my youngest brother asked me to, which is not very often in recent times. Roblox, on the other hand, he still manages to find new ways to get me back on that dumpster fire (which is one of the reasons I still need Windows - Roblox introduced an anti-cheat that broke Linux compatibility a year or two ago).

VisionCrizzal, 2025-06-04:
hey it's me again 
that bugs are back,it seem that the hotkey handler is error after reboot (config editer still fine)

Sparronator9999, 2025-06-05:
The issue is open once again.

It could be to do with how the hotkey handler is launched. In particular, it's probably started with a working directory of its own location (i.e. `C:\Program Files\Sparronator9999\YAMDCC` for a normal install), but some apps might start it with the default working directory of `C:\Windows\System32`, which the Config Editor is obviously not located in.

Sparronator9999, 2025-06-07:
> I can literally reproduce this with any Windows 10 version for some reason lol, not the exact issue but something similar to it, I let the laptop go to sleep on my desk, come back at it just to find that the fans are going jet mode and cpu is overheating even though task manager shows its not even doing anything, i never got to investigate why it only happens with windows 10

So I'm not crazy, Windows *is* trying to mine crypto on my computer when idle... /s

Seriously though, this happens on all my laptops, and can't possibly be good for battery runtime (it happens on my laptops even when unplugged sometimes...)

> honestly agreed, all BS aside windows 11 does look quite decent.... only if there was some sort of layer that took win32 apps and converted them to fluent ones

Bootleg OS creators are better than Microsoft at bringing dark themes to classic Win32 apps. That should tell you why Microsoft hasn't found a way to "convert" Win32 apps to the Fluent design.

> Doesn't StartAllBack fix like most of these? ive been using it since day one and havent seen any issues in years

I wouldn't know. I've heard of a program called ExplorerPatcher, which restores some Windows 10 features, and I'm pretty sure that the context menu can be reverted with a registry key, but my point is we shouldn't need these programs to get Windows the way we want it.

> lookingglass+virtual display drivers solve this

Cool, will keep in mind.

> Did you have the drivers installed properly? just installing the package doesnt really do it the "proper" way, e.g on my laptop running "sudo ubuntu-drivers install"(on an ubuntu based distro obviously) did all the config stuff and made it work properly

Last time I installed Linux on my MSI laptop was with Linux Mint, and that had a Driver Manager application that I installed the proprietary driver from.

> this is fixed, with the release of [Sober](https://sober.vinegarhq.org/)

Interesting, I might check it out at some point. Maybe I can squeeze a little bit more performance out of an ancient laptop my younger sibling plays Roblox on.

Sparronator9999, 2025-08-14:
> hey, the bug is happen again

I think I know why this bug is happening now.

Basically, when you launch Hotkey Handler directly, the "working directory" is set to the same directory as where Hotkey Handler is located. When it launches on system startup, Windows must start the app in the default working directory of `C:\Windows\System32`, resulting in Hotkey Handler (and possibly other programs) not being able to locate executables stored in the same directory as it.

The "fix" I did last time was to explicitly try to get the full executable path (using [`Path.GetFullPath()`](https://learn.microsoft.com/en-us/dotnet/api/system.io.path.getfullpath?view=netframework-4.8), and if that failed, show an error with where it thought the exe was, hoping someone would report it (thanks for that, by the way).

A fix will be released soon.

ios7jbpro, 2025-08-15:
@Sparronator9999 cant you create a shortcut/link(whatever the extension of shortcuts are), and move that to startup instead? afaik that way itll run on the directory its in