Sparronator9999/YAMDCC #98: v1.1 cannot get default fan profile + no error log folder

gemSquared, 2025-06-18:
### Prerequisites

- [x] This issue is not a [feature request](https://github.com/Sparronator9999/YAMDCC/issues/new?template=feature_request.md) or [question](https://github.com/Sparronator9999/YAMDCC/issues/new?template=question.md).
- [x] I have searched the [issue tracker](https://github.com/Sparronator9999/YAMDCC/issues?q=label:"type:+bug") for duplicate or similar bug reports.
- [ ] I have not answered the above questions truthfully.

### Describe the bug

v1.1

Cannot get default fan profile. Specified error log folder does not exist either.

Laptop default fan profile changed. Attempted to do BIOS update; fan profiles the same so not sure if it was intentional by MSI or one of my weird configs snuck in *somehow* but that's not the topic of this issue just yet.

### To Reproduce

1. Use v1.1
2. Load any profile, blank slate preferred
3. Options > Get default fan profile from EC
4. Reboot as instructed
5. Reopen v1.1
6. See error and no log folder created so no log

### Expected behavior

Program should've pulled default fan profile from EC

### Additional context

[image]

[image]

Sparronator9999, 2025-06-18:
Hello, can you please try performing an EC reset? You may be experiencing a bug caused by bad EC writes to a certain location (My laptop's fans straight up stopped working once until I did an EC reset...)

1. Turn off your computer (don't restart it yet).
2. Find the EC reset button (pin hole on the bottom of the laptop for me) and press it for a few seconds with a paper clip, SIM eject tool, or similar. If you can't find it, hold the power button for 30 seconds.
3. Turn the computer back on. You may need to connect it to the charger first.
4. Try pulling the default fan profile again.

Sparronator9999, 2025-06-18:
> Specified error log folder does not exist either.

You need to look in the `ProgramData` folder, not `Program Files`. It's hidden by default, so you might need to enable hidden files and folders, or just copy and paste the following into Explorer to navigate to YAMDCC's config directory directly: `C:\ProgramData\Sparronator9999\YAMDCC`

I can see how the two can get mixed up...

gemSquared, 2025-06-18:
Lol am blind hahah! On it o7

gemSquared, 2025-06-19:
New day, requested EC read
[yamdccsvc.exe.3.log.gz](https://github.com/user-attachments/files/20813708/yamdccsvc.exe.3.log.gz)

After reboot
[yamdccsvc.exe.log](https://github.com/user-attachments/files/20813709/yamdccsvc.exe.log)

So it worked and didn't work at the same time?

gemSquared, 2025-06-19:
[image]

[image]

(somewhat irrelevant still) What appears MSI Center has adopted an older version of my weird config as the default fan curve when I hit reset so *yaaay* that's weird haha.

Sparronator9999, 2025-06-19:
I think the YAMDCC service might need more logging. It does, however, look like the EC-to-config feature is working as intended.

Glad you figured out the problem. Personally I wouldn't run both MSI Center and YAMDCC at the same time, because as you saw, it can cause issues like this, where one picks up the fan profile of the other, among many other potential problems.

closed 2025-06-19 by Sparronator9999

gemSquared, 2025-06-19:
Yeah, they mess with each other. But after I supposedly removed MSI Center parts of it still remain like the GUI that pops up when I toggle camera on/off or full on can still change user scenarios. So clearly it leaves a ghost.

Sparronator9999, 2025-06-19:
Installing MSI Center on my system leaves behind remnants of the Nahimic audio driver and configuration utility, which even when uninstalled applies an EQ by default (instead of just straight, unfiltered audio), which persists until re-installing Windows (or possibly removing the kernel driver that it installs, but I haven't tried this).

The GUI doesn't get left behind, though. Weird. Have you:

1. Uninstalled it from the Microsoft Store
2. Uninstalled the extra programs it installs (I can't remember what it's called, but I'm pretty sure it was "MSI Center SDK" or something like that)

gemSquared, 2025-06-19:
[image]

Can only speculate that the latest update apparently made MSI Center think the current configuration was the default configuration and it saved it somewhere I can't touch. Either way, with MSI Center installed or taken out, when I reboot it resets to the config it thinks is "default" and I'm wee bummed it was the older weird config I made 😅

Either way. It be a me problem. Otherwise let's hope you figure out what that error box was o7

Sparronator9999, 2025-06-19:
> You can't just click uninstall to remove that shit.

Yeah, I know. That's why I re-installed Windows once YAMDCC was good enough to use on its own (also I somehow managed to corrupt something in that install IIRC)