Sparronator9999/YAMDCC #101: Can you improve the wiki please

Claod44, 2025-07-15:
### Prerequisites

- [x] This issue is not a [bug report](https://github.com/Sparronator9999/YAMDCC/issues/new?template=bug_report.md) or [feature request](https://github.com/Sparronator9999/YAMDCC/issues/new?template=feature_request.md).
- [x] I have searched the [FAQ](https://github.com/Sparronator9999/YAMDCC#faq) for answers.
- [x] I have searched the [issue tracker](https://github.com/Sparronator9999/YAMDCC/issues?q=label:"type:+question") for duplicate or similar questions.
- [x] I have not answered the above questions truthfully.

### Question
Hello,
I have a MSI katana GF76 11UEK
The pc in not yet supported
I was trying to understand the wiki for an hour now, and I'm sorry but it's unclear.

how can you obtain this : 

Up threshold: 0, down threshold: 0 (this will always be the same across YAMDCC configs)
Up threshold: 55, down threshold: 47
Up threshold: 64, down threshold: 61
Up threshold: 73, down threshold: 70
Up threshold: 76, down threshold: 73
Up threshold: 82, down threshold: 79
Up threshold: 88, down threshold: 85

From this
00 20 2A 36 43 52 64
7 hex values : 12 number as result except the 0 - 0.

using this : 
Tdown = Tup - Toffset

No matter what I try with your exemple I can't get those values.
The wiki need example, or some screenshots... 

I don't know if my hex will show correclty but here is mine : 

 0x | 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
----|-------------------------------------------------
 00 | 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
 10 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
 20 | 00 00 00 00 00 00 00 00 0A 05 00 00 00 04 0B 0B  ................
 30 | 03 09 00 0D 01 00 50 81 D2 11 88 2C C8 01 E0 00  ......P....,....
 40 | 00 00 5C 00 9C 0D 00 00 6C 0C 1D 2F E1 0B 00 00  ..\.....l../....
 50 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
 60 | 00 00 00 00 00 00 00 00 4D 00 37 40 49 4C 52 58  ........M.7@ILRX
 70 | 64 3C 26 2B 30 36 3C 46 55 64 08 03 03 03 03 03  d<&+06<FUd......
 80 | 38 00 37 3D 43 49 4F 54 63 2B 00 2B 30 36 3C 46  8.7=CIOTc+.+06<F
 90 | 55 64 08 03 03 03 03 02 86 0F 7D 06 0A 78 40 00  Ud........}..x@.
 A0 | 31 37 4C 31 45 4D 53 31 2E 31 30 37 30 36 32 38  17L1EMS1.1070628
 B0 | 32 30 32 32 30 39 3A 30 33 3A 33 39 00 00 00 28  202209:03:39...(
 C0 | 00 00 07 33 00 00 00 00 00 41 00 3A 00 00 00 00  ...3.....A.:....
 D0 | 00 00 C0 83 0D 00 05 80 04 03 00 00 00 00 00 00  ................
 E0 | E2 00 00 9C 0D 00 00 40 00 00 00 00 00 C0 00 00  .......@........
 F0 | 00 00 70 00 00 64 00 00 64 00 00 00 00 00 00 00  ..p..d..d.......

Sparronator9999, 2025-07-15:
Hello, the wiki hasn't been up to date for ages (if not ever - there's a reason there's so many TODOs). Maybe I should just open the wiki to public editing and let others figure out how to make it better.

As for your laptop, all you should need to do is load the generic config called `MSI-10th-gen-or-newer-dualfan.xml` (or `MSI-10th-gen-or-newer-singlefan.xml` if your laptop only has one fan): [link](https://github.com/Sparronator9999/YAMDCC/blob/main/Configs/Generic/MSI-10th-gen-or-newer-dualfan.xml). Then go to `Options` > `Get default fan profile from EC...` and follow the instructions.

Claod44, 2025-07-15:
I think there is 2 fans. So MSI-10th-gen-or-newer-dualfan.xml ? By the way I don't need the backlight control, I am already using OpenRGB for it, so ... is the MSI-10th-gen-or-newer-dualfan-nokeylight.xml ok for me too ?

Sparronator9999, 2025-07-15:
If you're already using another program for backlight control, chances are YAMDCC won't work with your keyboard backlight anyways (it only supports monochromatic (single-colour) keyboard backlights with 4 brightness settings including off). So **use `MSI-10th-gen-or-newer-dualfan-nokeylight.xml`**, then follow my instructions above.

Claod44, 2025-07-15:
The values of the xml for the temps, and fans will be ok according the hex values I sent earlier ? So bad I can't find how to proceed by myself, I would have edit the wiki to help others...

Claod44, 2025-07-15:
Sorry for the double post, it seems okfor now. The config is on, I'll test to see what happens.
By the way I d'ont know if the YAMDCC window needs to stay opened ? Or once the settings are applied the service is managing everything so it can be closed ?
Do I need to start YAMDCC on boot, and load the profile with the Task Scheduler with some parameter ?

Thank you for your time :)

Sparronator9999, 2025-07-15:
> Sorry for the double post, it seems okfor now. The config is on, I'll test to see what happens. By the way I d'ont know if the YAMDCC window needs to stay opened ? Or once the settings are applied [...]

You do not need to leave the YAMDCC config editor open. The YAMDCC service will ensure your settings stay applied across reboots and sleep cycles, assuming you don't stop or disable it.

Claod44, 2025-07-15:
:/ me again, when I close the window the full blast stops :'(

Sparronator9999, 2025-07-15:
> :/ me again, when I close the window the full blast stops :'(

That's a feature, not a bug. I'm probably going to remove that feature (or make it toggleable) soon though (you are not the first person to want full blast to stay active after closing the YAMDCC config editor).

Claod44, 2025-07-15:
Ok, thanks, I still got the Fn + up key so it's fine, if it's normal. If the next release let the toggle active through the service it would be amazing ;) thanks again

Sparronator9999, 2025-07-15:
I implemented the option to disable the Full Blast disabling when closing Config Editor in the latest commit.