Sparronator9999/YAMDCC #102: howto cmd?

porkmanager, 2025-07-20:
Hey bro! How to use cmd commands? I tried latest build and run ConfigEditor.exe -monitor it provided me nothing.

Sparronator9999, 2025-07-20:
The CLI is not included in GitHub Actions builds yet due to a grand total of about 5 (out of 16) commands being implemented at all (let alone properly tested). You'll have to build it yourself if you want to test what's already implemented.

These are the only commands that are actually implemented as of writing:

- `-help`, `-version`, and `-license` commands (program help and information)
- `-info` and `-monitor` commands (config info and EC monitor)
- `-fan` and `-profile` commands (to select a fan and fan profile to modify. Not used anywhere yet)
- `-speed` command (the code to parse the new temperature threshold is there but not used yet)

Sparronator9999, 2025-07-31:
Every initial YAMDCC CLI command is now implemented, and the CLI should now be available to install along with everything else. Please test and report issues with the CLI.

closed 2025-07-31 by Sparronator9999