Sparronator9999/YAMDCC #104: Updater not found message after Installation (v1.1.0) and program refuses to load config files

ruudtswaanen, 2025-08-04:
### Prerequisites

- [x] This issue is not a [feature request](https://github.com/Sparronator9999/YAMDCC/issues/new?template=feature_request.md) or [question](https://github.com/Sparronator9999/YAMDCC/issues/new?template=question.md).
- [x] I have searched the [issue tracker](https://github.com/Sparronator9999/YAMDCC/issues?q=label:"type:+bug") for duplicate or similar bug reports.
- [ ] I have not answered the above questions truthfully.

### Describe the bug

I have tried YAMDCC v1.1.0 on two MSI laptops (Prestige 15 A10SC and Cyborg 15 AI A1VEK) but on both laptops after installation and starting of the config editor, I get a dialog box showing "Updater is not found" (while it is in the program directory and I can run it manually).  

I don't mind that issue. 
The second issue is the main one.
I manually downloaded the config files:
MSI-10th-gen-or-newer-dualfan.xml -> Prestige 15 
MSI-10th-gen-or-newer-singlefan.xml -> Cyborg 15
But when I try to load them nothing happens. I don't get fan control sliders.

The YAMDCC service is running fine (no winring0 issues in Windows security) 

### To Reproduce

see above

### Expected behaviour

I would have expected to see fan controls when loading the config xml.

### Additional context

Prestige 15 is a clean install without MSI Creator Center software or other MSI software. 
Cyborg 15 is a MSI installation with MSI Center v2.0.54.0 on it.

Both running 24H2 Win11 fully updated, with local admin accounts.

Maybe I am doing something wrong, since both laptops behave exactly the same.

Sparronator9999, 2025-08-04:
This should really be two separate issues, but I can help you with both now:

What does the Updater Not Found message box look like? Is it shown after updating YAMDCC? (it should pop up a message box saying it was updated successfully).

For your second issue, can you post a screenshot of the Config Editor window (so I can see the "Fan control" tab)?

ruudtswaanen, 2025-08-05:
sorry about combining the two issues.

In the mean time I've tried v1.2.0Beta1 instead (on Cyborg 15) but the exact same thing(s) happen.
Instead me describing the events (poorly), I've made screenshots of the entire installation and config loading process, see attached.

I think it will give more clues than me writing about them.

Basically when loading the config file (MSI-10th-gen-or-newer-singlefan.xml), the program still shows "Please load a config to start".

There are no warnings or errors in Windows Eventlogs (System/Application).

[v1.2.0Beta1_install_screenshots.zip](https://github.com/user-attachments/files/21591850/v1.2.0Beta1_install_screenshots.zip)

Sparronator9999, 2025-08-05:
Oh, it looks like YAMDCC is failing to call the Updater to change the update preference. I will look into this.

As for the config not loading, I cannot reproduce the issue on my end while running v1.2 beta 1.

Sparronator9999, 2025-08-05:
@ruudtswaanen Can you please test the latest commit once it finishes compiling? I have attempted a fix for the failure to open Updater to set the auto-update state on first launch.

To make sure you get the "Do you want to enable automatic updates" dialog, delete the `GlobalConfig.xml`, located in `C:\ProgramData\Sparronator9999\YAMDCC`, before running the updated build.

Sparronator9999, 2025-08-05:
> I really love your idea, to give users, the control over the laptop fans.

This is actually a feature of MSI Center, but that program is so bloated and (probably) full of trackers that I nuked my old Windows install as soon as this program was stable enough to use daily.

Sparronator9999, 2025-08-05:
Those first two config loading errors (where YAMDCC tries to load `CurrentConfig.xml`) are harmless, since it doesn't exist yet.

Opening the XML in a web browser is useless. Don't do that.

Can you please upload your copy of `MSI-10th-gen-or-newer-singlefan.xml`? You will need to add it to a ZIP file, or rename it to have a .txt extension.

ruudtswaanen, 2025-08-05:
I downloaded this one:
[MSI-10th-gen-or-newer-singlefan.zip](https://github.com/user-attachments/files/21595499/MSI-10th-gen-or-newer-singlefan.zip)

ruudtswaanen, 2025-08-05:
I think I know what I did wrong, I went to Configs-> Generic and then use "Save Link As" and that xml file doesn't work. Sorry my bad.

If I click the link first and then download the xml contents, it works.
So sorry for wasting your time on this non-issue.

[image]

Thanks for your patience.

Sparronator9999, 2025-08-05:
> I think I know what I did wrong, I went to Configs-> Generic and then use "Save Link As" and that xml file doesn't work. Sorry my bad.

Yeah, you're supposed to click the "Download raw file" button on that page (or click "Raw", THEN right-click and select "Save page as" to save the XML). I figured you accidentally saved the link to the XML, or the webpage itself. Per-laptop configs will be a thing of the past in a few months* soon, anyways.

I also verified the updater fix actually works. Time to apply it to all the other times a YAMDCC program calls another YAMDCC program...

closed 2025-08-05 by Sparronator9999