Sparronator9999/YAMDCC #110: Turn off auto update

VisionCrizzal, 2025-09-18:
### Prerequisites

- [x] This issue is not a [bug report](https://github.com/Sparronator9999/YAMDCC/issues/new?template=bug_report.md) or [question](https://github.com/Sparronator9999/YAMDCC/issues/new?template=question.md).
- [x] I have searched the [issue tracker](https://github.com/Sparronator9999/YAMDCC/issues?q=label:"type:+enhancement") for duplicate or similar feature requests.
- [ ] I have not answered the above questions truthfully. 

### Feature request

When i start my computer, internet will take some time to connect (or it's take time to me for turn on wifi hotspot on my phone) and auto update will allway check and fail.So,how to turn off auto update ???

Sparronator9999, 2025-09-18:
Run `Updater.exe --setautoupdate false` in an elevated Command Prompt from YAMDCC's install directory. There is also an option in the update UI to disable it.

closed 2025-09-18 by Sparronator9999

VisionCrizzal, 2025-09-18:
oh.... ok thank a lot :))))